raspberrypi/linux #2412: [4.14.x] Unexpected delay when echoing to sysfs kobject, it's fine on 4.9.x

gtrainavicius, 2018-03-04:
The Pisound kernel module creates a sysfs object (see pisnd_led_attribute at https://github.com/raspberrypi/linux/blob/rpi-4.14.y/sound/soc/bcm/pisound.c#L1064) which allows flashing the MIDI LEDs from various scripts for visual feedback, as so:

`sudo sh -c "echo 10 > /sys/kernel/pisound/led"`

On 4.9.80 kernel it works without a perceptible delay. On 4.14.18 (The first 4.14 branch version available in rpi-update), however, this seems to act in a 'batched' way, where the very first call to this function waits for up to ~2.5 seconds, and successive calls that were made quickly all go through fast, until the next time it's batched.

Here I have run the same command a few times quickly by hitting up and enter a few times:

```
pi@raspberrypi:~ $ time { sudo sh -c "echo 10 > /sys/kernel/pisound/led"; }
real    0m2.486s
user    0m0.033s
sys     0m0.001s
pi@raspberrypi:~ $ time { sudo sh -c "echo 10 > /sys/kernel/pisound/led"; }
real    0m0.061s
user    0m0.033s
sys     0m0.001s
pi@raspberrypi:~ $ time { sudo sh -c "echo 10 > /sys/kernel/pisound/led"; }
real    0m0.062s
user    0m0.015s
sys     0m0.019s
pi@raspberrypi:~ $ time { sudo sh -c "echo 10 > /sys/kernel/pisound/led"; }
real    0m0.062s
user    0m0.020s
sys     0m0.014s
```

Any ideas what could be different from 4.9 kernel? (The module source code is the same)

Let me know if you'd like me to try something out.

pelwell, 2018-03-06:
Can you try your test with /sys/class/leds/led0/brightness (the green activity led)? Non-zero turns it on, zero turns it off. I don't see any appreciable delay.

gtrainavicius, 2018-03-08:
I can reproduce the same problem using led0 on 4.14.24:

```
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m2.335s
user    0m0.019s
sys     0m0.022s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.037s
user    0m0.012s
sys     0m0.008s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }
^[[A
real    0m2.163s
user    0m0.032s
sys     0m0.005s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.067s
user    0m0.024s
sys     0m0.013s
pi@raspberrypi:~ $
```

Just before upgrading from 4.9.79:

```
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.068s
user    0m0.020s
sys     0m0.010s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.068s
user    0m0.020s
sys     0m0.010s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.066s
user    0m0.030s
sys     0m0.000s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255 > /sys/class/leds/led0/brightness"; }

real    0m0.067s
user    0m0.030s
sys     0m0.000s
```

gtrainavicius, 2018-03-08:
Actually, turns out it's enough to run a 'sudo' command to get the strange delay in execution:

```
pi@raspberrypi:~ $ time { sudo sh -c "echo 255"; }
^[[A
255

real    0m2.341s
user    0m0.025s
sys     0m0.011s
pi@raspberrypi:~ $ time { sudo sh -c "echo 255"; }
255

real    0m0.066s
user    0m0.019s
sys     0m0.018s
```

So this is not specific to sysfs kernel objects.

popcornmix, 2018-03-08:
Possibly a hostname issue?
https://serverfault.com/questions/38114/why-does-sudo-command-take-long-to-execute

gtrainavicius, 2018-03-08:
I haven't changed the hostname from 'raspberrypi', and why would 4.9 kernel not have the same issue?

p.s., there is such an entry:

`127.0.1.1       raspberrypi`

I will try doing a clean Raspbian Lite install and check whether I can reproduce the issue on that.

gtrainavicius, 2018-03-08:
Ok, I have tried this on clean installs of lite and full versions, couldn't reproduce the issue.

I also did apt-get upgrade on all images, including the original on which I hit the issue, to bring the installed software to as similar versions as possible, and still the issue could be reproduces only on the original sd card, but not on clean installs.

What is weirdest is that after switching back to 4.9 kernel versions, the issue disappears.

Anyway, I will switch to a clean install and will keep an eye on whether this issue reappears and after what changes were done to the system.

closed 2018-03-08 by gtrainavicius