raspberrypi/linux #2414: NULL pointer dereference on RPI Zero after setting dtoverlay=dwc2

lategoodbye, 2018-03-04:
Hi,
i installed Raspbian Lite and updated to the latest firmware / kernel and rootfs. After that i enabled dwc2 via config.txt and then the kernel crashes always during boot with a NULL pointer dereference on my RPi Zero (strangly my Zero W isn't affected).

```
[    0.000000] Linux version 4.14.22+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1096 Mon Feb 26 18:53:26 GMT 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero Rev 1.3
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] random: fast init done
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=3e54eee7-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 435404K/458752K available (6329K kernel code, 561K rwdata, 1964K rodata, 440K init, 677K bss, 15156K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc06368c8   (6331 kB)
[    0.000000]       .init : 0xc08ac000 - 0xc091a000   ( 440 kB)
[    0.000000]       .data : 0xc091a000 - 0xc09a6728   ( 562 kB)
[    0.000000]        .bss : 0xc09ac074 - 0xc0a5551c   ( 678 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 23752 entries in 70 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000071] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000165] bcm2835: system timer (irq = 27)
[    0.000721] Console: colour dummy device 80x30
[    0.001310] console [tty1] enabled
[    0.001362] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060334] pid_max: default: 32768 minimum: 301
[    0.060862] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060917] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062241] Disabling memory control group subsystem
[    0.062461] CPU: Testing write buffer coherency: ok
[    0.063592] Setting up static identity map for 0x8200 - 0x8238
[    0.064890] devtmpfs: initialized
[    0.074732] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.075195] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.075269] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.076622] pinctrl core: initialized pinctrl subsystem
[    0.078006] NET: Registered protocol family 16
[    0.080953] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.087144] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.087216] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.087361] Serial: AMBA PL011 UART driver
[    0.090160] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.129589] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.132032] SCSI subsystem initialized
[    0.132319] usbcore: registered new interface driver usbfs
[    0.132457] usbcore: registered new interface driver hub
[    0.132690] usbcore: registered new device driver usb
[    0.140812] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-02-26 19:45
[    0.142705] clocksource: Switched to clocksource timer
[    0.228861] VFS: Disk quotas dquot_6.6.0
[    0.229057] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.229396] FS-Cache: Loaded
[    0.229751] CacheFiles: Loaded
[    0.247097] NET: Registered protocol family 2
[    0.248320] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.248456] TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
[    0.248576] TCP: Hash tables configured (established 4096 bind 4096)
[    0.248753] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.248813] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[... 104 lines cut ...]
[    4.098118] Process systemd-journal (pid: 60, stack limit = 0xda5de188)
[    4.098125] Stack: (0xda5dfe40 to 0xda5e0000)
[    4.098141] fe40: da52f120 da529d14 da4e5960 da4e5b1c da4e5ae0 da4e5960 da529cf8 da4e0114
[    4.098155] fe60: dac081a0 b6cfc000 da5dfed4 da5dfe78 c012e7cc c0124c14 00000000 00000000
[    4.098168] fe80: 00000000 00000000 0000001b b6cfb000 daacd714 b6cfc000 daef3960 00000000
[    4.098181] fea0: daacd724 00000000 00000000 da52f120 da4e5ae0 00000000 b6cfc000 00000000
[    4.098195] fec0: c09ac36c b6cfc000 da5dff04 da5dfed8 c012f840 c012e6c8 da52f120 00000000
[    4.098209] fee0: da4e5ae0 00000000 00100071 b6cfb000 dac081a0 00100073 da5dff14 da5dff08
[    4.098223] ff00: c012f910 c012f730 da5dff6c da5dff18 c01322d0 c012f8f0 00100071 da529cf8
[    4.098236] ff20: daef3960 0000001b 00000000 00000009 f3fa8ab8 00000000 00000001 00000000
[    4.098249] ff40: da5dff6c b6cfc000 00000000 00000001 00000005 b6cfc000 00000001 00000080
[    4.098263] ff60: da5dffa4 da5dff70 c0132438 c0132090 00100071 c00a53c8 da5dffa4 da4e5ae0
[    4.098277] ff80: b6f37ce8 b6f37ce8 b6f37ce8 0000007d c000ffa4 da5de000 00000000 da5dffa8
[    4.098291] ffa0: c000ff80 c0132324 b6f37ce8 b6f37ce8 b6cfb000 00001000 00000001 b6cfb000
[    4.098305] ffc0: b6f37ce8 b6f37ce8 b6f37ce8 0000007d b6cdcdb0 00000000 b6f37ce8 bebcfc14
[    4.098319] ffe0: 0002befc bebcfb44 b6f15c8c b6f23e6c 80000010 b6cfb000 1bffa861 1bffac61
[    4.098359] [<c0124c38>] (anon_vma_interval_tree_remove) from [<c012e7cc>] (__vma_adjust+0x110/0x644)
[    4.098382] [<c012e7cc>] (__vma_adjust) from [<c012f840>] (__split_vma+0x11c/0x1c0)
[    4.098404] [<c012f840>] (__split_vma) from [<c012f910>] (split_vma+0x2c/0x3c)
[    4.098422] [<c012f910>] (split_vma) from [<c01322d0>] (mprotect_fixup+0x24c/0x294)
[    4.098439] [<c01322d0>] (mprotect_fixup) from [<c0132438>] (SyS_mprotect+0x120/0x200)
[    4.098467] [<c0132438>] (SyS_mprotect) from [<c000ff80>] (__sys_trace_return+0x0/0x10)
[    4.098484] Code: e2807010 e5904014 e5905018 0a000002 (e5913004)
[    4.098495] ---[ end trace dbb4c81d4efb3deb ]---
```

JamesH65, 2018-05-11:
Sorry about the delayed response. Are you still seeing this issue?

lategoodbye, 2018-05-12:
What i've done today:
1. Copy the latest Raspbian-lite to sdcard
2. Try to boot on RPi Zero (without enabling dwc2) ->kernel oops (see below)
3. Place sdcard into RPi 1 B
4. Update to latest kernel/firmware
5. Place sdcard into RPi Zero (without enabling dwc2) -> still kernel oops (see below)

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.39+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1112 Sat May 5 11:40:12 BST 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero Rev 1.3
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=533cb580-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 435268K/458752K available (6418K kernel code, 587K rwdata, 1988K rodata, 436K init, 677K bss, 15292K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc064cda8   (6420 kB)
[    0.000000]       .init : 0xc08c9000 - 0xc0936000   ( 436 kB)
[    0.000000]       .data : 0xc0936000 - 0xc09c8e50   ( 588 kB)
[    0.000000]        .bss : 0xc09ce784 - 0xc0a77ce4   ( 678 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24082 entries in 71 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000028] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000072] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000163] bcm2835: system timer (irq = 27)
[    0.000714] Console: colour dummy device 80x30
[    0.001285] console [tty1] enabled
[    0.001336] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060338] pid_max: default: 32768 minimum: 301
[    0.060856] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060913] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062238] Disabling memory control group subsystem
[    0.062455] CPU: Testing write buffer coherency: ok
[    0.063580] Setting up static identity map for 0x8200 - 0x8238
[    0.064864] devtmpfs: initialized
[    0.073646] random: get_random_u32 called from bucket_table_alloc+0x88/0x1c4 with crng_init=0
[    0.074535] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.074978] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.075051] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.076395] pinctrl core: initialized pinctrl subsystem
[    0.077770] NET: Registered protocol family 16
[    0.080718] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.087005] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.087075] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.087219] Serial: AMBA PL011 UART driver
[    0.089987] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.130713] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.133048] SCSI subsystem initialized
[    0.133324] usbcore: registered new interface driver usbfs
[    0.133476] usbcore: registered new interface driver hub
[    0.133714] usbcore: registered new device driver usb
[    0.140808] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-05-08 17:30
[    0.142795] clocksource: Switched to clocksource timer
[    0.230413] VFS: Disk quotas dquot_6.6.0
[    0.230596] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.230924] FS-Cache: Loaded
[    0.231286] CacheFiles: Loaded
[    0.248158] NET: Registered protocol family 2
[... 45 lines cut ...]
[    3.791424] PC is at mutex_lock+0x30/0x5c
[    3.808274] LR is at __local_bh_enable_ip+0x98/0x114
[    3.825987] pc : [<c06482c0>]    lr : [<c0026504>]    psr: 60000013
[    3.845090] sp : da48fed8  ip : da48fea8  fp : da48feec
[    3.863096] r10: c06482ac  r9 : dac08000  r8 : c0026478
[    3.881049] r7 : bf00da5c  r6 : da48fec0  r5 : da48fed4  r4 : ffffbb44
[    3.900413] r3 : dae1ef00  r2 : 00000000  r1 : 00000001  r0 : 00000000
[    3.919780] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    3.939993] Control: 00c5387d  Table: 1a4e0008  DAC: 00000055
[    3.958737] Process kworker/0:2 (pid: 37, stack limit = 0xda48e188)
[    3.978027] Stack: (0xda48fed8 to 0xda490000)
[... cut ...]
[    4.124858] ff80: ffffffff dafe9320 c003f29c 00000000 00000000 00000000 00000000 00000000
[    4.146761] ffa0: 00000000 da48ffb0 c000fe6c c003f2a8 00000000 00000000 00000000 00000000
[    4.168597] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    4.190419] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555555
[    4.212104] Code: e1a0300d e3c33d7f e3c3303f e593300c (e1941f9f)
[    4.231906] ---[ end trace 28635624149ac4c1 ]---
[   48.378895] random: crng init done
```

lategoodbye, 2018-05-14:
Today i took the last Raspbian version with Linux 4.9 and this crashes before resizing the SD card. I guess this is a hardware issue :-(

So don't waste your time on this.

closed 2018-05-14 by lategoodbye