raspberrypi/linux #2415: Feature request: Move bluetooth firmware into brcm directory

lategoodbye, 2018-03-04:
Hi,
i hope this the right place for this feature request. Raspbian is a good rootfs for my upstream work. In order to make my life just a little bit easier it would be nice to move the bluetooth firmware ( /lib/firmware/BCM43430A1.hcd ) into the /lib/firmware/brcm directory. After this change the BT firmware could be loaded automatically without hciattach by the upstream driver (scheduled for 4.17).

XECDesign, 2018-03-04:
@pelwell Shall I?

lategoodbye, 2018-03-04:
Yes, with the current location i will get this error:

```
[    5.103374] bluetooth hci0: Direct firmware load for brcm/BCM43430A1.hcd failed with error -2
[    5.103390] Bluetooth: hci0: BCM: Patch brcm/BCM43430A1.hcd not found
```

No need to hurry. Maybe a symlink could also be an option.

XECDesign, 2018-03-06:
Done. Should show up in the repo soon.

lategoodbye, 2018-03-07:
Thanks a lot

closed 2018-03-07 by lategoodbye