raspberrypi/linux #2416: Overlay node order is inverted

lucastanure, 2018-03-05:
My machine driver receives the opposite order of nodes defined on device tree overlay. 
Using this overlay :
`/dts-v1/;
/plugin/;
/{
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&sound>;
		__overlay__ {
			status = "okay";
			compatible = "dts,test-order";

			label0 {
			};
			label1 {
			};
		};
	};
};`

With this driver:

`#include <linux/module.h>
#include <linux/of.h>
#include <linux/platform_device.h>

static int dts_test_probe(struct platform_device *pdev)
{
	struct device_node *dai;
	struct device_node *np = pdev->dev.of_node;

	dev_info(&pdev->dev,"dts_test_probe\n");
	for_each_child_of_node(np, dai) {
		dev_info(&pdev->dev,"Label Name %s\n", dai->name);
	}
	return 0;
}

#if IS_ENABLED(CONFIG_OF)
static const struct of_device_id dts_test_of_match[] = {
	{ .compatible = "dts,test-order", },
	{},
};
MODULE_DEVICE_TABLE(of, dts_test_of_match);
#endif /* CONFIG_OF */

static struct platform_driver dts_test_driver = {
	.driver		= {
		.name	= "dts-test-order",
		.owner	= THIS_MODULE,
		.of_match_table = of_match_ptr(dts_test_of_match),
	},

	.probe		= dts_test_probe,
};

module_platform_driver(dts_test_driver);`

I got this on dmesg:

[    4.590522] dts-test-order soc:sound: dts_test_probe2
[    4.590542] dts-test-order soc:sound: Label Name label1
[    4.590551] dts-test-order soc:sound: Label Name label0

lucastanure, 2018-03-05:
From  [https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/of/fdt.c#L451](url)

pelwell, 2018-03-05:
[ The issue report has been mangled - use ` ``` ` on the lines before and after for a block code quote ]

Overlays are applied by the firmware before the kernel runs. The order of fragments is irrelevant unless you are applying overlays at runtime, and specifically it has no impact on the order in which drivers are probed.

In general there is no easy way to guarantee drivers are loaded in a particular order. Instead drivers are supposed to handle the absence of any pre-requisites by return -EPROBE_DEFER from their probe functions. This causes the kernel to re-probe after all subsequent drivers are loaded until the probe succeeds. It is common not to output a kernel message when returning -EPROBE_DEFER to avoid worrying users about a harmless and unavoidable situation, but this does mean that a genuinely missing will cause a silent failure.

lucastanure, 2018-03-05:
I'm not worried about the order of fragments, but the order of nodes internally to a fragment.

pelwell, 2018-03-05:
See my second paragraph.

pelwell, 2018-03-05:
It would have been easier to answer if the code sections hadn't been deleted when I came to reply.

lucastanure, 2018-03-05:
Sorry, I was editing the issue, I did put the code there, you can see it now?

JamesH65, 2018-03-05:
As I understand it, there is no guarantee that the order of items in the DT is the order in which they are read. I think I fell over that one a while back.

pelwell, 2018-03-05:
Yes, I can see it. Think of nodes as being sets of sub-nodes and properties rather than lists - order cannot be guaranteed. I think that is the reason behind the reversal - to guarantee that assumptions about node order are not met.

If you want to explicitly enumerate sub-nodes from your code in a specific order then you will have to look for them by name, but if you are hoping to have their drivers probed in a specific order then you are going to be disappointed.

lucastanure, 2018-03-05:
But why commenting it out : 
```
	if (!dryrun)
		reverse_nodes(root);
```
from drivers/of/fdt.c, line 451, make it everything works ?

pelwell, 2018-03-05:
Looking at the history of the code in that area it appears that it was actually trying to preserve the order of sub-nodes, but for some reason it no longer works. That is all upstream code and problems in it should be reported to the Linux maintainers, but I think you would be better off not assuming a fixed node order.

pelwell, 2018-03-05:
This was bugging me, so I ran a few tests. Picking a node at random (the GPIO controller) I dumped out the names of the children in order and found them to be as expected. Tracing added to the reverse_nodes function showed that the labels were being processed in the correct order, so the reversal was putting them into the wrong order.

Finally I captured the combined DTB generated by the firmware before the kernel is launched and found found this:
```
        sound {
            compatible = "dts,test-order";
            status = "okay";
            linux,phandle = <0x00000072>;
            phandle = <0x00000072>;
            label1 {
            };
            label0 {
            };
        };
```

The firmware appears to be merging nodes in the reverse order (correction - it is merging them in the correct order but they are ending up in the reverse order), but it isn't currently obvious why that would be...

pelwell, 2018-03-05:
It's a feature of libfdt, the Open Source library for manipulating Flattened Device Trees. The firmware, the Pi-specific dtmerge utility and the (new) standard fdtoverlay utility all use a function called `fdt_add_subnode`(*) to add each node. The documentation says:
```
 * fdt_add_subnode() creates a new node as a subnode of the node at
 * structure block offset parentoffset, with the given name (which
 * should include the unit address, if any).
```
What it doesn't say is where the subnode should be added. A comment in the code makes it clearer:
```
	/* Try to place the new node after the parent's properties */
```
Since FDT stores the subnodes after the properties, this has the effect of inserting the new subnode before any others. And since the source nodes are enumerated in the order they appear, the end result is a reversal.

As `fdtoverlay` behaves the same way, I see this as an unexpected but harmless quirk.

(*) Actually the `fdt_add_subnode_namelen` variant, but they are the same for the purposes of this explanation.

lucastanure, 2018-03-06:
Great, Thanks for the help. 
This is something that we would like to fix ?

pelwell, 2018-03-06:
I'm not going to modify the way the overlays are merged - we are doing it the expected way, and it works. Now that libfdt supports overlay application we are more likely to use the upstream support than we are to modify our own downstream code.

If you care about the order (and I think you would be better not to) then you should reverse the order in your overlay.

closed 2018-03-13 by pelwell