raspberrypi/linux #2420: ieee80211 phy0: rt2800usb_tx_sta_fifo_read_completed: Warning - TX status read failed -71

ccztux, 2018-03-06:
I use the following usb wifi adapter on a raspberry pi 2 as an ap with hostapd:
```
ALFA Networks AWUS052NH | 2.4 / 5 GHz, USB WLAN Adapter, 300 MBit, Ralink RT3572
```

Kernel version:
```
Linux wurlitzer 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
```

I have deactivated hardware encryption:
```
cat /etc/modprobe.d/rt2800usb.conf
options rt2800usb nohwcrypt=1
```

Ip settings:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether b8:27:eb:10:21:87 brd ff:ff:ff:ff:ff:ff
3: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether e6:66:da:2f:90:72 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.1/24 brd 10.0.0.255 scope global br0
       valid_lft forever preferred_lft forever
    inet6 fe80::e466:daff:fe2f:9072/64 scope link 
       valid_lft forever preferred_lft forever
5: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br0 state DOWN group default qlen 1000
    link/ether 00:c0:ca:8f:c5:44 brd ff:ff:ff:ff:ff:ff
```

After running hostapd a short time, the connections break and there  plenty of this log messages are written to /var/log/messages:
```
Mar  6 19:31:55 localhost kernel: [77125.660664] ieee80211 phy0: rt2800usb_tx_sta_fifo_read_completed: Warning - TX status read failed -71
```

Extracted data from /var/log/messages:
```
Mar  6 19:27:45 localhost kernel: [76874.917882] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
Mar  6 19:27:45 localhost kernel: [76874.920113] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
Mar  6 19:27:46 localhost kernel: [76876.213589] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar  6 19:27:46 localhost kernel: [76876.220073] br0: port 2(wlan0) entered blocking state
Mar  6 19:27:46 localhost kernel: [76876.220089] br0: port 2(wlan0) entered disabled state
Mar  6 19:27:46 localhost kernel: [76876.221142] device wlan0 entered promiscuous mode
Mar  6 19:27:54 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.108 (10.0.0.1) from 60:d9:a0:68:43:6e (android-8c2ed927ce08942b) via br0
Mar  6 19:27:54 localhost dhcpd[490]: DHCPACK on 10.0.0.108 to 60:d9:a0:68:43:6e (android-8c2ed927ce08942b) via br0
Mar  6 19:27:57 localhost kernel: [76887.206972] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mar  6 19:27:57 localhost kernel: [76887.207133] br0: port 2(wlan0) entered blocking state
Mar  6 19:27:57 localhost kernel: [76887.207141] br0: port 2(wlan0) entered listening state
Mar  6 19:28:12 localhost kernel: [76902.268743] br0: port 2(wlan0) entered learning state
Mar  6 19:28:15 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.20 from 14:91:82:ed:e3:f2 via br0
Mar  6 19:28:15 localhost dhcpd[490]: DHCPACK on 10.0.0.20 to 14:91:82:ed:e3:f2 via br0
Mar  6 19:28:27 localhost kernel: [76917.628872] br0: port 2(wlan0) entered forwarding state
Mar  6 19:28:27 localhost kernel: [76917.628915] br0: topology change detected, sending tcn bpdu
Mar  6 19:28:50 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.43 from b8:27:eb:04:b3:82 via br0
Mar  6 19:28:50 localhost dhcpd[490]: DHCPACK on 10.0.0.43 to b8:27:eb:04:b3:82 via br0
Mar  6 19:28:51 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.41 from d8:9d:67:cd:ae:0e via br0
Mar  6 19:28:51 localhost dhcpd[490]: DHCPACK on 10.0.0.41 to d8:9d:67:cd:ae:0e via br0
Mar  6 19:30:03 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.108 from 60:d9:a0:68:43:6e (android-8c2ed927ce08942b) via br0
Mar  6 19:30:03 localhost dhcpd[490]: DHCPACK on 10.0.0.108 to 60:d9:a0:68:43:6e (android-8c2ed927ce08942b) via br0
Mar  6 19:30:45 localhost dhcpd[490]: DHCPDISCOVER from 84:3a:4b:81:95:44 via br0
Mar  6 19:30:46 localhost dhcpd[490]: DHCPOFFER on 10.0.0.109 to 84:3a:4b:81:95:44 (tux01) via br0
Mar  6 19:30:46 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.109 (10.0.0.1) from 84:3a:4b:81:95:44 (tux01) via br0
Mar  6 19:30:46 localhost dhcpd[490]: DHCPACK on 10.0.0.109 to 84:3a:4b:81:95:44 (tux01) via br0
Mar  6 19:31:26 localhost dhcpd[490]: DHCPDISCOVER from 1c:23:2c:6e:d3:d1 via br0
[... 46 lines cut ...]
Mar  6 19:31:57 localhost kernel: [77126.910607] usb 1-1.5: reset high-speed USB device number 5 using dwc_otg
Mar  6 19:31:57 localhost kernel: [77127.049749] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3572, rev 0223 detected
Mar  6 19:31:57 localhost kernel: [77127.071301] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 0009 detected
Mar  6 19:31:57 localhost kernel: [77127.076374] ieee80211 phy1: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
Mar  6 19:31:57 localhost kernel: [77127.076473] ieee80211 phy1: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
Mar  6 19:31:57 localhost mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
Mar  6 19:31:57 localhost mtp-probe: bus: 1, device: 5 was not an MTP device
Mar  6 19:31:58 localhost kernel: [77128.363965] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar  6 19:31:58 localhost kernel: [77128.450676] br0: port 2(wlan0) entered blocking state
Mar  6 19:31:58 localhost kernel: [77128.450699] br0: port 2(wlan0) entered disabled state
Mar  6 19:31:58 localhost kernel: [77128.451442] device wlan0 entered promiscuous mode
Mar  6 19:32:31 localhost dhcpd[490]: DHCPREQUEST for 10.0.0.10 from 24:5e:be:02:77:d0 via br0
Mar  6 19:32:31 localhost dhcpd[490]: DHCPACK on 10.0.0.10 to 24:5e:be:02:77:d0 via br0
```

Could this be the issue, of disconnection?:
```
Mar  6 19:31:56 localhost kernel: [77125.791302] usb 1-1-port2: over-current change
```

What's it all about the excessive logging?

I have tested this also with the newest kernel version, updated via raspi-config, but the error still exists. Now I have reinstalled the rasbpian stock kernel.

Please let me know, if I should provide more infos.

rodizio1, 2018-03-09:
That card is very power hungry, it creates a high current spike (above 1.3 amps) when the interface is brought up and pulls up to 2 amps (yes, 10 Watts) when transmitting. 

You need a good power supply and short good quality USB cables to rule out power-supply problems first.

pelwell, 2018-03-09:
Thanks for your insight, @rodizio1.

ccztux, 2018-03-09:
Thank you for pointing this out! I will use another USB wifi adapter.

closed 2018-03-09 by ccztux