raspberrypi/linux #2421: 'make ARCH=arm64 dtbs' in 4.15.6 doesn't generate overlays?

davidw5, 2018-03-07:
(This is my first post in this forum, so my apologies if this question is inappropriate for it.)

Running 'make ARCH=arm64 dtbs' for release 4.15.6 wasn't generating the overlay dtbo files for me. The overlays generated ok under 4.14.9, but I see there has been a substantial change to arch/arm64/boot/dts/Makefile since then.

Adding 'subdir-y += overlays' to arch/arm64/boot/dts/Makefile does 'fix' the problem (i.e. generates the .dtbo files under the make command), but I'm wondering if I'm missing something (being newish to kernel builds), since I would have thought if it were that obvious this would already have been added, and I can't see any changes in rpi-4.16.y that would change this behaviour?

Thanks.

pelwell, 2018-03-07:
I had to make similar changes in the arm dts Makefile, but although we host arm64 patches we don't support them ourselves. If you submit a patch as a Pull Request then I'll add it to the tree.

davidw5, 2018-03-07:
Thanks, but I'm also new to Pull Requests; do I have to create a fork of the full linux tree to generate a Pull Request?

pelwell, 2018-03-07:
Yes, but the GitHub GUI makes it easy. Find a file you want to change, click on it to view the contents, then click the pencil icon to edit it. At this point it will create a fork for you. Once you've made your changes, fill out the commit message at the bottom of the page and commit/submit/whatever the button says. Then you'll be able to create a PR from it. The official documentation is [here](https://help.github.com/articles/creating-a-pull-request/).

If for some reason you can't get it to work, post your diff here and I'll take care of the rest.

davidw5, 2018-03-07:
Thanks again; I appreciate your help! I believe I've now made a pull request #2423 "Update dts Makefile to include overlays directory".

closed 2018-03-07 by davidw5