raspberrypi/linux #2427: Raspberry Pi 3 wlan0 interface failing after network failure

FalcoGer, 2018-03-09:
I use a raspberry pi with 2017-11-29-raspbian-stretch. 

The issue:
I use wpa_supplicant as it is the default. I entered the ssid and psk into /etc/wpa_supplicant/wpa_supplicant.conf
and I also added to /etc/dhcpcd.conf:
```
interface wlan0
static ip_address=192.168.115.243
static routers=192.168.115.1
static domain_name_servers 192.168.115.1 8.8.8.8

interface eth0
static ip_address=192.168.115.17
static routers=192.168.115.1
static domain_name_servers 192.168.115.1 8.8.8.8
```
upon boot the wifi connects automatically (ethernet is disconnected at this point), which is fine.
Now the problem is when I simulate a network failure (by unplugging the power to my router and replugging it), the the pi doesn't automatically reconnect.
So to solve this problem i went to every corner of the internet and tried to find a solution.
1: ifup --force wlan0, fails with wlan0 doesn't exist
(ifconfig shows it as wlan0)
2: enter stuff to /etc/networking/interfaces, works for booting but wlan still fails after disconnecting

I noticed after disconnecting I could no longer scan networks:
> iwlist wlan0 scanning
results in
> wlan0 interface doesn't support scanning: connection timed out

ifconfig wlan0 down and up again, same result

after a reboot:
I killed the wpa_supplicant service and started it anew with
wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf and it works.
then I disconnected by cutting power to my router again and ran the same command after wlan was back up:
> Successfully initialized wpa_supplicant
> Failed to created interface p2p-dev-wlan0: -110 (connection timed out)
> nl80211: failed to create a P2P device interface p2p-dev-wlan0
> P2P: Failed to enable P2P device interface
> wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
> wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
> ...

I'm at a loss what to do except for doing a full on reboot every time I loose the net...

I don't think it matters but
Imediately after installing I did the following things (as root):

> apt-get purge wolfram-engine libreoffice libreoffice-core libreoffice-common
> apt-get update
> apt-get upgrade
> apt-get autoremove
> apt-get autoclean
> apt-get install xrdp mono-complete mono-vbnc nginx
> rpi-update

locale settings, hostname change and ssh enable with raspi-config

FalcoGer, 2018-03-09:
`uname -a`
> Linux hostname-pi 4.14.27-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l GNU/Linux

FalcoGer, 2018-03-09:
The problem appears to be assigning a static ip to the interface in dhcpcd.conf

pelwell, 2018-03-09:
Both interfaces are currently on the same network, i.e.:
```
(address(eth0) & netmask(eth0)) == (address(wlan0) & netmask(wlan0))
```
(192.168.x.y is a set of 254 class C networks - the default netmask is 255.255.255.0)

Can you try putting one of them on a different network, if only to rule that out as being a possible issue?

teddythomas, 2018-03-09:
I was going to report the same issue and as I was connecting on GitHub and I am seeing it just got reported. If it may help, I tested four scenarios:
- Latest official 4.9 kernel release: problem not happening
- Locally built kernel 4.9 (commit f263ce8  from March 6th): problem not happening
- Locally built kernel 4.14 (commit 5e2d4ac539f8007e7a2c0e86956b53680f764ff1 from Feb 9th): issue is happening
- Locally built kernel 4.14 (commit 80a14a56dacb7cc2b40d5f37d00bedb0ceace130 from March 5th): issue is happening

All of the tests above have been done with image 2017-11-29-raspbian-stretch, update to date (apt-get update && apt-get dist-upgrade) as of yesterday or the day before

pelwell, 2018-03-09:
`sudo rpi-update` will install an official build of the rpi-4.14.y branch, if you want to rule out finger trouble on your part.

JamesH65, 2018-03-09:
See here...

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=207345
https://www.raspberrypi.org/forums/viewtopic.php?t=205485

teddythomas, 2018-03-09:
Thank you to both of you.
- I did the `sudo rpi-update` and faced the issue.
After going through the page https://www.raspberrypi.org/forums/viewtopic.php?t=205485, I tested with the firmware shared in the post https://www.raspberrypi.org/forums/search.php?author_id=406&sr=post. I got the following (successful) results:
- Locally built kernel 4.14 (commit 80a14a5 from March 5th): problem is not happening
- Kernel updated via `sudo rpi-update`: problem is not happening
I did not test with a 4.9 kernel though.

teddythomas, 2018-03-19:
I am not the original reporter of this issue, but I believe this issue _could_ be closed.
The new firmware `firmware-brcm80211` published last week (installed via `apt-get dist-upgrade`) and running with the latest 4.14 kernel (installed via `rpi-update`) is working properly on my side, at least.

JamesH65, 2018-03-19:
Closing this issue as questions answered/issue resolved.

closed 2018-03-19 by JamesH65

FalcoGer, 2018-03-26:
After
```
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-update
reboot
```
i can now scan networks with
`sudo iwlist wlan0 scanning`
after cutting power to the router.
however the icon in the pannel that shows connection status ("Wireless & Wired Network Settings") doesn't show networks anymore.
Also the pi doesn't automatically reconnect when the router comes back on. What's with that?
What I could do however was
`sudo wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf`
and it connected. I guess that's a start.

JamesH65, 2018-03-26:
Odd. In my testing I saw all that working. 

Is there anything unusual with your setup?

Commander-Crash, 2019-11-05:
So what was the resolver for this issue? I upgrade the distro and rpi-upgrade only to make things worse. One may say to start a new but thats like 6 months of progress.