raspberrypi/linux #2430: Missing snd-bcm2835.ko audio kernel module for 64 bit aarch64

Roland74, 2018-03-10:
Hi,
I recently tried to build a 64 Bit kernen on my Raspberry Pi 3 on Raspbian following the instructions in http://www.tal.org/tutorials/raspberry-pi3-build-64-bit-kernel
git clone --depth=1 -b rpi-4.9.y https://github.com/raspberrypi/linux.git
mkdir kernel-out
cd linux
make O=../kernel-out/ ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-  bcmrpi3_defconfig
The build completed but the audio kernel module snd-bcm2835.ko is missing afterwards.
After installing the kernel and kernel modules I was able to boot the 64 Bit kernel with wifi and X11 but I was not able to use sound.  I looked into kernel-out and did not find the snd-bcm2835.ko. There were other snd\*.ko modules but that does not help me to get the sound working. I looked into bcm2709_defconfig and there were some lines bcmrpi3_defconfig does not have especially
CONFIG_SND_BCM2835=m
That looks strange. Are you able to get sound working for 64 Bit. What happened to snd.bcm2835.ko and why is 
CONFIG_SND_BCM2835=m
missing in bcmrpi3_defconfig?

Another issue I found after compiling the 64 Bit kernel was that I cannot get the processor temperature any more:
/opt/vc/bin/vcgencmd measure_temp
VCHI initialization failed

Even with sudo it does not help. Maybe I should open another issue? 
Ok for this issue it would be great if you could give me any advice. I currently try to build 
git clone --depth=1 -b rpi-4.10.y https://github.com/raspberrypi/linux.git

Kind regards,
  Roland

JamesH65, 2018-03-10:
We don't support a 64 bit kernel, most of the work done on it has been done by third parties. There are a number of reasons, but a big one is that the interface with the VC4 breaks in a number of places because the VC4 is 32 bit. Issues you are seeing are probably related to that, especially anything to do with VCHI.

Roland74, 2018-03-10:
Hi James,
thank you for your answer. It was reported that sound already worked with the 64 Bit kernel . After building kernel 4.10 sound works again. Thank God! :-) I can even watch youtube videos in the browser.
OK, I can do that with the 32 Bit kernel, too but I wanted to run Skype.
I foolowed the instructions on
https://eltechs.com/run-skype-on-raspberry-pi/
and bought exagear to be able to run 32 Bit x86 Skype 4.3. I was able to start Skype but found out that 32 Bit Skype for linux was stopped. The support team for exagear suggested to build a 64 Bit kernel on my Raspberry 3. Then they would offer me exagear 64 Bit that allows to run skype for x86_64 Linux. So that was my motivation to build the 64 Bit kernel with sound support.
For VC4 I don't know. As I wrote I just tested to run youtube in the browser and that worked so far. I didn't use full display so maybe that does not work?
For me it's OK to boot from time to time into 64 Bit to use Skype and boot afterwards into the 32 Bit kernel. So it's good that people work on the 64 Bit kernel, too and it would be great if people continue to support the 64 Bit kernel even if there are some parts still not working, yet.
For me it seems that there was a problem in bcmrpi3_defconfig that prevented to build the sound module. I will check later with kdiff3 and let you know.

Roland74, 2018-03-10:
Hi,
I want to share some output I could collect:

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.10.17-v8+ #1 SMP PREEMPT Sat Mar 10 10:32:23 CET 2018 aarch64 GNU/Linux
pi@raspberrypi:~ $ lscpu
Architecture:          aarch64
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
Model:                 4
CPU max MHz:           1200.0000
CPU min MHz:           600.0000
BogoMIPS:              38.40
Flags:                 fp asimd evtstrm crc32
pi@raspberrypi:~ $ lsmod | sort
bluetooth             393216  29 hci_uart,bnep,btbcm,rfcomm
bnep                   24576  2
brcmfmac              262144  0
brcmutil               20480  1 brcmfmac
btbcm                  16384  1 hci_uart
cfg80211              671744  1 brcmfmac
cmac                   16384  1
evdev                  24576  1
fuse                  114688  3
hci_uart               32768  1
hid_logitech_dj        24576  0
hid_logitech_hidpp     32768  0
i2c_dev                20480  0
ip_tables              28672  0
ipv6                  462848  40
joydev                 20480  0
Module                  Size  Used by
rfcomm                 53248  6
rfkill                 36864  6 bluetooth,cfg80211
snd                    98304  3 snd_timer,snd_bcm2835,snd_pcm
snd_bcm2835            36864  0
snd_pcm               126976  1 snd_bcm2835
snd_timer              36864  1 snd_pcm
uio                    24576  1 uio_pdrv_genirq
uio_pdrv_genirq        16384  0
x_tables               45056  1 ip_tables

Current CPU frequency (at idle state):
pi@raspberrypi:~ $  cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
600000
pi@raspberrypi:~ $ vcgencmd get_config arm_freq
vchiq_lib: Very incompatible VCHIQ library - cannot retrieve driver version
VCHI initialization failed

CPU temperature:
pi@raspberrypi:~ $ vcgencmd measure_temp
vchiq_lib: Very incompatible VCHIQ library - cannot retrieve driver version
VCHI initialization failed
pi@raspberrypi:~ $ cat /sys/class/thermal/thermal_zone0/temp
39704
pi@raspberrypi:~ $
So vcgencmd currently does not work but it's possible to get the temperature and CPU frequency.
I would say that is another issue but not so important.
Kind regards,
 Roland

Roland74, 2018-03-13:
Hi,
so now I had time to do a diff between bcmrpi3_defconfig of kernel 4.9 and 4.10
The version of kernel 4.9 is /home/pi/bcmrpi3_defconfig
and of kernel 4.10 is in ./arch/arm64/configs/bcmrpi3_defconfig

--- /home/pi/bcmrpi3_defconfig  2018-03-10 00:03:36.626718868 +0100
+++ ./arch/arm64/configs/bcmrpi3_defconfig      2018-03-10 07:11:30.834072383 +0100
@@ -114,8 +114,6 @@
 CONFIG_NF_CONNTRACK_ZONES=y
 CONFIG_NF_CONNTRACK_EVENTS=y
 CONFIG_NF_CONNTRACK_TIMESTAMP=y
-CONFIG_NF_CT_PROTO_DCCP=m
-CONFIG_NF_CT_PROTO_UDPLITE=m
 CONFIG_NF_CONNTRACK_AMANDA=m
 CONFIG_NF_CONNTRACK_FTP=m
 CONFIG_NF_CONNTRACK_H323=m
@@ -178,7 +176,6 @@
 CONFIG_NETFILTER_XT_MATCH_RATEEST=m
 CONFIG_NETFILTER_XT_MATCH_REALM=m
 CONFIG_NETFILTER_XT_MATCH_RECENT=m
-CONFIG_NETFILTER_XT_MATCH_SOCKET=m
 CONFIG_NETFILTER_XT_MATCH_STATE=m
 CONFIG_NETFILTER_XT_MATCH_STATISTIC=m
 CONFIG_NETFILTER_XT_MATCH_STRING=m
@@ -581,7 +578,6 @@
 # CONFIG_BCM2835_DEVGPIOMEM is not set
 # CONFIG_BCM2835_SMI_DEV is not set
 # CONFIG_LEGACY_PTYS is not set
-# CONFIG_DEVKMEM is not set
 CONFIG_SERIAL_8250=y
 # CONFIG_SERIAL_8250_DEPRECATED_OPTIONS is not set
 CONFIG_SERIAL_8250_CONSOLE=y
@@ -846,6 +842,7 @@
 CONFIG_SND_MTPAV=m
 CONFIG_SND_SERIAL_U16550=m
 CONFIG_SND_MPU401=m
+CONFIG_SND_BCM2835=m
 CONFIG_SND_USB_AUDIO=m
 CONFIG_SND_USB_UA101=m
 CONFIG_SND_USB_CAIAQ=m
@@ -1100,7 +1097,6 @@
 CONFIG_LIRC_IMON=m
 CONFIG_LIRC_RPI=m
 CONFIG_LIRC_SASEM=m
-CONFIG_LIRC_SERIAL=m
 CONFIG_FB_TFT=m
 CONFIG_FB_TFT_AGM1264K_FL=m
 CONFIG_FB_TFT_BD663474=m
@@ -1131,7 +1127,6 @@
 CONFIG_FB_TFT_WATTEROTT=m
 CONFIG_FB_FLEX=m
 CONFIG_FB_TFT_FBTFT_DEVICE=m
-# CONFIG_BCM2708_VCHIQ is not set
 CONFIG_MAILBOX=y
 CONFIG_BCM2835_MBOX=y
 # CONFIG_IOMMU_SUPPORT is not set

Now this diff can help you to get the 4.9 kernel running with sound support.
By the way it would be great to get at least a bit feedback ;)
Kind regards,
  Roland

pelwell, 2018-03-13:
```
vchiq_lib: Very incompatible VCHIQ library - cannot retrieve driver version
```
This isn't a good sign.

Post the output from dmesg and we may spot something, but as JamesH65 said, "We don't support a 64 bit kernel".

Roland74, 2018-03-13:
Hi Pelwell,
tank you for your answer. Here my dmesg after running the machine for about 2 days.
[dmesg.txt](https://github.com/raspberrypi/linux/files/1806298/dmesg.txt)

Currently what I tested ran stable. 
Best regards,
  Roland

pelwell, 2018-03-13:
I noticed one oddity:
```
[    0.310344] gpio gpiochip1: GPIO chip parent is NULL
```
but I can't see how that would explain your problem.

Why are you using 4.10? I suggest you switch to 4.14 which is what we are developing on now and is likely be supported for a year. Even if it doesn't immediately solve the problem it will make it easier to compare configurations, and if we did find a fix on 4.10 we won't be updating that branch.

Roland74, 2018-03-13:
Hi Pelwell,
thank you for your answer. Maybe I did not explain good enough. I tried out 4.9 as it was suggested on two sides that explained how to build a 64 Bit kernel for Raspberry Pi 3. The  kernel 4.9.80 had a problem in bcmrpi3_defconfig that leads to an uncomplete sound module build. In the config file 
CONFIG_SND_BCM2835=m
was missing => no snd-bcm2835.ko was build for 4.9.80
For the 4.10 branch this problem is already fixed. So my suggestion was to update bcmrpi3_defconfig in case someone wants ro build the 4.9.x kernel with full sound support. OK if that old 4.9 branch will not be updated then we need not update the config file there, too. In case another person has problems with snd-bcm2835.ko then the information is, that the branch 4.10 works with sound.
Nevertheless I will go to the latest 4.14 branch and will try that out.
The only problem my 4.10 kernel had was, that  the vcgencmd command does not work. I am curious if that works on 4.14 :-)
Best regards,
  Roland

pelwell, 2018-03-13:
Your methodology was good, it is only the choice of the alternative branch which was a problem. 4.10 was only ever a temporary stepping stone, as were 4.11. 4.12 and 4.13 - 4.14 is the new king in waiting, and it will probably be crowned in the next few weeks.

Roland74, 2018-03-13:
Thank you Phil for your explanation. I am curious to try out 4.14 and will report :)

raymondjclark, 2018-03-15:
I build 4.14 for 64 bit today and receive a [0.190323] dmi: firmware registration failed message when trying to boot.

Roland74, 2018-03-15:
Hi Ray,
I got that message for 4.10, too:
[    0.183808] dmi: Firmware registration failed.
[    0.184446] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-10-24 17:09
Neverheless it continued booting and  seemed to works fine on my Raspberry.
Did your boot process stop afterwards or did it continue and work?
The last days I had no time to build the latest 4.14.
Kind regards,
  Roland

raymondjclark, 2018-03-15:
It stops there and that is it.

Roland74, 2018-03-15:
Hi Ray,
Oh, that really sounds bad. What was the last 64 Bit kernel that you were able to run successfull? 
It seems I really had luck with my 4.10 kernel. Unfortunately I haven't had time to test a new kernel the last few days.
Kind regards,
  Roland

Roland74, 2018-03-16:
Hi,
today I built the latest 4.14 kernel and got problems during booting:
raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
This message is shown every second.
The 4.10 kernel seems to be more stable for 64 Bit than 4.14.
[image]
Kind regards,
  Roland

pelwell, 2018-03-16:
Those messages just indicate that your firmware doesn't support the new mailbox method of querying the voltage and temperature throttling status. Run rpi-update to get a new firmware, our just copy it from the firmware repo.

Roland74, 2018-03-16:
Hi Phil,
thank you for your answer.  OK, for now I need to boot in the old 64 Bit or 32 Bit kernel by changing /boo/config.txt to run the update. I will try to edit it today evening or tomorrow. Is it possible to use this the new "firmware" with the old 32 Bit 4.9 kernel of Raspbian, too?
Currently I have to admit that I don't know what this firmware files? exactly are and where they are stored. Is it a binary blob? It reminds me on a Bios update like for PCs but there is no Bios on Raspberry, is there?
Kind regards,
  Roland

pelwell, 2018-03-16:
What we call firmware is really just a different kind of software - software that runs on the VideoCore processors (VPUs) and loads the OS, along with a number of other support roles such as the legacy 3D driver, ISP, display compositing etc.

Running a 4.9 kernel with a current firmware works just fine - I switch back and forth between 4.9 and 4.14 on an almost daily basis with changing the firmware.

Roland74, 2018-03-16:
Thank you Phil for your explanation. By the way is there an easier possibility to boot a different kernel than changing the /boot/config.txt? A Lilo or grup is probably not available for Raspberry?
Kind regards,
  Roland

pelwell, 2018-03-16:
Some people run U-boot on Pis. It's not something we support directly, but we are happy to support the maintainers.

pelwell, 2018-03-16:
NOOBS is another option...

Roland74, 2018-03-16:
Thank you very much. :)

Roland74, 2018-03-16:
Hi Phil,
thank you very much. It works very well :-)
After updating the firmware with rpi-update I can boot the latest kernel 4.14.
The sound kernel module is build and works fine. I just tested to watch youtube which works fine :)
Even vcgencmd works fine. Super! 
[dmesg_4.14.txt](https://github.com/raspberrypi/linux/files/1820988/dmesg_4.14.txt)

I add some output below:

Here some output
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.26-v8+ #1 SMP PREEMPT Fri Mar 16 00:20:28 CET 2018 aarch64 GNU/Linux
pi@raspberrypi:~ $ lscpu
Architecture:          aarch64
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
Model:                 4
CPU max MHz:           1200.0000
CPU min MHz:           600.0000
BogoMIPS:              38.40
Flags:                 fp asimd evtstrm crc32 cpuid
pi@raspberrypi:~ $ vcgencmd get_config arm_freq
arm_freq=1200
pi@raspberrypi:~ $ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
600000
pi@raspberrypi:~ $ vcgencmd measure_temp
temp=41.9'C
pi@raspberrypi:~ $ cat /sys/class/thermal/thermal_zone0/temp
41856
pi@raspberrypi:~ $ lsmod | sort
bluetooth             393216  28 hci_uart,bnep,rfcomm
bnep                   24576  2
brcmfmac              270336  0
brcmutil               16384  1 brcmfmac
cfg80211              696320  1 brcmfmac
cmac                   16384  1
ecdh_generic           24576  1 bluetooth
evdev                  24576  1
fuse                  114688  3
hci_uart               28672  1
hid_logitech_dj        24576  0
hid_logitech_hidpp     40960  0
i2c_dev                20480  0
ip_tables              28672  0
ipv6                  483328  40
joydev                 28672  0
Module                  Size  Used by
rfcomm                 53248  6
rfkill                 32768  6 bluetooth,cfg80211
sdhci_iproc            16384  0
snd                    90112  3 snd_timer,snd_bcm2835,snd_pcm
snd_bcm2835            40960  0
snd_pcm               126976  1 snd_bcm2835
snd_timer              36864  1 snd_pcm
uio                    24576  1 uio_pdrv_genirq
uio_pdrv_genirq        16384  0
vchiq                 348160  1 snd_bcm2835
x_tables               45056  1 ip_tables

Best regards,
  Roland

raymondjclark, 2018-03-18:
Up on 4.14 in 64 bit mode using old dtbs... Looks good

Roland74, 2018-03-24:
Hi Raymond,
that is strange. It only sees one core with your new kernel. Thank God it works for me with all 4 cores.
Maybe the /boot/config.txt is not correct (see below in this message)

Hello all Raspberry 64 Bit users :-)
I did some testing comparing the old 32 Bit 4.9 kernel with the new 64 Bit 4.14 kernel comparing watching youtube videos. The videos are shown smooth even in full screen mode. The audio is synchron on 64 Bit but it has clicking sound sound (a little bit cracking). At first I thought that full screen is too much but even with small resolution an very low load the cracking is there.
Running the 32 Bit Kernel 4.9 the sound was without cracking, the video was smooth too but somehow the audio and video was a bit asynchron. Maybe it was after I installed pulse audio? 
Before installing pulse audio and did a firmware update the 32 Bit mode played the videos symchron in sound and video.
Can anyone test the chromium browser watching video and compare the sound audio quality for 32 Bit and 64 Bit kernel?

For multi boot I use NOOBS and edit /boot/config.txt
The last 4 lines are:
# 64 Bit kernel
device_tree=dtbs/4.14.26-v8+/broadcom/bcm2710-rpi-3-b.dtb
kernel=kernel8.img
#device_tree=bcm2710-rpi-3-b.dtb
#kernel=kernel7.img
So I can comment in and out the two 64 Bit or 32 lines.

I have a question about the bcm2710-rpi-3-b.dtb line.
In
/boot/dtbs/4.14.26-v8+/broadcom
I have two files bcm2710-rpi-3-b.dtb and bcm2837-rpi-3-b.dtb
Which is the correct one to choose in /boot/config.txt? 
Currently I use  bcm2710-rpi-3-b.dtb and it boots but I am unsure which one is better? 

Now I will try to update the 32 Bit kernel with the firmware upgrade.
Kind regards,
  Roland

Roland74, 2018-03-24:
Hi,
the 32 Bit kernel was not updated with the firmware upgrade. 
rpi-update wrote:
You appear to be using a custom kernel file.
Skipping installation of new kernel, as bundled dtb files may be incompatible with your kernel.

I think it would not be conflicting because the kernels use different directories as the kerner version is a bit different. No problem for 32 Bit I can live with kernel 4.9.

For the noisy sound I found a work around. I grepped for the process
 /usr/bin/pulseaudio
and killed it. After starting chromium again a new  /usr/bin/pulseaudio was started and the noisy sound was gone and the sound was fine in youtube and everything played synchroniously. Perfect :-)
I like the 64 Bit kernel.
Best regards,
  Roland

JamesH65, 2018-04-23:
@Roland74 Can this issue now be closed?

raymondjclark, 2018-04-24:
I suppose, I have moved on to opensuse 64 bit.

Raymond Clark
Expect tYopS

closed 2018-04-24 by JamesH65

Roland74, 2018-04-24:
Hi James, it's OK to close the issue now. As I see it's already closed.
Kind regards,
  Roland

JamesH65, 2018-04-24:
Thanks for the update.

k9gardner, 2023-05-02:
I wish only that someone would have answered the original question, which I took to be "where is module snd-bcm2835.ko"? I inadvertently ran a curl script to install an Adafruit I2S audio bonnet on the wrong RPi (without said bonnet) and now trying to get my analog 3.5mm audio back, unsuccessfully. Trying to look under every rock to find something I've missed! I see that the module is being called to load in config.txt but I don't know if it is in fact loading. The audio port is simply not showing up as a sink option for example in the pulseaudio volume control. Can someone tell me where to look to find that that module is in fact there? I'm not up on the architecture at all.