raspberrypi/linux #2431: rpi-4.9.y: pps-gpio does not capture rising edge on RPI3

hdoverobinson, 2018-03-12:
With the following /boot/config.txt entry and a several known-good PPS signals on multiple Raspberry Pi 3's I cannot see any PPS input via ppstest on 4.9.59:
`dtoverlay=pps-gpio,gpiopin=18
`

I can see PPS asserts but no clears when using this /boot/config.txt entry:
`dtoverlay=pps-gpio,gpiopin=18,assert_falling_edge
`

As I understand it, this option reports falling edges as rising ones, or "asserts". 

With small modifications to pps-gpio.c and pps-gpio-overlay.dts I am able to see falling edges as "clears" through ppstest with the following /boot/config.txt entry:
`dtoverlay=pps-gpio,gpiopin=18,capture_clear
`

added at line 123 of pps-gpio.c:
`if (of_get_property(np, "capture-clear", NULL))
                        data->capture_clear = true;
`

added at line 34 of pps-gpio-overlay.dts:
`capture_clear = <&pps>,"capture-clear?";
`

None of these options let the user track the PPS rising edge. I am puzzled as to how to do this via pps-gpio with controls in the dtoverlay. This is important for timing applications as most PPS outputs will have their rising edge set to be the top-of-second marker by default. Any guidance here would be appreciated. Thank you

pelwell, 2018-03-12:
That's not my experience. With `dtoverlay=pps-gpio,gpiopin=18` and the following loop:
```bash
# The sleep is to keep the output volume manageable, but try without
while true; do grep . /sys/class/pps/pps0/{assert,clear}; sleep 1; done
```
I see plenty of asserts and no clears. In fact I get too many asserts with a flying leading I can touch to +3.3V or GND because of the lack of a pull on the pin (the overlay doesn't configure one, causing the pin to float if nothing is connected).

You can temporarily add a pull using raspi-gpio:
```
$ raspi-gpio set 18 ip pd
```
With that in place I see asserts when touching +3.3V (with some bounce) and nothing when touching GND.

I approve of your addition to enable capture-clear from DT using the overlay - you should submit that as a Pull Request (yes it's an upstream module, but the maintenance overhead should be non-existent and it's a useful facility).

pelwell, 2018-03-12:
P.S. With `dtoverlay=pps-gpio,gpiopin=18,assert_falling_edge` and GPIO 18 set to pull down, I see asserts when the flying lead is grounded, and nothing when driven high, as you would expect.

hdoverobinson, 2018-03-13:
Thank you, due to a physical connection issue I was not reliably seeing the rising edge of the PPS...

I just made pull request #2433 to add in the capture_clear option for the PPS falling edge.

JamesH65, 2018-05-11:
@hdoverobinson OK to close this?

JamesH65, 2018-06-29:
Closing this issue as questions answered/issue resolved.

closed 2018-06-29 by JamesH65