raspberrypi/linux #2435: Built in wifi connection to router fails hard on router reboot

picameratk, 2018-03-14:
Whenever I reboot my router the wifi connection on my RPi3 fails and will not recover without a reboot.  I have tried all of the usual ifup, ifdown, ifconfig down....   I am also on the latest kernel which you can see below. 

The Pi is being used as a repeater with wlan0 pointing to a usb wifi dongle and wlan1 using the built in wifi adapter of the Pi. I have a set of udev rules to make the built in wifi adapter wlan1 rather than the usual wlan0.

This Pi setup has been running flawlessly since September 2017 and has only experienced this problem in March 2017.  I have periodically run rpi-update, apt-get update, and apt-get dist-upgrade.  Also note that the usb wifi dongle does not fail at all.  The usb wifi dongle is being controlled by hostapd and all of its device remained connected when wlan0 fails.

Environment:   
RPi3 Raspbian Stretch   
Linux motionpi3 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l GNU/Linux  
firmware-brcm80211                    1:20161130-3+rpt2  

/etc/network/interfaces  
auto wlan1  
iface wlan1 inet static  
address 192.168.135.92  
netmask 255.255.255.0  
gateway 192.168.135.1  
 wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf  
post-up iw dev wlan1 set power_save off  
post-up iwconfig wlan1 txpower 20  

Relevant syslog:
[syslog.txt](https://github.com/raspberrypi/linux/files/1812062/syslog.txt)

pelwell, 2018-03-14:
March 2018?

picameratk, 2018-03-14:
Yes, March 2018

picameratk, 2018-03-14:
Oops, did not spot the date typo.  THanks.

JamesH65, 2018-03-14:
This is a known issue and fixed in the latest trees. I *think* that apt
update/apt upgrade on Stretch will fix it.

Failing that you could update the kernel to 4.14 using rpi-update.

picameratk, 2018-03-14:
Actually I am already running 4.14.24 so that is not an alternative solution.  I will now go run apt-get dist-upgrade.

picameratk, 2018-03-14:
I keep the Pi fairly current so at this time apt-get dist-upgrade has no packages to upgrade.  This Pi is really current.

JamesH65, 2018-03-14:
Hmm, I wonder if you have the older wireless firmware. Might make a difference.

What does a `dmesg | grep brcmfmac` produce?

picameratk, 2018-03-14:
[   22.980275] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   22.987706] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[   22.988024] usbcore: registered new interface driver brcmfmac
[   23.166967] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[   23.167811] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:47 
[   23.236773] brcmfmac mmc1:0001:1 wlan1: renamed from wlan0
[   25.355567] brcmfmac: power management disabled
[   25.594742] brcmfmac: power management disabled

picameratk, 2018-03-17:
I can no longer recreate this problem after upgrading.
**uname -r = 4.14.27-v7+**
output of dmesg:
[   23.374187] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   23.382602] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[   23.382838] usbcore: registered new interface driver brcmfmac
[   23.604963] brcmfmac: brcmf_c_preinit_dcmds: **Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f**
[   23.605862] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14
[   23.673947] brcmfmac mmc1:0001:1 wlan1: renamed from wlan0
[   25.644756] brcmfmac: power management disabled
[   25.858948] brcmfmac: power management disabled