raspberrypi/linux #2436: rpi-4.14: default console changed for Pi 1

maxnet, 2018-03-15:
In 4.14 there seems to be a change of behavior in what is the default console on Pi 0/1 boards.

Have a script that outputs to /dev/console
On Pi 2/3 output still appears on tty1, like was the case before in 4.9

[image]

But on PI 0/1 it no longer is the case:

[image]

Seems output now goes to serial console by default.

I can workaround it by specifying `console=tty1` explicity in cmdline.txt
However I still think changes in behavior like this should be avoided.

maxnet, 2018-03-15:
Likely caused by https://github.com/raspberrypi/linux/commit/f08f58a2bf68900a84e782b8c7ad701c0654173c that sets stdout-path = "serial0:115200n8";

Think that setting doesn't have effect on Pi 3 because serial0 is an alias for serial@7e215040 which has status = "disabled";.
But does causes this change of behavior on Pi 1.

Device tree I get when running on a Pi 3 (obtained from /proc):

```
/dts-v1/;

/memreserve/	0x0000000000000000 0x0000000000001000;
/ {
	memreserve = <0x36000000 0xa000000>;
	serial-number = "0000000062429dd6";
	compatible = "raspberrypi,3-model-b-plus", "brcm,bcm2837";
	model = "Raspberry Pi 3 Model B Plus Rev 1.3";
	interrupt-parent = <0x1>;
	#address-cells = <0x1>;
	#size-cells = <0x1>;

	system {
		linux,serial = <0x0 0x62429dd6>;
		linux,revision = <0xa020d3>;
	};

	axi {

		vc_mem {
			reg = <0x3ec00000 0x40000000 0xc0000000>;
		};
	};

	aliases {
		i2c_arm = "/soc/i2c@7e804000";
		i2c = "/soc/i2c@7e804000";
		i2c_vc = "/soc/i2c@7e205000";
		serial0 = "/soc/serial@7e215040";
		serial1 = "/soc/serial@7e201000";
		audio = "/soc/audio";
		aux = "/soc/aux@0x7e215000";
		sound = "/soc/sound";
		soc = "/soc";
		dma = "/soc/dma@7e007000";
		intc = "/soc/interrupt-controller@7e00b200";
		watchdog = "/soc/watchdog@7e100000";
		random = "/soc/rng@7e104000";
		mailbox = "/soc/mailbox@7e00b880";
		gpio = "/soc/gpio@7e200000";
		uart0 = "/soc/serial@7e201000";
		sdhost = "/soc/mmc@7e202000";
		mmc0 = "/soc/mmc@7e202000";
		i2s = "/soc/i2s@7e203000";
		spi0 = "/soc/spi@7e204000";
		i2c0 = "/soc/i2c@7e205000";
		uart1 = "/soc/serial@7e215040";
		spi1 = "/soc/spi@7e215080";
		spi2 = "/soc/spi@7e2150c0";
		mmc = "/soc/mmc@7e300000";
		mmc1 = "/soc/mmc@7e300000";
		i2c1 = "/soc/i2c@7e804000";
		i2c2 = "/soc/i2c@7e805000";
		usb = "/soc/usb@7e980000";
		leds = "/leds";
		fb = "/soc/fb";
		vchiq = "/soc/vchiq";
		thermal = "/soc/thermal@7e212000";
		axiperf = "/soc/axiperf";
		ethernet0 = "/soc/usb@7e980000/usb1@1/usb1_1@1/usbether@1";
	};

[... 2241 lines cut ...]
		vc4 = "/soc/gpu";
		dpi = "/soc/dpi@7e208000";
		mmc = "/soc/mmc@7e300000";
		firmwarekms = "/soc/firmwarekms@7e600000";
		smi = "/soc/smi@7e600000";
		axiperf = "/soc/axiperf";
		firmware = "/soc/firmware";
		power = "/soc/power";
		fb = "/soc/fb";
		vchiq = "/soc/vchiq";
		vcsm = "/soc/vcsm";
		audio = "/soc/audio";
		sound = "/soc/sound";
		clk_osc = "/clocks/clock@3";
		clk_usb = "/clocks/clock@4";
		usbphy = "/phy";
		vdd_5v0_reg = "/fixedregulator_5v0";
		vdd_3v3_reg = "/fixedregulator_3v3";
		leds = "/leds";
		act_led = "/leds/act";
		pwr_led = "/leds/pwr";
	};
};

```

pelwell, 2018-03-15:
This is an upstream change that we've merged along with all the others. If it is considered harmful then we can revert it or find a workaround (which shouldn't be difficult). But first I'd like to understand your configuration a bit more, in particular how your console output ends up on tty1 - for me, having removed the console settings in cmdline.txt (which are present in Raspbian images), the output goes to tty0.

What does `dmesg | grep console` report?

pelwell, 2018-03-15:
It may save time if you post the entire output of dmesg after booting.

maxnet, 2018-03-15:
You are right, its tty0 not tty1
Just meant to say the tty that has HDMI, as opposed to the tty that has serial.

Pi 3:

```
console [tty0] enabled
```

Pi 1, since 4.14:

```
console [tty0] enabled
console [ttyAMA0] enabled
```

In which the last one becomes tied to /dev/console on Pi 1
If you do "echo test > /dev/console" it does appear on HDMI on Pi 3 but not anymore on Pi 1.

(If it's of any use:
A build that exhibit this problem: https://sourceforge.net/projects/berryboot/files/berryboot-20180314-pi0-pi1-pi2-pi3.zip
As long as Berryboot did not finish installation, console is available on ctrl-alt-f2, login: root, no password
After that, it is under "edit menu" -> "more options" -> "console"
)

pelwell, 2018-03-16:
I've pushed a commit to rpi-4.14.y that deletes the 'stdout-path' property, restoring the previous behaviour.

Thanks for reporting this - it's not something we would have noticed.

MilhouseVH, 2018-03-16:
Thanks, this fixes the console issue LibreELEC has had on Pi0/Pi1/Pi2 since 4.14, which caused LibreELEC to revert https://github.com/raspberrypi/linux/commit/f08f58a2bf68900a84e782b8c7ad701c0654173c as a workaround (which we will now stop reverting).

pelwell, 2018-03-16:
You're welcome. Reporting issues like that here helps them to not get lost.

closed 2018-03-18 by maxnet