raspberrypi/linux #2441: DS3231 on 400 khz baudrate I2C?

deblauwetom, 2018-03-16:
Hello,

I got the DS3231 RTC chip of maxim working by using the following overlay:

`
dtoverlay=i2c-rtc,ds3231
`

However, it doesn't work when I increase the I2C baudrate to 400000 like this:

`
dtparam=i2c_baudrate=400000
`

The driver loaded in linux is the "rtc-ds1307". 

When I remove the i2c-rtc overlay from the config.txt, then I can talk to the device using i2cdetect and i2cdump so it really works at 400 khz. 

I don't want to lower the baudrate to 100000, so apart from reading the device myself using just a userspace program, what are the options I could try? It is like the rtc-ds1307 driver only wants to talk to the device on 100 khz. This makes sense for the ds1307 device itself because I saw in specs this only supports 100 khz, but the ds3231 device also supports 400 khz. Is there maybe a better suited driver for the DS3231 device? Something else?

I am using kernel 4.4.50, with the tag "raspberrypi-kernel_1.20170405-1". The hardware is a CM3, on a waveshare io baseboard plus, which has the DS3231 attached to pin 2 & 3.

pelwell, 2018-03-16:
Two questions:
1. What's the highest frequency you've got it running at?
2. Do you have to use 4.4? We are on 4.9 now, about to switch to 4.14, and we're not going to be able to provide 4.4-specific support.

deblauwetom, 2018-03-16:
Hello,

1. I only tried 100000(so just the default)
2. I would rather not upgrade now, but am willing to backport/check any fixes you would have for newer kernels. I upgraded to 4.9 at one time but had some trouble I can't remember, so I stayed at 4.4 :)

deblauwetom, 2018-03-19:
Hi, it seems I ran into the issue that baudrate 400 kHz actually requires the value to be 640000 instead of 400000. So now it works. Maybe this is fixed in later kernels and this issue can be closed?

pelwell, 2018-03-19:
I2C clock rate is slaved off the core clock, so when the VPUs enter and leave turbo mode the I2C speed changes. If you really need a specific clock speed (which seems odd for I2C) you can add `core_freq=250` to config.txt and reboot.

closed 2018-03-24 by deblauwetom