raspberrypi/linux #2442: Rpi3B+ kernel oops phy_ethtool_ksettings_get

penfold42, 2018-03-17:
A nod was starting early in the boot process and triggered an oops just like:

https://patchwork.kernel.org/patch/10282365/

Apologies if this patch is already on the list but I couldn’t find any recent commits that include it

graysky2, 2018-03-27:
Just FYI - I applied the patch building 4.14.29, upon booting into it, my network was unable to come up. 
 Any ideas why?

I see nothing in dmesg or journalctl with errors.  The status of the service:
```
● systemd-networkd.service - Network Service
   Loaded: loaded (/usr/lib/systemd/system/systemd-networkd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-03-23 15:02:48 EDT; 20s ago
     Docs: man:systemd-networkd.service(8)
 Main PID: 441 (systemd-network)
   Status: "Processing requests..."
    Tasks: 1 (limit: 1831)
   CGroup: /system.slice/systemd-networkd.service
           └─441 /usr/lib/systemd/systemd-networkd

Mar 23 15:02:48 pi3 systemd[1]: Starting Network Service...
Mar 23 15:02:48 pi3 systemd-networkd[441]: Enumeration completed
Mar 23 15:02:48 pi3 systemd[1]: Started Network Service.
Mar 23 15:02:48 pi3 systemd-networkd[441]: lo: Link is not managed by us
```
I am building 4.14.30 now with the patched source to test as well.

pelwell, 2018-03-27:
There may be an interaction with the other patches we've applied.

graysky2, 2018-03-27:
> There may be an interaction with the other patches we've applied.

Confirmed with 4.14.30 as well.  At least it allows for a boot without kernel oops with a bridge network... you just can't bring the network up :D

penfold42, 2018-03-27:
Sounds like what I saw with OSMC:
https://github.com/osmc/osmc/commit/21eb49b83ef36f2a27a083f4320609a54242dbfc

https://discourse.osmc.tv/t/rpi3b-kernel-oops-phy-ethtool-ksettings-get/71548

graysky2, 2018-03-28:
@pelwell - Do you think either of the recent commits will help this issue?

https://github.com/raspberrypi/linux/commit/9696aab22bb8163fb3a2a262dd6c67f6d05b70a1
https://github.com/raspberrypi/linux/commit/661230038a8736ba9023978c53cd21cd65739406

popcornmix, 2018-04-04:
Latest rpi-update kernel has a potential fix for this issue. Please test.

graysky2, 2018-04-04:
@popcornmix - Arch ARM doesn't use that script but I am building:
* raspberrypi-firmware from https://github.com/raspberrypi/firmware/commit/0dff9ecc00d8114164bb9a73799793dad24d8593
* linux-raspberrypi from https://github.com/raspberrypi/linux/commit/b5b6bb9bdb5e62bc7dfc64b40bd833bffd971693

Will try and report back, thank you!

graysky2, 2018-04-04:
> Will try and report back, thank you!

@popcornmix - I am now able to successfully boot with the bridge setup I described in https://github.com/systemd/systemd/issues/8503.

The network there isn't completely stable though... using `dd` to write out to an NFS mount seems flakey:

```
% dd if=/dev/zero of=/opt/armv7/scratch/fill bs=4M count=250 status=progress
1006632960 bytes (1.0 GB, 960 MiB) copied, 21 s, 47.9 MB/s
250+0 records in
250+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 25.4407 s, 41.2 MB/s

% dd if=/dev/zero of=/opt/armv7/scratch/fill bs=4M count=250 status=progress
1048576000 bytes (1.0 GB, 1000 MiB) copied, 136 s, 7.7 MB/s
250+0 records in
250+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 138.733 s, 7.6 MB/s

% dd if=/dev/zero of=/opt/armv7/scratch/fill bs=4M count=250 status=progress
1015021568 bytes (1.0 GB, 968 MiB) copied, 22 s, 45.7 MB/s
250+0 records in
250+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 25.5034 s, 41.1 MB/s
```

I am booting with a simple dhcp/eth0 (no bridge) to see if I can confirm this.  Progress though, it does boot now so good job!

closed 2018-04-30 by pelwell