raspberrypi/linux #2443: RPi3 Can't connect a USB wifi adapter after the March 2018 firmware updates

rsfitz, 2018-03-17:
**Description:**
The USB WiFi adaptor will connect and immediately disconnect. Attempts to manually reconnect will prompt for passphrase and fail to connect.

**Steps to Reproduce:**
- Install compatible USB WiFi Adapter
- Disable on-board WiFi (dtoverlay=pi3-disable-wifi)
- apt-get update && upgrade

JamesH65, 2018-03-17:
Which USB adapter are you using? dmesh to get the vendor ID would be useful, and try lsub to get its name.

rsfitz, 2018-03-18:
It is _generic_ Realtek Adapter
idProduct=7811

This appears to be an issue with re-connection to the AP: If the connection to the access point is lost it will not automatically or manually reconnect. Removing and replacing the USB adapter will restore the connection.

JamesH65, 2018-03-18:
We had a recent problem with the onboard Wireless doing the same thing. It was a Linux driver problem. The firmware on the wireless device was fine, but it was new firmware. Older firmware worked with the older driver.

Using the same logic  this is not a Pi firmware issue, but a Realtek kernel driver issue or Realtek firmware issue. You will need to approach the Realtek maintainer for help on this one, usually done via the Linux wireless mailing list.

rsfitz, 2018-03-18:
Will do, I did find a similar issue dating back to 2015 with the kernel driver that appears to have the same symptoms.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=776816

jakemagee, 2018-03-20:
@rsfitz, Sounds like this issue can be closed here?

closed 2018-03-20 by rsfitz