raspberrypi/linux #2444: Loop Message Kernerl :raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001

Cyr-ius, 2018-03-17:
I compiled the last kernel 4.14.27 and since then I have the following message in loop in the kern.log
raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
and i have no problem of under voltage
Pi Zero W
vcgencmd measure_clock arm
frequency(45)=1000106000
vcgencmd measure_clock core
frequency(1)=400000000
vcgencmd measure_volts
volt=1.2000V

vcgencmd measure_temp; vcgencmd get_throttled; vcgencmd measure_temp
temp=37.9'C
throttled=0x0
temp=37.9'C

I have the problem on my Pi Zero and Pi 2

How to stop this message ?

pelwell, 2018-03-17:
What on earth would possess you to write the same question three times in three different issues? All the issues go the same people, so you just annoy them.

Answered elsewhere.

closed 2018-03-17 by pelwell

marcusinacio, 2018-04-11:
I have a RPi 2 running Arch, think I have latest firmware but still seeing the message repeat in syslog.
Can you help?

Cheers, Marcus

pelwell, 2018-04-11:
You can't have the latest firmware - what does this command return:
```
pi@raspberry:~ $ strings /boot/start.elf | grep VC_BUILD
```

marcusinacio, 2018-04-11:
Seems you're right... however pacman thinks "raspberrypi-firmware-20180330-1-any.pkg.tar.xz" is installed?
Output:
`VC_BUILD_ID_USER: dc4
VC_BUILD_ID_TIME: 20:15:34
VC_BUILD_ID_BRANCH: master
VC_BUILD_ID_TIME: Jul 28 2017
VC_BUILD_ID_HOSTNAME: dc4-XPS13-9333
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 6716e816fd1614c326071d86188976b5fbf9c22b (clean)`

TomJohnZ, 2018-04-22:
The package raspberrypi-firmware doesn't update /boot. You're most likely still using the same bootloader and firmware from your original install. To update these files, you can install the raspberrypi-bootloader (or raspberrypi-bootloader-x) package. I'd recommend backing up your /boot directory first, just to be safe.

AGenchev, 2018-06-17:
Running rpi-update as root fixes you. Internet must be connected of course.