raspberrypi/linux #2445: Pi Configuration Locale Missing

ptdesign, 2018-03-17:
Set Locale Language pt (Portugal Portuguese) is missing since last versions of raspberry pi.

pelwell, 2018-03-23:
While looking at a similar issue we've spotted a problem with sub-languages (like Austrian German and Braziliian Portuguese). The app was displaying the name of the first (alphabetically) language in the list with the same major language - so de -> de_at, pt -> pt_br. However, the problem is confined to the displayed name of the locale - the correct locale is actually selected.

The fixed app requires some testing and packaging, but it should be available in a week or so.

pelwell, 2018-03-23:
I'm assured that is also fixed.

ptdesign, 2018-04-25:
New release from 2018-04-18 doesn't provide info about this fix.

pelwell, 2018-04-25:
This is the specific GUI commit that fixed it: https://github.com/raspberrypi-ui/rc_gui/commit/3e6fc4a27e6cf7c0655140d6e239055e27325857

ptdesign, 2018-04-25:
Ok, but is included in last release? Thanks

JamesH65, 2018-06-26:
@ptdesign OK to close this?

ptdesign, 2018-06-26:
Ok, thanks for heads up!

closed 2018-06-26 by ptdesign