raspberrypi/linux #2446: OOPS in latest kernel

dave-p, 2018-03-17:
[oops.txt](https://github.com/raspberrypi/linux/files/1821950/oops.txt)

This is on an original Pi2B running fully up-to-date Arch Linux. The application is TVHeadend which was compiled from source on this machine and which uses a DVB-S2 USB stick. Networking is via Ethernet.

The OOPS is triggered when I stream live TV from TVHeadend to VLC (v3.0) running on a remote Linux machine, then close the session by exiting VLC. TVHeadend becomes unresponsive and cannot be killed.

Reverting the kernel and firmware to the previous version cures the problem.

The same behaviour is seen if the SD card is moved to a Pi3B.

dave-p, 2018-03-18:
To clarify, the last good kernel (that I can test without building my own) is 4.14.24, commit 80a14a5. The oldest failing kernel is 4.14.26, commit ad350a5.
I know nothing of kernel internals but this commit, introduced in 4.14.25, looks possibly relevant: https://github.com/raspberrypi/linux/commit/abb4a8b870b5c644c9bb6d9bd372298002f0a65f

popcornmix, 2018-03-18:
Are you able to build a kernel with that commit reverted?
https://www.raspberrypi.org/documentation/linux/kernel/building.md

dave-p, 2018-03-18:
Working on it now, but I may be some time...

lategoodbye, 2018-03-18:
According to netdev this [patch](https://www.spinics.net/lists/netdev/msg489445.html) should be the fix.

dave-p, 2018-03-18:
I built a kernel from the head of the rpi-4.14 branch using the Arch packaging system; as expected the Oops occurs.
Reverting commit https://github.com/raspberrypi/linux/commit/abb4a8b870b5c644c9bb6d9bd372298002f0a65f and rebuilding, the issue is gone.
Reinstating https://github.com/raspberrypi/linux/commit/abb4a8b870b5c644c9bb6d9bd372298002f0a65f and also applying the patch from https://www.spinics.net/lists/netdev/msg489445.html, the issue is still gone.

popcornmix, 2018-03-18:
Okay, looks like an upstream bug with a pending upstream fix.
Once it is merged we can pull it in here (hopefully it will appear in 4.14.28 or shortly afterwards) but if not we can pull it in earlier.

dave-p, 2018-03-18:
Thanks.
@lategoodbye, many thanks for spotting the solution. I'll carry on using my homebrewed kernel until the fix comes through the system.

closed 2018-03-18 by dave-p

popcornmix, 2018-03-28:
The linked patch has been pushed to linux tree and is in latest rpi-update kernel