raspberrypi/linux #2447: Pi3b+: "kevent 4 may have been dropped" with lan78xx

jens-maus, 2018-03-17:
Every second or third boot with a 4.14.26 kernel where the LAN78XX support had been enabled I receive the following notices in `dmesg`:

```
...
[   16.345186] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.346150] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.347155] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.348168] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.349165] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.350173] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.351150] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.352149] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.353164] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.354165] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.355150] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
...
```

Sometimes the system still continues and is able to setup eth0 properly. However, every now and then the system is not able to bring up eth0 and thus the boot up process is interrupted.

Any idea where this might come from or how to debug this situation?

JamesH65, 2018-03-17:
Bugger, thought the kevent issue was ancient history. IIRC, it is because an event is stuck on the system event queue and then another one turns up before it is serviced. But it's not been reported for a couple of years, and that was on the smsc75xx driver. Not sure this is the same thing though, might be a systemd initialisation issue.

jens-maus, 2018-03-17:
The system I am developing and compiling the raspberrypi kernel for is not using systemd at all but is based on buildroot without systemd.

JamesH65, 2018-03-18:
OK, do you get the same problem with Raspbian?

jens-maus, 2018-03-18:
Haven‘t tested it yet with Raspbian. However, this is the kernel defconfig I am using:

https://github.com/jens-maus/RaspberryMatic/blob/master/buildroot-external/board/raspberrypi3/kernel_defconfig

Bilg21, 2018-03-22:
I have the same issue with Compute Module 3. But it looks like a hardware issue. It's not happening on another setup which is identical.

danijelt, 2018-04-13:
I am also using Buildroot to make my own image, and I managed to resolve this issue by disabling VLAN support in kernel.
In my case, I do have systemd, and this message appears after systemd starts hostname service:
```
[  OK  ] Started Hostname Service.
[   17.326467] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   17.326507] 8021q: adding VLAN 0 to HW filter on device eth0
[   17.338617] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   17.345081] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
```
After disabling VLAN support (`CONFIG_BRIDGE_VLAN_FILTERING`, `CONFIG_VLAN_8021Q` and `CONFIG_VLAN_8021Q_GVRP`) in latest rpi-4.14-y (4.14.33) this message stopped appearing almost completely, and I don't have the lockup bug anymore.

Firmware is ae9a493932e47e08cabb25a2728037298075fd00 and userland is a343dcad1dae4e93f4bfb99496697e207f91027e.

JamesH65, 2018-04-23:
@danijelt @jens-maus We have committed three or four fixes to VLAN support in the lan78xx driver since 4.14.33, so its probably worth getting the latest kernel and seeing if it has fixed the kevent issue with VLAN enabled. There are also other lan78xx fixes in there, so well worth trying to see if the kevent issues has gone.

If you do find it has gone, please close this issue.

JamesH65, 2018-05-11:
@danijelt @jens-maus If no further reports/updates are forthcoming, I'm inclined to close this report.

danijelt, 2018-05-11:
We have released a stable production image without VLAN and I'm currently occupied with another project.

I'll keep this in mind during the next update and report here if kevent/VLAN issue persists.

fhunleth, 2018-06-02:
I was able to easily reproduce this issue with the VLAN options disabled when running Linux 4.14.29 at commit c117a8bccf37bfba323065b566cf999ed4629a4a. I just tried updating to the latest rpi-4.14-y (4.14.44) at commit 4fca48b7612da3ff5737e27da15b0964bdf4928f, and I can no longer reproduce the `kevent 4` issue.

bcutter, 2018-06-06:
I also saw this some times on my Pi 3 B+. Currently the Pi is completely dead, on console screen the

`"** 1 printk messages dropped ** [ 3808.632569] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped"`

messages go wild meaning every second about 50 of those lines are shown. Only one option: hard reset of the system. Resulting in fsck errors and so on.

Raspbian Stretch by the way. 1 Gbit/s ethernet port.
**Anything I can do on this? Will there be a fix? A workaround?**

JamesH65, 2018-06-06:
@bcutter Are you using the latest kernel? You can apt update, to get an official release, or rpi-update to get the latest bleeding edge one. You might need rpi-update, but backup first. There have been a number of kernel updates so worth trying.

bcutter, 2018-06-06:
Currently on 4.14.34-v7+. Latest official release. I try to avoid using rpi-update, cause of the risk of running into other issues (maybe fix one and maybe grab some others - no perfect chance/risk ratio :-))... maybe my opinion will change temporarily if this kevent4 keeps on appearing.

vnd, 2018-06-14:
I'm also experiencing this issue occasionally, 4.14.48-v7+.

bcutter, 2018-06-14:
Updated to 4.14.48-v7+ recently, will monitor future behaviour.
But @vnd´s comment doesn´t make me feel positive about this...

JamesH65, 2018-06-14:
Please note, the 3B error reporting kevent 0 is, I think,  a different issue - please keep this thread to 3B.

For those people using 3B+, do you still see the issue after a rpi-update? Report above seems to indicate it has gone away with the latest, but I would like to confirm that.

JamesH65, 2018-06-14:
Also, can I just confirm this only happens during startup? Or have people encountered this error in general usage?

bcutter, 2018-06-14:
I always experienced this during normal usage a few hours or even days after startup.

fhunleth, 2018-06-14:
I'm no longer able to reproduce it, but when I could, it was by running `ip link set dev eth0 down` and `ip link set dev eth0 up` repeatedly.

JamesH65, 2018-06-14:
Just some notes from a a quick look at the driver code.  kevent 4 is EVENT_LINK_RESET. This is only fired off in two places, first during opening the driver, and secondly when an interrupt is received (via USB, so not a 'direct line') from the PHY. I believe the only time an interrupt is received requires the link to be reset so this makes sense. 

The action taken when the kevent is processed is fairly minimal, most work done is in lan78xx_link_reset, which does a bunch of phy/register reads and writes. Nothing I've seen that would take any real amounts of time.  Note though that register reads all go via USB, so they could be delayed if the other end is busy.

JamesH65, 2018-06-14:
Been giving the latest kernel some stress testing on a debug enabled lan78xx driver. Running stress-ng --cpu 4, so loading the CPU's right up, then making the ethernet go up and down as quickly as possible (ip link set dev eth0 down/up), whilst monitoring for any kevents. No errors at all after well over an hour of this. Will leave it running for a few more hours.

JamesH65, 2018-06-15:
Cannot get this to go wrong. Run test scripts for hours, no issues. Clearly not exercising the right runes. If anyone seeing this has anything unusual in their networking setup, please let me know here. But for the moment I have to look at some other stuff.

bcutter, 2018-06-15:
The only two „special“ things I have are:

1) Second virtual interface, so:
eth0
eth0:0
where the second one has a different IP configuration.
2) Various HDDs connected directly or with a hub via USB to the Pi - which is quite different to many other setups e.g. of friends of mine I can compare to (and they don’t see this issue at all)

Hope this helps a bit.

danijelt, 2018-06-20:
I have tried with latest 4.14.50 kernel (commit 3b01f05):
```
[    0.000000] Linux version 4.14.50 (danijel@dev) (gcc version 5.4.0 (Buildroot 2017.05-git-gd5acf12b1)) #1 SMP Wed Jun 20 10:59:49 CEST 2018
```

Still having the same problem when I enable VLAN support:
```
[  OK  ] Started Network Manager Script Dispatcher Service.
         Starting Hostname Service...
[  OK  ] Started Hostname Service.
[   17.853152] 8021q: adding VLAN 0 to HW filter on device eth0
[   17.853178] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   17.853924] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
```

My setup: Buildroot 2017.05, systemd 232, NetworkManager 1.8.0. kevent error appears appears after systemd-hostnamed service is started and kernel notifies that VLAN 0 is added.

JamesH65, 2018-06-20:
@danijelt Does this only occur during startup? Does it ever cause any problems or does everything still work OK?

6by9, 2018-06-20:
Configuring the VLAN mask and setting up multicast subscriptions both use worker threads 
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L1036
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L2325

VLAN takes data that is precomputed, and multicast works out the settings in the worker, so being requested twice before having been called for the first time should have no side effect (minor assumption that the flag is cleared before the worker task is executed).
They're independent work queues, so there shouldn't be a conflict between the two. Unless it's indicative of a worker thread having wedged totally, and that is then preventing any other worker thread from running. I don't know enough about how ```INIT_WORK``` works when it comes to scheduling.

danijelt, 2018-06-21:
@JamesH65 Yes, only during startup. Once it boots (if kevent 4 doesn't happen), I don't have any problems anymore.

I don't do anything special with the network, just bring it up at boot and leave it that way. I'm not even touching anything VLAN related, adding VLAN 0 is probably NetworkManager's (or kernel's) default behaviour.

JamesH65, 2018-06-26:
I suspect that the kevent reports on startup are benign, but if people are getting them after the startup process is complete, that's a bit more worrisome.

vnd, 2018-06-29:
Update: Ethernet port died (mii-tool eth0 => eth0: no link, "link ok" through USB<->Ethernet adapter, but ping and everything else doesn't work), so may be it was an early manifestation of hardware problem.

yarmooh, 2018-07-16:
Hello there,
I'm using ComputeModule3 with Lan78xx.
Today I've upgraded kernel to latest - 4.14.54-v7+ and `kevent 4 may have been dropped` is gone.

yarmooh, 2018-07-16:
I'm using ComputeModule 3, with kernel 4.14.54-v7+, and I'm getting that error too:

root@raspberrypi:~# dmesg | grep eth0
[    6.275493] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.275541] lan78xx 1-1.6:1.0 eth0: kevent 4 may have been dropped
[    6.303397] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

However, have no issues with LAN78xx at all, it works, just everytime showing that message.

bcutter, 2018-07-17:
Today I discovered my Pi "crazy dead" again, to be more specific: exactly same console output as shown in https://github.com/raspberrypi/linux/issues/2447#issuecomment-400570396.

I really don´t know what else to do or try on this! I even rpi-update´d a few weeks ago. Something I can check or provide some logs or anything like that? I don´t see any progress (in my case) :-(

pelwell, 2018-07-17:
You need to rpi-update again - a fix went in last week for a network driver bug that seems to have resolved many of the crashes and lockups people have been experiencing.

bcutter, 2018-07-17:
Thx for that information. I had the release feed of the GitHub project in my RSS reader but accidentally removed it meanwhile so I had no idea there´s a newer version.

According to "uname -a" the Pi 3 B+ is now on
`Linux RPi1Raspbian 4.14.54-v7+ #1126 SMP Wed Jul 11 20:01:03 BST 2018 armv7l GNU/Linux`

Will monitor it closely the next days and weeks.

bcutter, 2018-07-17:
Thx for that information. I had the release feed of the GitHub project in my RSS reader but accidentally removed it meanwhile so I had no idea there´s a newer version.

According to "uname -a" the Pi 3 B+ is now on
`Linux raspberrypi 4.14.54-v7+ #1126 SMP Wed Jul 11 20:01:03 BST 2018 armv7l GNU/Linux`

Will monitor it closely the next days and weeks.

bcutter, 2018-07-18:
Great. Crashed again, but this time with a different response:

"kevent 0 may have been dropped" at the end of the line, repeated every few seconds. System unresponsive but allowed a controlled reboot with STRG + ALT + ENTF.

So i´m lost again...

JamesH65, 2018-07-18:
@bcutter Going to need to know what you are doing on the Pi to provoke the error, how you have it set up, what going on at the time of error etc. The fixes that have gone in seem to have fixed a lot of lan78 errors, but this one is a new one, so presumably something to do wtih your setup.

bcutter, 2018-07-18:
I think so too. At the time of error the following happened:
- streamed a file using SMB from one client

I can´t provoke the error. Changes in configuration next to the rpi-update from yesterday: removed and disabled swap file, today a message "RAM is low, I´m going to kill php", but that was hours before the kevent 0 thing.

My "special" setup: https://github.com/raspberrypi/linux/issues/2447#issuecomment-397648281

JamesH65, 2018-07-19:
Had a quick glance through the code. Kevent 0 appears to be a tx stall, cause by the USB system returning an EPIPE error. This seems to be caused by a failure in the USB system itself, a protocol stall (drivers/usb/core/hcd.c!700 seems a likely culprit).  There are a number of ways in that function that coudl cause the stall. All of these completely out of my realm of 'expertise'. We woudl need a reliable way of replicating this in order to get to the bottom of it.

rstanislav, 2018-09-04:
Any updates on this ? This affects every 3b+ i have and makes them close to useless.. about every 1/3 or 1/4 cold boots this bug occurs..

JamesH65, 2018-09-05:
Odd, you appears to be one of the few experiencing this error. I pressume you are running the latest Raspbian. Is there anythingunusual about anythingplugged in to the Pi, or your network setup? Do you have another Pi3B+you can test on,  to see if it might be a HW problem?

rstanislav, 2018-09-05:
I have tested this on ~4 rpi 3b+, all works the same, OS is not raspbian, i'm using this project(and discussion of this problem): https://github.com/Ysurac/openmptcprouter/issues/128#issuecomment-418644313

But that project using current version of rpi kernel with all patches in this thread applied

JamesH65, 2018-09-05:
Would you be able to test with Raspbian? Since most people do not appear to see this issue, it woudl be useful to know if its OS specific.

akaher, 2018-09-05:
Getting this issue and eth0 is not working, using 4.14.54 Linux kernel:

[   10.101247] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
[   10.124946] libphy: lan78xx-mdiobus: probed
[   10.204327] usbcore: registered new interface driver lan78xx
[   10.388623] lan78xx 1-1.1.1:1.0 eth0: No External EEPROM. Setting MAC Speed
[   10.390004] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped

rstanislav, 2018-09-05:
I don't think its OS specific because openmptcp using 4.14 kernel and has all patches related to lan78xx (https://github.com/Ysurac/openmptcprouter/tree/master/root/target/linux/brcm2708/patches-4.14) also this happens when system if not fully loaded and only if LAN interface connected, without LAN cable it boots without any problems every time.

JamesH65, 2018-09-05:
It would still be worth testing with raspbian since it might be some userspace setup. Also, you may be taxing the USB system quite heavily. Does everything work without the USB devices plugged in?

jens-maus, 2018-09-05:
As I am the originator of this ticket, I can say that since I updated to recent kernel+firmware combinations (using 4.14.62 lately) I haven't seen these `kevent` outputs anymore and everything seems to work flawlessly in my environments (buildroot-based Linux environment).

rstanislav, 2018-09-05:
Strange thing is that this happens only at boot, if system loaded it works without any problem for hours, sometimes it can say something about kevent 4 - but like 1 or 2 messages during work (installed in car, so average work time is 2-5 hours between full reboots)  so with very high load on usb subsystem (4 modems with mptcp pushing 90-92 mb/sec thru ethernet connected wifi router (using it only for wifi))(wi fi router connected via 100mb link (4 wires)) - i never saw this problem during work, only at boot time.

pelwell, 2018-09-06:
These messages, although annoying, should be harmless most of the time - they just indicate that an operation (which ought to be idempotent) is being requested for a second time before the system has managed to handle it the first time. If you are seeing a stream of them and the interface isn't working then the message is likely to be a symptom of a wider problem rather than the cause.

rstanislav, 2018-09-06:
Yes, kevent 4 always appears at start, but only few of them(also when disconnecting/connecting usb modem on working system), in that case all is working and system loads successfully, but sometimes there is stream of them on the screen, and in that case system not loading - thats main problem/bug i'm talking about.

openmptcprouter is based on openwrt LEDE with all rpi patches to kernel added, current kernel version is 4.14.67

Is there any way i can help to find where problem is ? I can reproduce it without any problems

JamesH65, 2018-09-06:
As I asked above, does the same issue happen in Raspbian? For the moment we still haven't determined if it's actually something to do with the OS you are using or whether it's kernel/driver related.

pelwell, 2018-09-06:
Multi-path TCP is not available in Raspbian - it isn't even part of the standard kernel source, instead being made available as a patch set to be applied on top of the specific kernel version. As James says, unless you are seeing the same problem on Raspbian (or at the very least on a stock RPi kernel) there is nothing we can do.

rstanislav, 2018-09-06:
**6by9** can you please give me example code that needs to be added?

6by9, 2018-09-06:
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L1242
```
static void lan78xx_defer_kevent(struct lan78xx_net *dev, int work)
{
	set_bit(work, &dev->flags);
	if (!schedule_delayed_work(&dev->wq, 0)) {
		netdev_err(dev->net, "kevent %d may have been dropped\n", work);
		WARN(1, "LAN78xx EVENT DROPPED");
	}
}
```

rstanislav, 2018-09-07:
sorry for mess, but it looks like this.. output from uart, and it never stops

```
[   12.587504] usbcore: registered new interface driver rtl8192cu
[   12.594868] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.531519] random: crng init done
[   14.536148] random: 6 urandom warning(s) missed due to ratelimiting
[   14.714066] usb 1-1.2: USB disconnect, device number 5
[   15.142183] IPv6: ADDRCONF(NETDEV_UP): lo: link is not ready
[   15.150438] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   15.192801] usb 1-1.1.3: USB disconnect, device number 8
[   15.261161] usb 1-1.2: new high-speed USB device number 10 using dwc_otg
[   15.273915] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   15.273919] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   15.273925] ------------[ cut here ]------------
[   15.273938] WARNING: CPU: 0 PID: 0 at drivers/net/usb/lan78xx.c:1249 intr_complete+0x68/0xe8
** 123 printk messages dropped ** [   15.277899] [<8011cf44>] (tasklet_hi_action) from [<80101570>] (__do_softirq+0xe0/0x240)
** 370 printk messages dropped ** [   15.292015] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.14.67 #0
** 337 printk messages dropped ** [   15.303869]  libcrc32c tunnel4 ip_tunnel veth tun snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore l1oip mISDN_dsp mISDN_core nls_utf8 broadcom bcm_phy_lib md5 hmac vfat fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp pps_core
** 1079 printk messages dropped ** [   15.345796]  arp_tables sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dss1_divert isdn_bsdcomp isdn slhc ifb dm9000 sit sctp
** 2644 printk messages dropped ** [   15.438767]  arp_tables sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dss1_divert isdn_bsdcomp isdn slhc ifb dm9000 sit sctp
** 2510 printk messages dropped ** [   15.533839]  libcrc32c tunnel4 ip_tunnel veth tun snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore l1oip mISDN_dsp mISDN_core nls_utf8 broadcom bcm_phy_lib md5 hmac vfat fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp pps_core
** 1166 printk messages dropped ** [   15.575843]  arp_tables sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dss1_divert isdn_bsdcomp isdn slhc ifb dm9000 sit sctp
** 2488 printk messages dropped ** [   15.670890] CPU: 0 PID: 2897 Comm: sh Tainted: G        W       4.14.67 #0
** 359 printk messages dropped ** [   15.683979] [<801579b0>] (__handle_domain_irq) from [<8010c50c>] (__irq_svc+0x6c/0x90)
** 382 printk messages dropped ** [   15.697937] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.14.67 #0
** 375 printk messages dropped ** [   15.711763]  arp_tables sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conn [... cut ...]  nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp pps_core

** 1160 printk messages dropped ** [   31.511816]  libcrc32c tunnel4 ip_tunnel veth tun snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore l1oip mISDN_dsp mISDN_core nls_utf8 broadcom bcm_phy_lib md5 hmac vfat fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp pps_core
```

akaher, 2018-09-07:
https://github.com/raspberrypi/linux/issues/2447#issuecomment-418743834

moved to linux 4.18.6 ... issue mentioned in above link got fixed.

rstanislav, 2018-09-07:
Tried a few times more, once got different log:
```
[   15.620656] usb 1-1.2: USB disconnect, device number 5
[   15.947246] IPv6: ADDRCONF(NETDEV_UP): lo: link is not ready
[   15.955486] IPv6: ADDRCONF(NETDEV_UP): eth0: link is `not` ready
[   16.021202] usb 1-1.2: new high-speed USB device number 9 using dwc_otg
[   16.033775] usb 1-1.1.3: USB disconnect, device number 8
[   16.092280] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   16.099968] ------------[ cut here ]------------
[   16.105812] WARNING: CPU: 0 PID: 2124 at drivers/net/usb/lan78xx.c:1249 intr_complete+0x68/0xe8
[   16.116934] LAN78xx EVENT DROPPED
[   16.116936] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL
[   16.206312]  xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset sr9700 sierra_net ser_gigaset rtl8150 rndis_host r8712u r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asix arptable_filter arpt_mangle
[   16.294038]  arp_tables sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dss1_divert isdn_bsdcomp isdn slhc ifb dm9000 sit sctp
[   16.382184]  libcrc32c tunnel4 ip_tunnel veth tun snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore l1oip mISDN_dsp mISDN_core nls_utf8 broadcom bcm_phy_lib md5 hmac vfat fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp pps_core
[   16.421831] CPU: 0 PID: 2124 Comm: netifd Not tainted 4.14.67 #0
[   16.429508] Hardware name: BCM2835
[   16.434508] [<80110188>] (unwind_backtrace) from [<8010b94c>] (show_stack+0x10/0x14)
[   16.445423] [<8010b94c>] (show_stack) from [<805b14f0>] (dump_stack+0x98/0xb8)
[   16.454328] [<805b14f0>] (dump_stack) from [<8011887c>] (__warn+0xe4/0x100)
[   16.462929] [<8011887c>] (__warn) from [<801188d0>] (warn_slowpath_fmt+0x38/0x48)
[   16.473499] [<801188d0>] (warn_slowpath_fmt) from [<803df2e8>] (intr_complete+0x68/0xe8)
[   16.484699] [<803df2e8>] (intr_complete) from [<803f0168>] (__usb_hcd_giveback_urb+0xb0/0x118)
[   16.496430] [<803f0168>] (__usb_hcd_giveback_urb) from [<80416c28>] (completion_tasklet_func+0x64/0x90)
[   16.508980] [<80416c28>] (completion_tasklet_func) from [<8011cf44>] (tasklet_hi_action+0x94/0x10c)
[   16.521171] [<8011cf44>] (tasklet_hi_action) from [<80101570>] (__do_softirq+0xe0/0x240)
[   16.532380] [<80101570>] (__do_softirq) from [<8011d3dc>] (irq_exit+0xd4/0x138)
[   16.541331] [<8011d3dc>] (irq_exit) from [<801579b0>] (__handle_domain_irq+0x9c/0xac)
[   16.552258] [<801579b0>] (__handle_domain_irq) from [<8010c50c>] (__irq_svc+0x6c/0x90)
[   16.563339] [<8010c50c>] (__irq_svc) from [<805b0370>] (__memzero+0x30/0x7c)
[   16.572057] [<805b0370>] (__memzero) from [<801a2e54>] (get_page_from_freelist+0x5e8/0x88c)
[   16.583578] [<801a2e54>] (get_page_from_freelist) from [<801a36a8>] (__alloc_pages_nodemask+0xcc/0xc80)
[   16.596217] [<801a36a8>] (__alloc_pages_nodemask) from [<801b91d8>] (kmalloc_order+0x18/0x4c)
[   16.607986] [<801b91d8>] (kmalloc_order) from [<801b4778>] (kvmalloc_node+0x8c/0xa8)
[   16.618951] [<801b4778>] (kvmalloc_node) from [<804caf00>] (fq_codel_init+0x110/0x198)
[   16.630084] [<804caf00>] (fq_codel_init) from [<804c1d70>] (qdisc_create_dflt+0x60/0x7c)
[   16.641403] [<804c1d70>] (qdisc_create_dflt) from [<804c1de8>] (dev_activate+0x5c/0x200)
[   16.652726] [<804c1de8>] (dev_activate) from [<8049d5d4>] (__dev_open+0xe4/0x104)
[... cut ...]
** 35 printk messages dropped ** [   39.508449] ------------[ cut here ]------------
[   39.508454] WARNING: CPU: 0 PID: 2124 at drivers/net/usb/lan78xx.c:1249 intr_complete+0x68/0xe8
** 35 printk messages dropped ** [   39.531356] ------------[ cut here ]------------
[   39.531361] WARNING: CPU: 0 PID: 2124 at drivers/net/usb/lan78xx.c:1249 intr_complete+0x68/0xe8
[   39.531362] LAN78xx EVENT DROPPED
** 34 printk messages dropped ** [   39.554265] ------------[ cut here ]------------
[   39.554270] WARNING: CPU: 0 PID: 2124 at dr

jens-maus, 2018-09-07:
Nope, the system I am developing is a home automation platform (https://github.com/jens-maus/RaspberryMatic) and there no systemd oder NetworkManager is used. However, feel free to install it youself and test accordingly.

6by9, 2018-09-07:
The fact that several of these backtraces go back to ```do_vfs_ioctl``` (ie a system call from userspace) would indicate that userspace is doing something odd and repeatedly triggering an interface reset.

https://github.com/raspberrypi/linux/issues/2447#issuecomment-419394938 with the kworker thread apparently calling itself looks a little odd, but needs some more understanding of the networking and USB stacks to make sense of it.

rstanislav, 2018-09-07:
That userspace call could be ethtool --offload eth0 rx off tx off command that initiated at boot time to fix ethernet speed issue

[52 more replies]

closed 2018-12-19 by pelwell