raspberrypi/linux #2448: Xwayland Tainted: 4.9.80-v7+ #1098

Jamtek, 2018-03-17:
After upgrading to the latest kernel release on Raspian Stretch, on my Rpi 3b with dtoverlay=vc4-kms-v3d driver, I am getting a steady stream of trace dumps in kernel.log or dmesg when running Weston or kwin_wayland compositors with xwayland.

Seems more of a nuisance for now, but the log file file grows in size quickly. Are the Tainted messages supposed to be there or is there away to keep them from writing continuously to the log?

[31687.060180] ------------[ cut here ]------------
[31687.060221] WARNING: CPU: 0 PID: 29974 at kernel/irq/manage.c:527 __enable_irq+0x54/0x80
[31687.060226] Unbalanced enable for IRQ 40
[31687.060231] Modules linked in: cmac bnep hci_uart btbcm bluetooth evdev joydev panel_raspberrypi_touchscreen hid_logitech_hidpp vc4 drm_kms_helper brcmfmac brcmutil drm cfg80211 snd_soc_core snd_compress rfkill snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops snd_bcm2835 snd_pcm i2c_gpio i2c_algo_bit snd_timer snd i2c_bcm2835 bcm2835_gpiomem fixed uio_pdrv_genirq uio hid_logitech_dj ip_tables x_tables ipv6
[31687.060399] CPU: 0 PID: 29974 Comm: kwin_wayland Tainted: G        W       4.9.80-v7+ #1098
[31687.060404] Hardware name: BCM2835
[31687.060432] [<8010fa48>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[31687.060449] [<8010c058>] (show_stack) from [<80457a04>] (dump_stack+0xd4/0x118)
[31687.060468] [<80457a04>] (dump_stack) from [<8011d31c>] (__warn+0xf8/0x110)
[31687.060481] [<8011d31c>] (__warn) from [<8011d37c>] (warn_slowpath_fmt+0x48/0x50)
[31687.060495] [<8011d37c>] (warn_slowpath_fmt) from [<80175da8>] (__enable_irq+0x54/0x80)
[31687.060512] [<80175da8>] (__enable_irq) from [<80175e18>] (enable_irq+0x44/0x7c)
[31687.060594] [<80175e18>] (enable_irq) from [<7f499c40>] (vc4_irq_postinstall+0x24/0x40 [vc4])
[31687.060713] [<7f499c40>] (vc4_irq_postinstall [vc4]) from [<7f49c8f0>] (vc4_v3d_runtime_resume+0x58/0x60 [vc4])
[31687.060784] [<7f49c8f0>] (vc4_v3d_runtime_resume [vc4]) from [<8050391c>] (pm_generic_runtime_resume+0x3c/0x48)
[31687.060802] [<8050391c>] (pm_generic_runtime_resume) from [<80507724>] (__genpd_runtime_resume+0x3c/0x9c)
[31687.060818] [<80507724>] (__genpd_runtime_resume) from [<805091f4>] (genpd_runtime_resume+0xd8/0x1a4)
[31687.060832] [<805091f4>] (genpd_runtime_resume) from [<80504eb4>] (__rpm_callback+0x48/0x98)
[31687.060844] [<80504eb4>] (__rpm_callback) from [<80504f34>] (rpm_callback+0x30/0x90)
[31687.060855] [<80504f34>] (rpm_callback) from [<80506320>] (rpm_resume+0x4bc/0x704)
[31687.060868] [<80506320>] (rpm_resume) from [<805065d8>] (__pm_runtime_resume+0x70/0x9c)
[31687.060931] [<805065d8>] (__pm_runtime_resume) from [<7f49678c>] (vc4_submit_cl_ioctl+0x160/0x954 [vc4])
[31687.061275] [<7f49678c>] (vc4_submit_cl_ioctl [vc4]) from [<7f310c60>] (drm_ioctl+0x20c/0x428 [drm])
[31687.061522] [<7f310c60>] (drm_ioctl [drm]) from [<80283f68>] (do_vfs_ioctl+0xac/0x820)
[31687.061540] [<80283f68>] (do_vfs_ioctl) from [<80284720>] (SyS_ioctl+0x44/0x6c)
[31687.061558] [<80284720>] (SyS_ioctl) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
[31687.061565] ---[ end trace 843ae48489738508 ]---

Jamtek, 2018-03-19:
Another user has opened bug report on Launchpad for the same issue.

[Raspbian Launchpad](https://bugs.launchpad.net/raspbian/+bug/1756553)

gfcittolin, 2018-03-21:
I can confirm this is happening on the same kernel version, with the same stack trace (except from the process that's generatig it, in our case, electron). This happens every time the screen is painted in our setup.

DestyNova, 2018-05-27:
I'm seeing this too, same kernel version but with Xorg instead of Wayland:
```
May 27 16:32:49 turlpi kernel: [ 4341.349953] ------------[ cut here ]------------
May 27 16:32:49 turlpi kernel: [ 4341.349985] WARNING: CPU: 3 PID: 1649 at kernel/irq/manage.c:527 __enable_irq+0x54/0x80
May 27 16:32:49 turlpi kernel: [ 4341.349990] Unbalanced enable for IRQ 40
May 27 16:32:49 turlpi kernel: [ 4341.349994] Modules linked in: binfmt_misc fuse rfcomm bnep hci_uart btbcm bluetooth vc4 drm_kms_helper drm snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops brcmfmac brcmutil snd_bcm2835 snd_pcm cfg80211 snd_timer snd rfkill bcm2835_gpiomem uio_pdrv_genirq uio fixed evdev joydev hid_pl i2c_dev ip_tables x_tables ipv6
May 27 16:32:49 turlpi kernel: [ 4341.350135] CPU: 3 PID: 1649 Comm: Xorg Tainted: G        W       4.9.80-v7+ #1098
May 27 16:32:49 turlpi kernel: [ 4341.350140] Hardware name: BCM2835
May 27 16:32:49 turlpi kernel: [ 4341.350165] [<8010fa48>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
May 27 16:32:49 turlpi kernel: [ 4341.350182] [<8010c058>] (show_stack) from [<80457a04>] (dump_stack+0xd4/0x118)
May 27 16:32:49 turlpi kernel: [ 4341.350199] [<80457a04>] (dump_stack) from [<8011d31c>] (__warn+0xf8/0x110)
May 27 16:32:49 turlpi kernel: [ 4341.350212] [<8011d31c>] (__warn) from [<8011d37c>] (warn_slowpath_fmt+0x48/0x50)
May 27 16:32:49 turlpi kernel: [ 4341.350226] [<8011d37c>] (warn_slowpath_fmt) from [<80175da8>] (__enable_irq+0x54/0x80)
May 27 16:32:49 turlpi kernel: [ 4341.350240] [<80175da8>] (__enable_irq) from [<80175e18>] (enable_irq+0x44/0x7c)
May 27 16:32:49 turlpi kernel: [ 4341.350314] [<80175e18>] (enable_irq) from [<7f4a8c40>] (vc4_irq_postinstall+0x24/0x40 [vc4])
May 27 16:32:49 turlpi kernel: [ 4341.350429] [<7f4a8c40>] (vc4_irq_postinstall [vc4]) from [<7f4ab8f0>] (vc4_v3d_runtime_resume+0x58/0x60 [vc4])
May 27 16:32:49 turlpi kernel: [ 4341.350498] [<7f4ab8f0>] (vc4_v3d_runtime_resume [vc4]) from [<8050391c>] (pm_generic_runtime_resume+0x3c/0x48)
May 27 16:32:49 turlpi kernel: [ 4341.350516] [<8050391c>] (pm_generic_runtime_resume) from [<80507724>] (__genpd_runtime_resume+0x3c/0x9c)
May 27 16:32:49 turlpi kernel: [ 4341.350531] [<80507724>] (__genpd_runtime_resume) from [<805091f4>] (genpd_runtime_resume+0xd8/0x1a4)
May 27 16:32:49 turlpi kernel: [ 4341.350544] [<805091f4>] (genpd_runtime_resume) from [<80504eb4>] (__rpm_callback+0x48/0x98)
May 27 16:32:49 turlpi kernel: [ 4341.350556] [<80504eb4>] (__rpm_callback) from [<80504f34>] (rpm_callback+0x30/0x90)
May 27 16:32:49 turlpi kernel: [ 4341.350568] [<80504f34>] (rpm_callback) from [<80506320>] (rpm_resume+0x4bc/0x704)
May 27 16:32:49 turlpi kernel: [ 4341.350580] [<80506320>] (rpm_resume) from [<805065d8>] (__pm_runtime_resume+0x70/0x9c)
May 27 16:32:49 turlpi kernel: [ 4341.350642] [<805065d8>] (__pm_runtime_resume) from [<7f4a578c>] (vc4_submit_cl_ioctl+0x160/0x954 [vc4])
May 27 16:32:49 turlpi kernel: [ 4341.350959] [<7f4a578c>] (vc4_submit_cl_ioctl [vc4]) from [<7f3e7c60>] (drm_ioctl+0x20c/0x428 [drm])
May 27 16:32:49 turlpi kernel: [ 4341.351197] [<7f3e7c60>] (drm_ioctl [drm]) from [<80283f68>] (do_vfs_ioctl+0xac/0x820)
May 27 16:32:49 turlpi kernel: [ 4341.351215] [<80283f68>] (do_vfs_ioctl) from [<80284720>] (SyS_ioctl+0x44/0x6c)
May 27 16:32:49 turlpi kernel: [ 4341.351233] [<80284720>] (SyS_ioctl) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
May 27 16:32:49 turlpi kernel: [ 4341.351240] ---[ end trace dd1f75fd91d2bf24 ]---
pi@turlpi:/var/log $ uname -a
Linux turlpi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```
I only noticed because the root partition ran out of space due to massive `/var/log/(messages|kern.log|syslog)` which are all full of the same stack traces (they got to about 800 megs each by the time the disk ran out of space and everything stopped working).

pelwell, 2018-05-27:
The latest Raspbian image is based on the 4.14 kernel - have you considered updating?

DestyNova, 2018-05-27:
Yes I was just coming back to report that it seems fine so far in 4.14.34-v7+ :+1:

JamesH65, 2018-06-26:
@DestyNova @Jamtek Since we do not plan to make any change to 4.9, and all development is now on 4,.14 (and has been for some time), can you doubly confirm that all is well on 4.14, and if so, close the issue? Thanks.

Jamtek, 2018-06-29:
@JamesH65 Yes, in 4.14 branch I am no longer seeing the messages.  
--
Thank you

closed 2018-06-29 by Jamtek