raspberrypi/linux #2449: Pi3B+ : USB+ethernet file transfer problem

toto8551, 2018-03-18:
Hi,

I have problems while transferring files from an USB external HDD or USB flash drive connected on the new RPi 3B+ to another computer via SFTP or samba. After a random amount of data transfered, the transfer rate drop to 0 and stay like that for at least 30 min ( I canceled the transfer after that time).

I cannot see any error on the standard raspbian system logs. I am running raspbian stretch with the last updates, I also did a firmware upgrade with rpi-update.

The RPi 3B+ is powered with an official raspberry power supply (2.5A) and I tried with a spare one.
I tried to give more power with an external power supply for the USB HDD without any change, but like I got the same problem with an USB flash drive, I don't expect a power issue or ?

I tried to transfer files with thunar, pcmanfm, scp or rsync. rsync is the only one working, but with a strong fluctuation of the transfer rate (from 0 to 20Mo/s). scp tell me that the file is "stalled" and stop any progress.

If I start the same SD card with a RPi 3B (not plus) the file transfer is working like a charm ( I used it for almost 3 years like that).

I followed tips from this link without success.

I having no problems during ssh sessions or with vnc.

Any ideas ?

mkreisl, 2018-03-18:
Having same strange issues with Pi3B+ [1]

IMO it's network driver (lan78xx) related or the usb host driver does not handle usb transfers correctly.

After switching from eth to wlan, everything is ok

[1]
Was never able to make an 3GB image backup (root fs is on external usb disk) over network (kernel 4.9, 4.14 and 4.15 tested). Same configuration works on an older Pi3B without any issues

toto8551, 2018-03-19:
I did another test.

Now I am back with the official kernel of the raspbian stretch : 4.9.80-v7, and it is working up to now.
I transfered 10Go from an USB HDD with a rate of 20 Mo/s.

PGallagher69, 2018-03-20:
I've got the same problem here... Trying to transfer large files over ethernet fails after a minute or so... Fine on Wifi... However, when I try to downgrade to 4.9.80-v7, I get four flashing green lights on the PI3B+... I can take the card out and plug it into a Pi2 and it boots up just fine... I then have to re-update the firmware and swap everything back to the Pi3B+... Frustrating...

fiery-, 2018-03-21:
Is this bug related to #2446? If so, then the forthcoming kernel 4.14.28 should fix it. Reverting to an old kernel may be a temporary solution, but only for advanced users.

Are other distros affected: xbian, osmc, LibreELEC? 

At least we know that OSMC used [pre-production units](https://osmc.tv/2018/03/osmcs-march-update-is-here-with-pi-3-b-support/), so may be they have solutions to this problem.

I have 10 days to decide to return the RPi 3B+, so I want to start with a working Ethernet or just return the unit unopened.

mkreisl, 2018-03-21:
> Is this bug related to #2446? If so, then the forthcoming kernel 4.14.28 should fix it. Reverting to an old kernel may be a temporary solution, but only for advanced users.

IMO there is a good chance to solve some strange issues.

As already reported, I'm struggling with my Pi3B+ since I got that part. Raspbian seems to work in my testing scenario [1], but XBian was crashing always, sooner or later. The major difference between Raspbian and XBian is, that Rasbian is using ipv6, whereas XBian is using ipv4 (ipv6 is disabled per default), and those patches fixing the ipv4 stack. After enabling tcp6 on XBian, my test seems so finish successfully.

[1] Root fs on usb disk (60GB), mounting Samba share on my server to /mnt, and then run dd if=/dev/sda of=/mnt/test.img bs=1M status=process. Raspbian succeeded, XBian crashes always in differen ways: kernel oops, kernel panic of just completely freezed.

pelwell, 2018-03-22:
If you have a Linux target machine, can you try running this one-liner (you'll need to change the login credentials for the target):
```
dd if=/dev/zero bs=1M status=progress | ssh user@target "cat >/dev/null"
```
I've had that running overnight and it's approaching 2TB transferred without issue. The channel is encrypted, so the 29.4MB/s (235Mb/s) it's achieving isn't bad.

If that's solid, switch to:
```
while true; do date 1>&2; sudo dd if=/dev/mmcblk0 bs=1M status=progress; done | ssh user@target "cat >/dev/null"
```
Then:
```
while true; do date 1>&2; sudo dd if=/dev/sda bs=1M status=progress; done | ssh user@target "cat >/dev/null"
```

mkreisl, 2018-03-22:
@pelwell 
Sure, there are many many ways to run a test. Unfortunately, after my tests has been finished successfully, my real application making an image backup started from Kodi GUI still kills the Pi3B+ completely [1], in different ways (4.14.27+ mostly throws kernel Oops #5, so built kernel with netdev patch, Oops were gone, but now still getting VFS CIFS stuck for more than 15s message, and Kernel 4.14.29+ build tonight did not make any difference).

Conclusion:
Pi3B+ is at the moment absolutely unusable, crappiest Pi ever had. Extremely frustrating

[1] That backup is using btrfs send/receive, sends all local btrfs subvolumes (located on external USB disk) to an image mounted on a network share (cifs of nfs).

pelwell, 2018-03-22:
> Pi3B+ is at the moment absolutely unusable, crappiest Pi ever had. Extremely frustrating

For you, apparently, but not for me - that's why I'm trying to be methodical and work out which step is causing the problem.

Ferroin, 2018-03-22:
@mkreisl Out of curiosity, could you possibly try using sshfs instead of CIFS or NFS?  It's a FUSE module instead of a native filesystem driver, and it uses a completely different (and much more efficient) protocol on-the-wire, so it may work even though CIFS and NFS aren't.  `sudo apt-get install sshfs` should get it for you (I'm about 99% certain Raspbian has it).  If that works, it should at least give you something you can use in the short term until this gets figured out.

mkreisl, 2018-03-22:
> For you, apparently, but not for me - that's why I'm trying to be methodical and work out which step is causing the problem.

Those basic tests you are running does cause the issue. IMO many many data in and out of the USB is causing the problem. That's the main difference between Pi3B nd Pi3B+, Pi3B has one usb hub, and Pi3B+ has two:

Pi3B:
```
Bus 001 Device 004: ID 046d:c503 Logitech, Inc. Cordless Mouse+Keyboard Receiver
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Pi3B+:
```
Bus 001 Device 004: ID 046d:c503 Logitech, Inc. Cordless Mouse+Keyboard Receiver
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mkreisl, 2018-03-22:
@Ferroin sshfs doesn't make a difference, Pi3B+ dying with kernel Oops again (Pi3B works like a charm)
```
Mar 22 17:40:00 kmxbilr2 kernel: [  555.453472] Unable to handle kernel NULL pointer dereference at virtual address 0000000d
Mar 22 17:40:00 kmxbilr2 kernel: [  555.453480] pgd = a12d8000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.453484] [0000000d] *pgd=00000000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.453493] Internal error: Oops: 5 [#1] SMP ARM
Mar 22 17:40:00 kmxbilr2 kernel: [  555.453578] Modules linked in: dm_mod dax fuse loop hci_uart bluetooth ecdh_generic sg uio_pdrv_genirq uio lirc_rpi(C) lirc_dev fixed frandom ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables ipv6 snd_bcm2835(C) snd_pcm snd_timer snd brcmfmac cfg80211 rfkill brcmutil evdev rpcsec_gss_krb5 tun uinput
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454081] CPU: 1 PID: 10626 Comm: sudo Tainted: G         C      4.14.29+ #1
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454170] Hardware name: BCM2835
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454216] task: 91ab1e00 task.stack: 93ea0000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454287] PC is at locks_remove_posix+0x30/0x14c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454364] LR is at filp_close+0x68/0x8c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454447] pc : [<802e08b0>]    lr : [<80286ac4>]    psr: 20000013
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454550] sp : 93ea1eb0  ip : 93ea1f58  fp : 93ea1f54
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454616] r10: 00000000  r9 : 93ea0000  r8 : 80108204
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454680] r7 : 00000006  r6 : acc63900  r5 : 9d640180  r4 : ae690440
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454754] r3 : 00000001  r2 : ad1bb940  r1 : acc63900  r0 : 9d640180
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454831] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454914] Control: 10c5383d  Table: 212d806a  DAC: 00000055
Mar 22 17:40:00 kmxbilr2 kernel: [  555.454983] Process sudo (pid: 10626, stack limit = 0x93ea0210)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455058] Stack: (0x93ea1eb0 to 0x93ea2000)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455112] 1ea0:                                     802fe824 91ab1e00 00000044 802a8f5c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455210] 1ec0: 00000004 00000000 00000000 00000100 81f22e80 80d093bc 00000017 808ac294
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455316] 1ee0: 76ebfe2c 93ea1fb0 7ed36760 808a8780 93ea1f14 93ea1f00 808a8780 806fbae8
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455440] 1f00: add51c00 add51d74 93ea1f34 93ea1f18 806fbae8 808a7758 add52400 7f0220bc
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455561] 1f20: add52400 add52444 93ea1f54 00000000 9d640180 acc63900 00000006 80108204
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455679] 1f40: 93ea0000 00000000 93ea1f74 93ea1f58 80286ac4 802e088c 0000003c acc63900
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455787] 1f60: 9d640180 00000006 93ea1f94 93ea1f78 802aad38 80286a68 76fbd218 0000000b
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455892] 1f80: 00000000 00000006 93ea1fa4 93ea1f98 80286b18 802aac7c 00000000 93ea1fa8
Mar 22 17:40:00 kmxbilr2 kernel: [  555.455993] 1fa0: 80108060 80286af4 76fbd218 0000000b 0000003c 7ed36760 76fbcb5c 76fbcb5c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456102] 1fc0: 76fbd218 0000000b 00000000 00000006 01d185b8 7ed36760 76fface8 0000003c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456205] 1fe0: 7ed36738 7ed36728 76fbcb6c 76e94fc2 60000030 0000003c 00000000 00000000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456325] [<802e08b0>] (locks_remove_posix) from [<80286ac4>] (filp_close+0x68/0x8c)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456435] [<80286ac4>] (filp_close) from [<802aad38>] (__close_fd+0xc8/0xec)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456529] [<802aad38>] (__close_fd) from [<80286b18>] (SyS_close+0x30/0x58)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456626] [<80286b18>] (SyS_close) from [<80108060>] (ret_fast_syscall+0x0/0x28)
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456728] Code: e59430e8 f57ff05b e3530000 0a000025 (e5b3200c) 
Mar 22 17:40:00 kmxbilr2 kernel: [  555.456857] ---[ end trace 124dc21b94998788 ]---
Mar 22 17:40:00 kmxbilr2 kernel: [  555.477670] Unable to handle kernel NULL pointer dereference at virtual address 0000000d
Mar 22 17:40:00 kmxbilr2 kernel: [  555.477836] pgd = 80004000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.477896] [0000000d] *pgd=00000000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.477964] Internal error: Oops: 5 [#2] SMP ARM
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478032] Modules linked in: dm_mod dax fuse loop hci_uart bluetooth ecdh_generic sg uio_pdrv_genirq uio lirc_rpi(C) lirc_dev fixed frandom ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables ipv6 snd_bcm2835(C) snd_pcm snd_timer snd brcmfmac cfg80211 rfkill brcmutil evdev rpcsec_gss_krb5 tun uinput
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478580] CPU: 3 PID: 10620 Comm: sudo Tainted: G      D  C      4.14.29+ #1
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478678] Hardware name: BCM2835
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478725] task: 82818f00 task.stack: 86ea8000
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478800] PC is at locks_remove_posix+0x30/0x14c
Mar 22 17:40:00 kmxbilr2 kernel: [  555.478867] LR is at filp_close+0x68/0x8c
[... cut ...]
Mar 22 17:40:05 kmxbilr2 kernel: [  561.144836] [<8029a720>] (user_path_at_empty) from [<80287344>] (SyS_faccessat+0x9c/0x1f8)
Mar 22 17:40:05 kmxbilr2 kernel: [  561.155787] [<80287344>] (SyS_faccessat) from [<802874c4>] (SyS_access+0x24/0x28)
Mar 22 17:40:05 kmxbilr2 kernel: [  561.166755] [<802874c4>] (SyS_access) from [<80108060>] (ret_fast_syscall+0x0/0x28)
Mar 22 17:40:05 kmxbilr2 kernel: [  561.177608] Code: e1a07005 da000015 f590f000 e1b04f9f (e1340006) 
Mar 22 17:40:05 kmxbilr2 kernel: [  561.188475] ---[ end trace 124dc21b9499878b ]---
```

pelwell, 2018-03-22:
If you have vcgencmd, please report the output of `vcgencmd get_throttled`.

Ferroin, 2018-03-22:
Hmm, that's interesting...

All of the stack traces in those kernel OOPS messages implicate absolutely nothing from the networking stack, they've just got generic core kernel stuff and VFS layer functions referenced.

The complaint about a kernel NULL pointer dereference at a very low virtual address is also somewhat suspicious.  Stuff like that is usually indicative of either hardware failure, or a very poorly behaved kernel driver scribbling on memory locations it shouldn't be touching.

mkreisl, 2018-03-22:
> All of the stack traces in those kernel OOPS messages implicate absolutely nothing from the networking stack, they've just got generic core kernel stuff and VFS layer functions referenced.

I know, could it be that corrupted  data read from usb disk caused this?

mkreisl, 2018-03-22:
> Thanks - that rules out a power supply issue.

Already tried different PSU's, usually using brand new 2.5A PSU for Pi3B+ and an older one (2A) for Pi3B

pelwell, 2018-03-22:
Yes, so you said, but I'd rather hear it from the Pi.

mkreisl, 2018-03-22:
@pelwell 
Are there any tests I can run to make absolutely sure that hardware is ok?

Btw, I have similar issues if root fs is on iSCSI target and no usb drive is connected and used.

mkreisl, 2018-03-22:
@pelwell @Ferroin 
Oh, and before I forget: 
If I'm using onboard WLAN instead of onboard ethernet, issues are still there
and, already changed network cable and port on the network switch
and, switching root to sd-card does not make any difference, getting this on my final test for today:
```
[  515.046170] CIFS VFS: sends on sock a715d500 stuck for 15 seconds
[  515.046196] CIFS VFS: Error -11 sending data on socket to server
[  517.443871] systemd-logind[3010]: New session c2 of user root.
[  530.166192] CIFS VFS: sends on sock a715d500 stuck for 15 seconds
[  530.166217] CIFS VFS: Error -11 sending data on socket to server
[  540.648102] Status code returned 0xc0000008 NT_STATUS_INVALID_HANDLE
[  540.648134] CIFS VFS: Send error in read = -9
[  540.648161] print_req_error: I/O error, dev loop0, sector 231328
[  540.648195] BTRFS error (device dm-1): bdev /dev/mapper/loop0p2 errs: wr 0, rd 1, flush 0, corrupt 0, gen 0
[  735.206538] INFO: task kworker/u8:2:219 blocked for more than 120 seconds.
[  735.206553]       Tainted: G         C      4.14.29+ #1
[  735.206558] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  735.206567] kworker/u8:2    D    0   219      2 0x00000000
[  735.206596] Workqueue: writeback wb_workfn (flush-btrfs-3)
[  735.206633] [<808a5ecc>] (__schedule) from [<808a6544>] (schedule+0x50/0xa8)
[  735.206652] [<808a6544>] (schedule) from [<80476418>] (btrfs_tree_lock+0x15c/0x278)
[  735.206672] [<80476418>] (btrfs_tree_lock) from [<804578e0>] (lock_extent_buffer_for_io+0xec/0x270)
[  735.206694] [<804578e0>] (lock_extent_buffer_for_io) from [<8045b160>] (btree_write_cache_pages+0x248/0x344)
[  735.206716] [<8045b160>] (btree_write_cache_pages) from [<80422eb0>] (btree_writepages+0x84/0x8c)
[  735.206734] [<80422eb0>] (btree_writepages) from [<8022edb4>] (do_writepages+0x30/0x8c)
[  735.206750] [<8022edb4>] (do_writepages) from [<802bb208>] (__writeback_single_inode+0x44/0x434)
[  735.206766] [<802bb208>] (__writeback_single_inode) from [<802bbb08>] (writeback_sb_inodes+0x214/0x4c4)
[  735.206781] [<802bbb08>] (writeback_sb_inodes) from [<802bbe48>] (__writeback_inodes_wb+0x90/0xd0)
[  735.206797] [<802bbe48>] (__writeback_inodes_wb) from [<802bc120>] (wb_writeback+0x298/0x33c)
[  735.206811] [<802bc120>] (wb_writeback) from [<802bc9b8>] (wb_workfn+0xdc/0x4d8)
[  735.206831] [<802bc9b8>] (wb_workfn) from [<801374b8>] (process_one_work+0x158/0x454)
[  735.206850] [<801374b8>] (process_one_work) from [<80137810>] (worker_thread+0x5c/0x5b0)
[  735.206868] [<80137810>] (worker_thread) from [<8013d8a4>] (kthread+0x13c/0x16c)
[  735.206886] [<8013d8a4>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  735.206920] INFO: task kworker/u8:3:399 blocked for more than 120 seconds.
[  735.206926]       Tainted: G         C      4.14.29+ #1
[  735.206931] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  735.206936] kworker/u8:3    D    0   399      2 0x00000000
[  735.206954] Workqueue: btrfs-endio-write btrfs_endio_write_helper
[  735.206972] [<808a5ecc>] (__schedule) from [<808a6544>] (schedule+0x50/0xa8)
[  735.206988] [<808a6544>] (schedule) from [<808aa12c>] (schedule_timeout+0x1d0/0x3e4)
[  735.207005] [<808aa12c>] (schedule_timeout) from [<808a71cc>] (wait_for_common+0xc0/0x184)
[  735.207021] [<808a71cc>] (wait_for_common) from [<808a72b0>] (wait_for_completion+0x20/0x24)
[  735.207036] [<808a72b0>] (wait_for_completion) from [<8040c478>] (btrfs_async_run_delayed_refs+0x12c/0x14c)
[  735.207053] [<8040c478>] (btrfs_async_run_delayed_refs) from [<8042d7ec>] (__btrfs_end_transaction+0x228/0x324)
[  735.207069] [<8042d7ec>] (__btrfs_end_transaction) from [<8042d904>] (btrfs_end_transaction+0x1c/0x20)
[  735.207085] [<8042d904>] (btrfs_end_transaction) from [<80438b40>] (btrfs_finish_ordered_io+0x230/0x848)
[  735.207102] [<80438b40>] (btrfs_finish_ordered_io) from [<8043955c>] (finish_ordered_fn+0x1c/0x20)
[  735.207118] [<8043955c>] (finish_ordered_fn) from [<8046a1d8>] (normal_work_helper+0xb0/0x3e0)
[  735.207135] [<8046a1d8>] (normal_work_helper) from [<8046a8f0>] (btrfs_endio_write_helper+0x1c/0x20)
[  735.207152] [<8046a8f0>] (btrfs_endio_write_helper) from [<801374b8>] (process_one_work+0x158/0x454)
[  735.207170] [<801374b8>] (process_one_work) from [<80137810>] (worker_thread+0x5c/0x5b0)
[  735.207187] [<80137810>] (worker_thread) from [<8013d8a4>] (kthread+0x13c/0x16c)
[  735.207203] [<8013d8a4>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  735.207217] INFO: task kworker/u8:6:759 blocked for more than 120 seconds.
[  735.207224]       Tainted: G         C      4.14.29+ #1
[  735.207229] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  735.207233] kworker/u8:6    D    0   759      2 0x00000000
[  735.207250] Workqueue: btrfs-extent-refs btrfs_extent_refs_helper
[  735.207269] [<808a5ecc>] (__schedule) from [<808a6544>] (schedule+0x50/0xa8)
[  735.207284] [<808a6544>] (schedule) from [<80475e2c>] (btrfs_tree_read_lock+0x120/0x198)
[  735.207303] [<80475e2c>] (btrfs_tree_read_lock) from [<803f9660>] (btrfs_read_lock_root_node+0x38/0x50)
[  735.207321] [<803f9660>] (btrfs_read_lock_root_node) from [<803ff188>] (btrfs_search_slot+0x8b0/0xb10)
[  735.207339] [<803ff188>] (btrfs_search_slot) from [<80400f10>] (btrfs_insert_empty_items+0x7c/0xd4)
[  735.207357] [<80400f10>] (btrfs_insert_empty_items) from [<8040f01c>] (__btrfs_run_delayed_refs+0xe90/0x16fc)
[... cut ...]
[  858.088975] [<8043c074>] (btrfs_setattr) from [<802a9318>] (notify_change+0x1cc/0x408)
[  858.088993] [<802a9318>] (notify_change) from [<80286c98>] (do_truncate+0x90/0xc0)
[  858.089008] [<80286c98>] (do_truncate) from [<80286eac>] (vfs_truncate+0x1e4/0x25c)
[  858.089022] [<80286eac>] (vfs_truncate) from [<80286fa0>] (do_sys_truncate+0x7c/0xac)
[  858.089036] [<80286fa0>] (do_sys_truncate) from [<80287200>] (SyS_truncate64+0x18/0x1c)
[  858.089052] [<80287200>] (SyS_truncate64) from [<80108060>] (ret_fast_syscall+0x0/0x28)
```

JamesH65, 2018-03-22:
Have you got another Pi3B+ you can try this on. It really is very bizarre, and bizarre always makes me think HW fault.

mkreisl, 2018-03-22:
> Have you got another Pi3B+ you can try this on. It really is very bizarre, and bizarre always makes me think HW fault.

No, unfortunately not

pelwell, 2018-03-22:
It doesn't feel like a hardware fault to me - if so, you would expect to see it on a stock Raspbian kernel as well, and so far I don't think we have.

mkreisl, 2018-03-22:
> if so, you would expect to see it on a stock Raspbian kernel as well

Already posted It happens on Raspbian using standard kernel (4.14.27+) as well

pelwell, 2018-03-23:
Yes, reading your earlier comment again I can see how it means that.

Can you try limiting the ARM cores to 1.2GHz by adding the following to config.txt?:
```
arm_freq=1200
```

mkreisl, 2018-03-23:
@pelwell 
I'll try this. Currently running another test to a different machine, with very strange result

getting periodically message like this:
```
[ 1416.176886] nfs: server kmxbmc not responding, still trying
[ 1417.217789] nfs: server kmxbmc not responding, still trying
[ 1417.575681] nfs: server kmxbmc OK
[ 1417.588839] nfs: server kmxbmc OK
```

The interval is exactly 239s. Any idea where this value comes from. NFS share is mounted as follows

```
kmxbmc:/srv on /mnt type nfs (rw,relatime,vers=3,rsize=524288,wsize=524288,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.1.5,mountvers=3,mountport=20048,mountproto=udp,local_lock=none,addr=192.168.1.5)
```

mkreisl, 2018-03-23:
@pelwell
Doesn't help :disappointed:

mkreisl, 2018-03-23:
... and it does not depend on kernel version. 4.9.xx producing same issues

mkreisl, 2018-03-23:
@fiery- 
Nothing special. Connected directly to my router (4-port GBit switch) via 5m CAT5 ethernet cable
Sending data from /dev/zero to server 33MB/s
Receiving data from server 29MB/s
You haven't read this thread completey. Issue appears on WLAN connection also

lategoodbye, 2018-03-23:
@mkreisl 
Any suspicious output in dmesg?
What is the MTU of your onboard ethernet interface?
Does it help to reduce the MTU to 1496?
What the output of `ethtool -S eth0` after the issue appeared?

mkreisl, 2018-03-23:
@lategoodbye 

> Any suspicious output in dmesg?

Already posted dmesg (see above). But output differs extremely. Sometimes nothing, copy process just stucks, sometimes system completely freezed, sometimes message below, sometimes kernel Oops ...

```
[  347.207072] CIFS VFS: sends on sock 81c33340 stuck for 15 seconds
[  347.207099] CIFS VFS: Error -11 sending data on socket to server
[  362.328036] CIFS VFS: sends on sock 81c33340 stuck for 15 seconds
[  362.328066] CIFS VFS: Error -11 sending data on socket to server
[  362.418887] CIFS VFS: Free previous auth_key.response = ad8a3540
[  368.135439] CIFS VFS: Free previous auth_key.response = 9cd2a840
[  372.970214] Status code returned 0xc0000128 STATUS_FILE_CLOSED
[  372.970259] CIFS VFS: Send error in read = -9
[  372.970297] print_req_error: I/O error, dev loop0, sector 296512
[  372.970334] BTRFS error (device dm-1): bdev /dev/mapper/loop0p2 errs: wr 0, rd 1, flush 0, corrupt 0, gen 0
[  389.918475] print_req_error: I/O error, dev loop0, sector 0
[  389.918755] BTRFS error (device dm-1): bdev /dev/mapper/loop0p2 errs: wr 0, rd 1, flush 1, corrupt 0, gen 0
[  389.918775] BTRFS warning (device dm-1): chunk 1048576 missing 1 devices, max tolerance is 0 for writeable mount
[  389.918808] BTRFS: error (device dm-1) in write_all_supers:3670: errno=-5 IO failure (errors while submitting device barriers.)
[  389.918825] BTRFS info (device dm-1): forced readonly
[  389.918841] BTRFS warning (device dm-1): Skipping commit of aborted transaction.
[  389.918851] BTRFS: error (device dm-1) in cleanup_transaction:1873: errno=-5 IO failure
[  389.918863] BTRFS info (device dm-1): delayed_refs has NO entry
```

> What is the MTU of your onboard ethernet interface?

Default (1500)

> Does it help to reduce the MTU to 1496?

No

> What the output of ethtool -S eth0 after the issue appeared?

```
NIC statistics:
     RX FCS Errors: 0
     RX Alignment Errors: 0
     Rx Fragment Errors: 0
     RX Jabber Errors: 0
     RX Undersize Frame Errors: 0
     RX Oversize Frame Errors: 0
     RX Dropped Frames: 0
     RX Unicast Byte Count: 82919099
     RX Broadcast Byte Count: 32537
     RX Multicast Byte Count: 199380
     RX Unicast Frames: 460932
     RX Broadcast Frames: 394
     RX Multicast Frames: 554
     RX Pause Frames: 708426
     RX 64 Byte Frames: 329
     RX 65 - 127 Byte Frames: 428916
     RX 128 - 255 Byte Frames: 2601
     RX 256 - 511 Bytes Frames: 592
     RX 512 - 1023 Byte Frames: 307
     RX 1024 - 1518 Byte Frames: 29135
     RX Greater 1518 Byte Frames: 0
     EEE RX LPI Transitions: 0
     EEE RX LPI Time: 0
     TX FCS Errors: 0
     TX Excess Deferral Errors: 0
     TX Carrier Errors: 0
     TX Bad Byte Count: 0
     TX Single Collisions: 0
     TX Multiple Collisions: 0
     TX Excessive Collision: 0
     TX Late Collisions: 0
     TX Unicast Byte Count: 1673525473
     TX Broadcast Byte Count: 4804
     TX Multicast Byte Count: 26394
     TX Unicast Frames: 1126605
     TX Broadcast Frames: 44
     TX Multicast Frames: 118
     TX Pause Frames: 8
     TX 64 Byte Frames: 37
     TX 65 - 127 Byte Frames: 13575
     TX 128 - 255 Byte Frames: 8217
     TX 256 - 511 Bytes Frames: 1747
     TX 512 - 1023 Byte Frames: 144
     TX 1024 - 1518 Byte Frames: 1103047
     TX Greater 1518 Byte Frames: 0
     EEE TX LPI Transitions: 0
     EEE TX LPI Time: 0
```

mkreisl, 2018-03-23:
Now I did final test, put 100MBit switch between, to force a 100MBit link instead of 1GBit. 

Makes no difference, now sending back that crap :angry:

Knoppix1, 2018-03-23:
The same with samba and share (NFS / etx4 ...) In HDD extern or SD internal. After reboot its work perfectly for 30/40min and after kick of share and reconnect again and again.
Network work perfectly (no ping out ...) I feel owerload the queue or something like that.

Otw-cz, 2018-03-23:
Same problem here. I'm using Samba and external USB HDD. It's quite random, but it just a matter of time, no dmesg problems indicated, throttled=0x0. RPi is connected directly to 8 port GB switch via cat5e cable.

Phlo5Cof, 2018-03-24:
I have exactly the same problem and I'm glad to see I'm not alone in this, after rebooting I can transfer files through sftp (using filezilla on windows) without issue but after half an hour or so after booting up the transfers don't work anymore: It hangs after a second or two of high speed transfer (18MBps ), then times out after and tries to reconnect after 20 seconds only to fail again after a second or two of transfer. I also tried the same with transferring the files from the sd card (without hdd connected) and it has the same issues, when I put the sd card into a 3B it has no issues whatsoever tranferring these files even from the hard drive.
I'm using Raspbian and the official Raspberry Pi charger if that helps.
If it is hardware related I'd love to know so I can get it exchanged by the seller.

mkreisl, 2018-03-24:
Seems that the Raspberry Pi team hasn't tested anything before releasing the Pi3B+ :angry:

JamesH65, 2018-03-24:
Yes, that right, because we just knocked up some hardware, and never
checked anything worked before releasing it.....

Or on the other hand perhaps we spent a year making this product and did,
actually, test it, but just never saw this error?

Might also be worth considering that we have sold >200k B+ already, and the
number of reports of this error are small in comparison, so its perhaps
some weird interaction between hubs, drivers, firmware and hardware.
Whatever it is, we are looking in to it. My suspicion is that its a driver
issue that we will figure out, although sometimes these things can take a
while. On the 3B, there was a bug in the smsc95xx driver, matched with a
bug in the brcmfmac software, that we didn't get to the bottom off for
about a year - but its was a difficult bug to reproduce. Neigher pieces of
software written by us incidentally, in much the same way that the lan78xx
driver or the chip firmware, which is presumably at fault here, wasn't
written by us either.

PGallagher69, 2018-03-24:
I'd just like to say that the team at Raspberry Pi are awesome! There's always niggles with new hardware and software, it's par for the course, and impossible to test every permutation out there. I have full confidence that they'll look into whatever issues there are and get a fix out when they can!

mkreisl, 2018-03-24:
@JamesH65 
You havent read this thread completely

**This issue happens regardless if lan78xx driver is used or not. I was building lan78xx as module and unloaded it before starting test
And issue remains**

JamesH65, 2018-03-24:
Not sure I mentioned the lan78xx driver in my post at all so not sure why you brought that up. I did say I think its a driver problem, and having skimmed  the thread that still seems the most likely explanation. Something somewhere is dumping on memory it shouldn't (which, incidentally, was the cause of the bug I referred to above, one driver was dumping on another drivers skb allocation). As always with these things, if we can replicate the issue at our end, then we have a much better chance of figuring it out.

pelwell, 2018-03-24:
The effort to find and fix this problem would be helped enormously if someone could write down the steps to get from a known OS image (ideally Raspbian, but it doesn't have to be) to an observed failure.

mkreisl, 2018-03-24:
@pelwell 
> The effort to find and fix this problem would be helped enormously if someone could write down the steps to get from a known OS image (ideally Raspbian, but it doesn't have to be) to an observed failure.

1) Install XBian
2) mount network share
3) run backup inside of xbian-config, menu item 6 (xbian copier), use file:/<mount-to network-share>/test.img as destination

Maybe some more data has to be written before running test (in my case I have about 2.5GB data to copy)

That's in short words what I'm doing

ps
Actually I have been successfuly written 5 images after

1) building lan78xx driver as module
2) using wlan and not ethernet, unloading lan78xx module (this makes things a bit better but issue still remains and Kodi crashes randomly)
3) powering off the usb part by running script (after this everything seems to be stable and Kodi does not crash)

```
#!/bin/bash

BUSPOWER="/sys/devices/platform/bcm2708_usb/buspower"
for USB in $(find /sys/devices/platform/soc/ -name *.usb 2>/dev/null); do
    [ -e $USB ] && { BUSPOWER=$USB/buspower; break; }
done

echo 0x0 > $BUSPOWER
```

For me it sounds like the usbhub/ethernet (that's one of the major difference to Pi3B) part does not get stable power for a  reliable work (that chip has higher power consumption that the chip used on Pi3B)

pelwell, 2018-03-24:
Thank you - your instructions look mercifully short and simple. As you say, the lan78xx and its driver seems the most likely culprit, but I don't want to leap to any conclusions.

mkreisl, 2018-03-25:
> Thank you - your instructions look mercifully short and simple. As you say, the lan78xx and its driver seems the most likely culprit, but I don't want to leap to any conclusions.

... and for running test in non-interactive loop a script like this:

```
#!/bin/bash

for i in $(seq 1 10); do
    logger running test $i
    /usr/sbin/btrfs-auto-snapshot xbiancopy --helper --img /dev/root /srv/backup/test.img
done

```

That's what I'm running at the moment

Knoppix1, 2018-03-25:
@pelwell @JamesH65

it's very simple to reproduce !!! mount a network share paste a video and read it from the share. You will systematically logout the video after several minutes of reading...
it works also with a copy of a very large file.

lategoodbye, 2018-03-25:
Here some questions for those who can reproduce the issue:
Are you able to reproduce it with Raspbian?
Do you have an external HDD connected via USB during the issue?
Which filesystem is the base on the server?
Are there any modifications to the cmdline.txt or config.txt?

Knoppix1, 2018-03-25:
@lategoodbye

Me yes I am on Raspbian with firmware basic (4.9)

Again on HDD or internal storage it's the same things.

My cmdline and config

pi@MyBilly:~ $ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=8d333b2b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait rootdelay=3
pi@MyBilly:~ $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

dtoverlay=pi3-disable-bt
dtoverlay=pi3-disable-wifi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=off

mkreisl, 2018-03-25:
> Here some questions for those who can reproduce the issue:
> Are you able to reproduce it with Raspbian?
Yes
> Do you have an external HDD connected via USB during the issue?
No/Yes, does not matter
> Which filesystem is the base on the server?
Doesn't matter, cifs, nfs4 or sshfs share mounted on server
> Are there any modifications to the cmdline.txt or config.txt?
No

PGallagher69, 2018-03-25:
> Here some questions for those who can reproduce the issue:
> Are you able to reproduce it with Raspbian?

Yes... Using the latest Stretch.

> Do you have an external HDD connected via USB during the issue?

Yes, both direct and through a hub.

> Which filesystem is the base on the server?

Filesystem on HD = NTFS. File System on Connecting PC = Windows 10 + Android.

> Are there any modifications to the cmdline.txt or config.txt?

Nope, neither... Tried a few different MTU's with no effect.

lategoodbye, 2018-03-25:
Sorry, i'm not able to reproduce the issue with Raspbian (kernel 4.14.27) and my RPI 3 B+. Yesterday i transfered the uncompressed Raspbian image (~ 4 GB) via sftp and today via NFS. No issues

pelwell, 2018-03-25:
And that is the sticking point. But tomorrow we'll set our team of highly trained engineers on the problem and hopefully make some progress.

PGallagher69, 2018-03-25:
For reference, I'm using Samba...

pelwell, 2018-03-25:
Yes, Samba or cifs does often seem to be a common element.

pelwell, 2018-03-25:
@mkreisl  (nachteule?) Am I right in thinking the current Pi 2/3 XBian release from the Downloads page should run on the 3B+? Version numbers and dates are conspicuously absent.

mkreisl, 2018-03-25:
> @mkreisl (nachteule?) Am I right in thinking the current Pi 2/3 XBian release from the Downloads page should run on the 3B+? Version numbers and dates are conspicuously absent.

Yes the latest release supports Pi3B+

Knoppix1, 2018-03-28:
how to patch that by staying on the official firmware?
Thank for reply

popcornmix, 2018-03-28:
The latest rpi-update kernel has some potential fixes for kernel panics in networking. Can you test?

asavah, 2018-03-28:
I've stumbled into this.
Will duplicate here what I posted on the forums.

Equipment:
pi3b+
ubuntu server, i5 skylake, intel gigabit NIC
managed L2 switch DLINK DGS-1210-10

Software:
own GNU/Linux homemade OS
kernel rpi-4.14.y @ https://github.com/raspberrypi/linux/commit/9d2ad143e40c38d34be86578840499a976c0a5b0
latest firmware at the moment of writing

Symptoms:
RX on pi3 very poor
TX good
`iperf3 -c rpi3plus -t 60` reports hundreds of drops, very poor speed (sorry I don't have bad numbers handy)

What I have tried:

- dtparam=eee=off : no change as eee was already disabled on all switch ports

- force link to 100Mbit FD on the switch: all good , at 100 mbit speed no drops, no retrs in iperf3, nfs file copy speed is around 95mbit/s

- remove forced link speed, **enable flow control** on 3b+'s port on the switch - **GOOD**,
```
iperf3 -c rpi3plus -t 60
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec  1.69 GBytes   242 Mbits/sec    8             sender
[  4]   0.00-60.00  sec  1.69 GBytes   242 Mbits/sec                  receiver
```
```
dd if=/path/to/large/file/on/nfs4/mount/file.iso of=/dev/null iflag=direct bs=4M status=progress
5028970496 bytes (5.0 GB, 4.7 GiB) copied, 172 s, 29.2 MB/s
1200+1 records in
1200+1 records out
5035036672 bytes (5.0 GB, 4.7 GiB) copied, 172.307 s, 29.2 MB/s
```
```
ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:  on
RX:             on
TX:             on
RX negotiated:  on
TX negotiated:  on
```
So flow control on the switch's definitively fixes the situation for me.
Hope this provides some clues.

lategoodbye, 2018-03-29:
The corruption fix 1ad1d52e6cb6a9fcee5d3fb08258b417ffda37fd looks very interesting isn't this necessary for rpi-4.14, too?

6by9, 2018-03-29:
@asavah 
```
ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:  on
RX:             on
TX:             on
RX negotiated:  on
TX negotiated:  on
```
> So flow control on the switch's definitively fixes the situation for me.
> Hope this provides some clues.

If you **don't** enable flow control on your switch does the output of ethtool change?
On my baby unmanaged Netgear switch pause parameters report auto-negotiate and on by default, so I'm just trying to understand is it something that we'd normally expect to be on.

asavah, 2018-03-29:
@6by9 
flow control turned off on the switch:
```
root@rpi3plus:~# ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:	on
RX:		on
TX:		on
RX negotiated:	off
TX negotiated:	off
```

6by9, 2018-03-29:
@asavah Thanks, that helps a lot.
The baby unmanaged switches we do most of our testing through allow flow control by default. Managed switches appear not to. Switching to sending the data over the main managed switches we see the issue.

So the question is why is disabling 802.3x flow control the default in managed switches.
I'm not an expert in networking, but it sounds like without it the Pi will have issues with no real way around it (I'm assuming that if the Pi sends pause frames when the switch has disabled 802.3x then they'll be ignored. @JamesH65 is looking to see if we can send them if not negotiated).

[133 more replies]

closed 2018-11-13 by pelwell