raspberrypi/linux #2453: wlan freezes in raspberry pi 3+

flok99, 2018-03-19:
(see also https://github.com/raspberrypi/linux/issues/1342 )

I've also got that problem with wifi dying.

```
Mar 17 18:25:28 hassass kernel: [10279.186321] brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012
Mar 17 18:25:30 hassass kernel: [10281.665090] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Mar 17 18:25:30 hassass kernel: [10281.665622] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Mar 17 18:25:30 hassass kernel: [10281.665638] brcmfmac: brcmf_run_escan: error (-110)
Mar 17 18:25:30 hassass kernel: [10281.665647] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
Mar 17 18:26:30 hassass kernel: [10341.665866] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
```

This is with 4.14.27-v7+ and with
/sbin/iw dev wlan0 set power_save off
/sbin/ifconfig wlan0 promisc
in /etc/rc.local.

```
[    4.112717] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.119827] brcmfmac: brcmf_fw_map_chip_to_name: using
brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[    4.120314] usbcore: registered new interface driver brcmfmac
[    4.440371] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb
27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.440958] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2
Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09
18:56:28
[   10.911757] brcmfmac: power management disabled
[   12.016088] brcmfmac: power management disabled
[ 2074.090674] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg
failed w/status -5
[ 2074.090687] brcmfmac: brcmf_cfg80211_get_tx_power: error (-5)
[ 2074.090745] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2074.090753] brcmfmac: brcmf_link_down: WLC_DISASSOC failed (-5)
[ 2074.610583] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2074.611992] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2074.613945] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2074.613971] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-5)
[ 2074.729716] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2074.729733] brcmfmac: brcmf_cfg80211_reg_notifier: Country code iovar
returned err = -5
[ 2074.871693] usbcore: deregistering interface driver brcmfmac
[ 2074.929084] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 2074.936897] brcmfmac: brcmf_fw_map_chip_to_name: using
brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[ 2074.937139] usbcore: registered new interface driver brcmfmac
[ 2075.118180] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb
27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[ 2075.118706] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2
Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09
18:56:28
[ 2075.215365] brcmfmac: power management disabled
[ 2075.263751] brcmfmac: power management disabled
[ 2085.475001] brcmfmac: power management disabled
[ 2124.380808] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2124.381146] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2124.381156] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-5)
[ 2124.622345] usbcore: deregistering interface driver brcmfmac
[ 2124.705432] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 2124.714194] brcmfmac: brcmf_fw_map_chip_to_name: using
brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[ 2124.716213] usbcore: registered new interface driver brcmfmac
[... 86 lines cut ...]
brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[ 8374.771137] usbcore: registered new interface driver brcmfmac
[ 8377.411255] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 8377.431924] brcmfmac: brcmf_c_get_clm_name: retrieving revision info
failed (-110)
[ 8377.431934] brcmfmac: brcmf_c_process_clm_blob: get CLM blob file name
failed (-110)
[ 8377.431941] brcmfmac: brcmf_c_preinit_dcmds: download CLM blob file
failed, -110
[ 8377.431949] brcmfmac: brcmf_bus_started: failed: -110
[ 8377.432003] brcmfmac: brcmf_sdio_firmware_callback: dongle is not
responding
[ 8424.133114] usbcore: deregistering interface driver brcmfmac
[ 8424.229631] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 8424.237210] brcmfmac: brcmf_fw_map_chip_to_name: using
brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[ 8424.239352] usbcore: registered new interface driver brcmfmac
[ 8424.460736] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb
27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[ 8424.461174] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2
Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09
18:56:28
[ 8424.646993] brcmfmac: power management disabled
[ 8424.708633] brcmfmac: power management disabled
```

digisaster, 2018-03-21:
Same problem here.

If i do a:
sudo BRANCH=next rpi-update
Then i have a working situation.
After this i got:
uname -a
Linux raspberrypi 4.14.17-v7+ #1090 SMP Mon Feb 5 21:02:18 GMT 2018 armv7l GNU/Linux

So i guess its kernel related.

JamesH65, 2018-03-21:
Interesting. So not in the current rpi-update, but  in the next branch? Should make the particular change a bit easier to find.

JamesH65, 2018-04-23:
Not had a chance to look yet. How do you provoke the problem? We've not had many reports on B+ Wifi failing in this way, so apparently it's unusual. Have you updated to the very latest 3.14 kernel? Does that make any difference?

JamesH65, 2018-04-23:
Once I have more details on how to replicate the issues, I can send the data to Cypress for investigation. The mailbox error, IIRC, is a firmware crash, so its not something we can really deal with here, since we do not have access to the firmware.

llamasoft, 2018-05-25:
It may be slightly off topic because it's a different distribution, but it's pretty easy to trigger this error when using [Kali Pi](https://whitedome.com.au/re4son/kali-pi/).  
Putting the device into monitor mode (using `mon0up` in Kali-Pi) and running `aireplay-ng --test` causes it to emit the `Unknown mailbox data content: 0x40012` error almost immediately.  From then on, the wifi is worthless until you reboot.

For reference, `mon0up` is a [short shell script](https://github.com/Re4son/Kali-Pi/blob/master/Menus/RAS-AP/mon0up) that runs `iw phy phy0 interface add mon0 type monitor` and `ifconfig mon0 up` and displays some brief info.

cdouglas97, 2018-06-05:
On a project I am working on, I am getting this constantly with Kali Pi when in monitor mode.  Unloading and reloading the driver works sometimes and sometimes not.  Sometimes it happens after 5 seconds, sometimes after 5 minutes, very random.   Verified its not a hardware problem since I have 2 Pi's and both do it. I also loaded Raspbian latest and installed latest Nextmon drivers and I get the exact same thing,

JamesH65, 2018-06-06:
Do you get any errors listed in dmesg?

cdouglas97, 2018-06-06:
Yes I do.

The following is while running the following command, The Set Channel failed aren’t really errors it seems since I am scanning a range

/usr/sbin/airodump-ng -C 2412-5825 --write-interval 10 --write test --output-format netxml wlan0mon

Working until this point……..
[78833.305809] brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012
[78835.790416] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78835.797410] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78835.803824] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53409, -110
[78838.590434] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78838.597330] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78838.603733] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53413, -110
[78841.390457] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78841.397407] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78841.403867] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4098, -110
[78844.190483] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78844.197466] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78844.204088] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4102, -110
[78846.990502] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78846.997678] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78847.004601] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4106, -110
[78849.790519] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78849.797885] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78849.804936] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -110
[78852.590543] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78852.597968] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78852.605103] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53284, -110
[78855.390563] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78855.398051] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78855.405211] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53288, -110
[78858.190583] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78858.198319] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[78858.205591] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=53292, -110
[78860.990609] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[78861.001145] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
…….

After shutting down airodump, I detected that it wasn’t capturing anything and attempted to unload at 79803.439351 and reload the driver at 79923.711128 and that failed.
I have a 120sec timer between unload and reload of driver

[79782.517551] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[79782.529849] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[79782.541559] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[79785.157565] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[79785.169936] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[79785.181530] brcmfmac: _brcmf_set_multicast_list: Setting mcast_list failed, -110
[79787.717590] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[79787.729703] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[79792.837633] brcmfmac: _brcmf_set_multicast_list: Setting allmulti failed, -110
[79797.957685] brcmfmac: brcmf_cfg80211_del_ap_iface: interface_remove failed -110
[79800.517694] brcmfmac: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
[79803.078132] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[79803.089706] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-5)
[79803.187727] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[79803.199528] brcmfmac: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[79803.210553] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-5)
[79803.439351] usbcore: deregistering interface driver brcmfmac
[79923.696268] brcmfmac: F1 signature read @0x18000000=0x15264345
[79923.700624] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[79923.711128] usbcore: registered new interface driver brcmfmac
[79926.358647] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[79926.389146] brcmfmac: brcmf_c_get_clm_name: retrieving revision info failed (-110)
[79926.400840] brcmfmac: brcmf_c_process_clm_blob: get CLM blob file name failed (-110)
[79926.412661] brcmfmac: brcmf_c_preinit_dcmds: download CLM blob file failed, -110
[79926.423176] brcmfmac: brcmf_bus_started: failed: -110
[79926.431387] brcmfmac: brcmf_sdio_firmware_callback: dongle is not responding

…  Tried unload and reload a little while later and it worked

[80685.169424] usbcore: deregistering interface driver brcmfmac
[80805.455360] brcmfmac: F1 signature read @0x18000000=0x15264345
[80805.460216] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[80805.471390] usbcore: registered new interface driver brcmfmac
[80805.805728] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Apr 10 2018 18:33:56 version 7.45.154 (nexmon.org: 2.2.2-178-gd64f-1) FWID 01-4fbe0b04
[80805.814840] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28

Click here<https://www.mailcontrol.com/sr/2xffoH0VKMPGX2PQPOmvUsgYVF5ojvBykkW9UzADy!8LKeLB79zZKi4csdmxhkuDqLqNh6I9BzeTKOxjbsImtQ==> to report this email as spam.

Nemesis7, 2018-06-09:
Any news about this? I'm having this same issue. From dmesg:
`[    4.584121] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[    4.868470] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.869048] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28 
 I wish I never executed the rpi-update.`

I finally got the Raspberry Pi 3 B+ working as an AP + Managed Wifi at 5GHz, but now the AP doesn't work anymore after the update. How can I downgrade the brcmfmac to a stable/working version?

gdb-power, 2018-06-10:
After "Unknown mailbox data content: 0x40012" is received,
sometimes the communication can be recovered with the following commands:
```
modprobe -r brcmfmac
modprobe brcmfmac
```
Sometimes it doesn't recover even after modprobe -r (device is stuck).
In that case, the following heavy-handed commands will fix the communication with the wifi device:

```
echo -n "3f300000.mmc" > /sys/devices/platform/soc/3f300000.mmc/driver/unbind
sleep 1
echo -n "3f300000.mmc" > /sys/bus/platform/drivers/mmc-bcm2835/bind
```
technical comment:
rebinding the driver will call probe(), 
which will call bcm2835_reset_internal(), 
which will power-cycle the SDIO device (SDVDD_POWER_OFF),
which will properly reset the wedged WIFI SDIO device. Phew!

cdouglas97, 2018-06-10:
I’ll try it Monday and let you know. Thank you very much.

pelwell, 2018-06-10:
@Nemesis7 You can revert to any previous firmware+kernel package using rpi-update by putting the hash (string of hexadecimal digits) on the command line. The hashes can be found on the right hand side of the list of commits (releases): https://github.com/Hexxeh/rpi-firmware/commits/master

Alternatively you could return to the standard Raspbian kernel using:
```
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
```

cdouglas97, 2018-06-11:
The heavy handed method seems to be working to get the driver working again, but is there a chance of this being fixed so its not necessary?  Is this a hardware or firmware issue?  Thank you

JamesH65, 2018-06-12:
@cdouglas97 We currently do not know the cause of this. The mailbox error is the result of the firmware on the wireless chip dying, but the cause of that is unclear. We do NOT have access to the firmware source, that is provided as a binary by Cypress so we do rely on them to fix firmware issues. 

I'm not looking at it at the moment, I have some other stuff to clear first. However, if anyone has a clear set of steps to replicate the problem, that would be very useful once I do start to look at it.

cdouglas97, 2018-06-12:
Thank you.  I can reproduce it very easily.    Steps:

wlan0 is not up, only active interface is eth0

# Use Airmon-ng to create monitor port:
/usr/sbin/airmon-ng start wlan0

# Run airodump that scans all 2.4 to 5.8Ghz frequencies
/usr/sbin/airodump-ng -C 2412-5825 --write-interval 10 --write OUTPUTME --output-format netxml wlan0mon

It usually happens within 30 seconds to 5 minutes.  I run the airodump for 60 seconds at a time with a 15 min gap between runs and then kill it so sometimes it completes its run and sometimes not.   I added the heavy-handed commands before I start and it didn't seem to make a difference on how long it took for it  to die. It just fixed the firmware if it died during a previous run.

-Chris

JamesH65, 2018-06-12:
@cdouglas97 I am not familiar with airmon/airodump. What package do I need to install in Raspbian to get those?

pelwell, 2018-06-12:
`sudo apt-get install aircrack-ng`

pelwell, 2018-06-12:
With wlan0 available but not associated with an AP I get:
```
pi@raspberrypi:~ $ sudo airmon-ng start wlan0

Found 4 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to run 'airmon-ng check kill'

  PID Name
  319 avahi-daemon
  353 dhcpcd
  364 avahi-daemon
  400 wpa_supplicant

PHY	Interface	Driver		Chipset

phy0	wlan0		brcmfmac	Broadcom 43430

ERROR adding monitor mode interface: command failed: Operation not supported (-95)
```
Any suggestions?

cdouglas97, 2018-06-12:
pelwell, the default Rasbian latest install firmware doesn't allow monitoring, have to use this:  https://github.com/seemoo-lab/nexmon to make it work. Kali uses this and I took a Raspbian image a built patch as directed and both get same result.

JamesH65, 2018-06-12:
Er, so the way to make this go wrong is to install some random third party stuff that plays around with the firmware in a way that probably wasn't intended by the developers? Why am I not surprised this might go wrong?

Unless this goes wrong with our standard firmware, I'm not sure we should spend any more time on this.

cdouglas97, 2018-06-12:
I don't blame you, I thought that it was understood from the first post that said he was putting into promiscuous mode that's what we were doing.  Unless was are supposed to be able to put default firmware into monitor mode making the nexmon unnecessary.

Thanks for your time and the command to force the wifi reload.

-Chris

pelwell, 2018-06-12:
I don't know why you would think it should have been understood - promiscuous mode is not the same as monitor mode.

cdouglas97, 2018-06-12:
The OP might not be doing the same thing as I am with the nexmon fw since he is just trying promiscuous mode so it still might be something you can investigate for him.

gdb-power, 2018-06-13:
@pelwell @JamesH65 please don't give up on all of us who use the regular firmware... It is a real problem.

In the following conditions, the firmware crashes several times per day:

1. Raspberry PI running in an environment with alot a *diverse* wifi packets, such as a busy train station or a busy mall. Not a "regular" office environment or an shielded RF room...

2. Raspberry PI running as hotspot (thus continuously listening to incoming packets).

Note: the raspberry PI wifi crashes randomly in *any* environment and also when running just as station (not AP). I'm just describing the conditions that *increase* the probability of crashing.

pelwell, 2018-06-13:
Nobody's giving up. We have a number of difficult problems ongoing, and the Ethernet stalls are currently getting most of the attention, but now we have what should be a low-impact workaround for that the spotlight will turn to this issue (which looks suspiciously like an old Pi3B problem).

gdb-power, 2018-06-13:
Another tip: when the firmware crashes, you can collect internal firmware stack traces here (you may need to compile the driver with DEBUG enabled):

> cat /sys/kernel/debug/brcmfmac/mmc1:0001:1/forensics

In a busy train station/mall/office, when there are alot of people, you can sometimes catch several crashes per hour.

In an ideal world, broadcom/cypress would release the firmware source code so the community would be able to fix it, just like qualcomm released their internal wifi firmware. There are no big wifi secrets, all wifi firmware is pretty trivial:

https://github.com/qca/open-ath9k-htc-firmware

llamasoft, 2018-06-13:
@gdb-power - Your comment gives me an idea.  If anybody has a second device with a monitor enabled WiFi adapter, you may be able to capture the series of frames that causes the firmware to lock up.  The experiment would be something like:

1. Set up the monitoring device running `airodump-ng` to capture all frames on the Raspberry Pi's channel.
2. Wait for the Raspberry Pi's WiFi to lock up, then terminate the capture and reboot the Raspberry Pi.
3. Bisect the captured frames and replay them with `aireplay-ng` until the triggering frames are isolated.

Of course, this assumes the hang is caused by *incoming* frames.  If it's caused by invalid *outgoing* frames... I'm not sure how to capture that.

gdb-power, 2018-06-13:
Fuzzing the cypress firmware for crashes is pretty trivial, several groups have done it in the past (including Google project zero). In order to *fix* the firmware, Broadcom/cypress should be doing the fuzzing, since they're the only one with access to the source code.

pelwell, 2018-06-13:
@gdb-power If you have any forensics dumps then we'd like to see them.

@llamasoft That sounds like a plan. The issue is likely to be triggered by packet reception - transmission is massively simpler because the driver gets to determine the timing and packet content.

cdouglas97, 2018-06-13:
I know I am out of this since I am using a patched driver to support monitoring, but this might help.  All I am doing is running airodump-ng to scan for access points and it crashes.  There are 42 in my vicinity that I usually detect.

JamesH65, 2018-06-13:
As with all of these ethernet issues, replication is the crux. We need simple (ish) ways of making the problem happen, preferably in a situation where we can have debuggers and diagnostics tools attached (which makes a trip out of the office a real issue). The recent 3B+ issue suddenly got easier to solve when we had a user who could replicate at will and was able to help, and when I finally managed to be able to cause it on demand. So if ANYONE has a guaranteed way of causing this mailbox error that would be very useful. 

On the 3B this mailbox error has been around since launch, but I thought it had been fixed with the most recent firmware upgrade. Certainly until this thread it hadn't been recently reported. Cypress actually closed the case, but it looks like it will need to be reopened, once we can replicate it reliably.

pelwell, 2018-06-13:
Yes, it's possible that the 43438 fix will also apply to the 43455.

cdouglas97, 2018-06-13:
I’ll help but I don’t know if I am a valid test since I am using patched firmware.  I replicate it every 15 minutes like clockwork.

Nemesis7, 2018-06-14:
I am totally new to this but I hope my input can help you solve this issue (even without fixing the firmware).
I am encountering this problem when I try to setup a managed WiFi with AP and ip forwarding on a Raspberry Pi 3 B+. The first time around I followed the guide, which seemed to work: https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/, eventually it failed, I think it's because I didn't disable dhcpcd which didn't play nice with /etc/network/interfaces

Now I've done what is described here: https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=138730&start=125#p1321390

It works pretty awesome until the AP is activated, then I lose the WiFi connection and I get the mailbox error preceded by:
`brcmfmac brcmf_link_down wlc_disassoc failed (-11)`

Is there a firmware/kernel version I can try to see if it works there?

gdb-power, 2018-06-14:
@pelwell @JamesH65 I sent you forensics dump by email.

pelwell, 2018-06-14:
Yes, thank you. We are sending them direct to Cypress who may then have enough information to fix the crashes.

gdb-power, 2018-06-14:
Also, it might be a good idea to enable brcmfmac DEBUG flag in the raspberry pi default kernel, so that anyone can send forensic dumps.

JamesH65, 2018-06-15:
The Cypress case is live, and they have assigned 'collaborators' to it, so hopefully we will hear something soon.

Nemesis7, 2018-06-15:
Guys, for me the crash stopped happening when I changed the order in which I put interfaces live: the wlan0 must be disabled, then enable the ap0, then enabled wlan0. In another sequence, it will crash.

brubbel, 2018-07-04:
@Nemesis7. Yes, but when wlan0 associates with an AP (e.g. using wpa_supplicant), traffic over ap0 (e.g. using hostapd) becomes slow/unreliable after some time. This does not happen as long as wlan0 is not used.

JamesH65, 2018-07-13:
OK, we have some beta software from Cypress, that may or may no help with this. 
The files on the following two links need to be copied to the /lib/firmware/brcm folder on the Pi. Note, there do appear to be some error messages reported when the driver first starts up, but doesn't seem to affect usage, however, I recommend backing up the two original files first. 

https://drive.google.com/file/d/1bqugahKmfz1uQe8u5VHijAUnuZTxGkvG/view?usp=sharing
https://drive.google.com/file/d/1mbfEOMShLrul-qprmlcPSuERdCTdh4e7/view?usp=sharing

brubbel, 2018-07-15:
No success: driver fails to load.
[image]

Steps undertaken:
1. brcmfmac43455 is for 3B+ only (CYW43455): /proc/cpuinfo says rev=a020d3
2. sudo dd if=2018-06-27-raspbian-stretch-lite.img of=/dev/sdb bs=4M
3. cp brcmfmac43455-sdio.* /lib/firmware/brcm/
4. sha256sum /lib/firmware/brcm/brcmfmac43455-sdio.*
```
644b1afe735232a1b0c447e6f80650a9992f6977b80dc1d468c7302c769aa5d5  /lib/firmware/brcm/brcmfmac43455-sdio.bin
635bdcbf9dc2cf7dd3bb72480566f347966e95f3deb2fdb5615a4001c7dd2e77  /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
15698c62457bcf25e60d063e6c666d6e1b7dacdf2b03e6d14ebbc619de6da6b7  /lib/firmware/brcm/brcmfmac43455-sdio.txt
```
5. halt + powercycle: new driver fails
6. sh -c "echo options brcmfmac debug=0x100000 > /etc/modprobe.d/brcmfmac.conf"
7. apt update; apt upgrade + halt + powercycle: new driver fails
8. rpi-update + halt + powercycle: new driver fails

9. modprobe -r brcmfmac
10. sudo echo -n "3f300000.mmc" > /sys/devices/platform/soc/3f300000.mmc/driver/unbind
11. echo -n "3f300000.mmc" > /sys/bus/platform/drivers/mmc-bcm2835/bind
12. modprobe brcmfmac (fails + stack trace)

13. copy original driver to /lib/firmware/brcm + powercycle: old driver ok. 
14 repeat steps 9-12: old driver ok.

gdb-power, 2018-07-15:
Same here - driver crashes and burns. No wifi interface.

```
[    3.987473] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Jun 20 2018 20:26:28 version 7.45.165 (r692055 CY) FWID 01-1de59a68
[    3.988042] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.116 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-06-20 20:12:36 
[    4.965242] uart-pl011 3f201000.serial: no DMA platform data
[    7.751431] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[    7.751442] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   10.311449] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   10.311462] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   11.465157] Bluetooth: Core ver 2.22
[   11.465238] NET: Registered protocol family 31
[   11.465244] Bluetooth: HCI device and connection manager initialized
[   11.465263] Bluetooth: HCI socket layer initialized
[   11.465277] Bluetooth: L2CAP socket layer initialized
[   11.465311] Bluetooth: SCO socket layer initialized
[   11.479666] Bluetooth: HCI UART driver ver 2.3
[   11.479681] Bluetooth: HCI UART protocol H4 registered
[   11.479687] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.479884] Bluetooth: HCI UART protocol Broadcom registered
[   11.655598] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.655606] Bluetooth: BNEP filters: protocol multicast
[   11.655623] Bluetooth: BNEP socket layer initialized
[   12.871437] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   12.871450] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   15.431426] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   15.431437] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   17.991448] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   23.111431] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   25.671431] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[   28.231434] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   28.231446] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   30.791428] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   30.791440] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   33.351436] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   38.471439] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   41.031441] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211

```

JamesH65, 2018-07-15:
Hmm, although I get those messages, the WiFi interface does come up. I'll
report back to cypress.

brubbel, 2018-07-15:
As a side-note, Firmware version 7.45.154 (Feb 27 2018) has a discrepancy between the powerstate reported by the driver and the powerstate reported by iwconfig. I suppose that it is somewhat related to [this commit](https://github.com/raspberrypi/firmware/commit/8b4e5482b52e6fb438dddc0d88ba0ba8d44af54b) which disabled powersave, but it confusing. Anyway, I understand it is always off.

`$ iw wlan0 set power_save **on**`
dmesg says: brcmfmac: power management **disabled**
iwconfig says: Power Management:**on**

`$ iw wlan0 set power_save **off**`
dmesg says: brcmfmac: power management **disabled**
iwconfig says: Power Management:**off**

brubbel, 2018-07-15:
> So if ANYONE has a guaranteed way of causing this mailbox error that would be very useful.

1. Start from the setup in my post above: 2018-06-27-raspbian-stretch-lite and rpi 3B+ rev=a020d3
2. remove /lib/dhcpcd/dhcpcd-hooks/10-wpa-supplicant to prevent wpa_supplicant from starting + reboot
3. apt install hostapd
3. $ iw dev wlan0 interface add uap0 type __ap
4. terminal 1: $ wpa_supplicant -dd -iwlan0 -c /home/pi/wpa_supplicant.conf
<wait until connected with your wifi network>
5. terminal 2: $ hostapd /home/pi/hostapd.conf
6. terminal 3: $ dmesg -ew
```
[   79.680414] brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012
[   82.154516] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   82.155047] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   84.714914] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   84.715403] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   84.715417] brcmfmac: brcmf_c_set_joinpref_default: Set join_pref error (-110)
[   87.154353] brcmfmac: brcmf_cfg80211_connect: BRCMF_C_SET_SSID failed (-110)
[   89.675516] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   89.676054] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
```
/home/pi/wpa_supplicant.conf:
```
ap_scan=1

ctrl_interface=/var/run/wpa_supplicant
network={
    ssid="E2000"
    scan_ssid=1
    proto=WPA RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP TKIP
    group=CCMP TKIP
    psk="myrouterpassword"
}
```

/home/pi/hostapd.conf:
```
interface=uap0
ssid=raspberrypi
hw_mode=g
channel=6
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=myhostapdpassword
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_pairwise=TKIP
ap_isolate=1
```

JamesH65, 2018-07-16:
I've passed on the details of the issues with this test firmware to Cypress. Interstingly, although this is new and in test from our point of view, I believe it's actually their top of tree build, so I would not be expecting issues this clear to be turning up. Most strange. Anyway, we now wait for Cypress.

brubbel, 2018-07-16:
More info.
Compiling hostapd/wpa_supplicant from [source](https://salsa.debian.org/debian/wpa.git) with the [same version](https://salsa.debian.org/debian/wpa/commit/57c9d49f61e45b2d82f614105aff2f11e61e3bed) as indicated in the wpasupplicant source package (wpa 2:2.4-1+deb9u1) stops the immediate crashes, in any order of starting hostapd and wpa_supplicant, multiple times.
Also, compiling upstream (https://w1.fi/cvs.html) does not show the crashes.

So, what is the difference with a clean 2018-06-27-raspbian-stretch-lite installation?

gdb-power, 2018-07-17:
@brubbel the crash you're investigating is occurring at firmware instruction pointer = 0x00021e2c (somewhere in ROM). 

There are also firmware crashes at other locations that cause "Unknown mailbox data content: 0x40012", those occur _during normal operation_. I've collected a few of these crashlogs during 2 days of operation, they all seem to be in the RAM region of the firmware (0x180000-0x25ffff, more or less):

```
$ grep -a TRAP forensics.log | sort | uniq -c
      2 TRAP 4(25fd10): pc 1a4bdc, lr 1a4bcb, sp 25fd68, cpsr 19f, spsr 1bf
      1 TRAP 4(25fd78): pc 19ecb6, lr 19ecaf, sp 25fdd0, cpsr 2000019f, spsr 200001bf
      7 TRAP 4(25fd78): pc 19ecc2, lr 19ecaf, sp 25fdd0, cpsr 2000019f, spsr 200001bf
      1 TRAP 4(25fdc0): pc 19ed20, lr 1a2545, sp 25fe18, cpsr 2000019f, spsr 200001bf
      3 TRAP 4(25fdc0): pc 19ed2c, lr 1a2545, sp 25fe18, cpsr 6000019f, spsr 600001bf
     59 TRAP 4(25fdc8): pc 1a246a, lr 1a246b, sp 25fe20, cpsr 6000019f, spsr 600001bf
```

Those are probably buggy information-element parsing functions, as described below, but without firmware source code we can't really fix them:
https://googleprojectzero.blogspot.com/2017/10/over-air-vol-2-pt-2-exploiting-wi-fi.html

brubbel, 2018-07-17:
@gdb-power Correct, I also noticed that the stack pointer had different addresses, but I had thought this was due to loading the firmware at random locations in memory.

brubbel, 2018-07-23:
Concerning the firmware crash at ROM location 0x00021e2c for CYW43455, RPI3B+, firmware driver brcmfmac43455-sdio.bin (Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04)

## Bringing the AP up.
The following line in the [brcmfmac driver](https://patchwork.kernel.org/patch/2115341/) is most likely the culprit.
`brcmf_fil_iovar_int_set(ifp, "apsta", 0);`
When the AP is started, this command is issued to the ap0 interface (ifidx=1), but it scares the hell out of the CYW43455 RTOS when interface with ifidx=0 is already in associated state.
This command is issued when RSDB (2.4/5GHz simultanuous dual band) is not supported by the device.

As ap0 is already in AP mode (iw dev wlan0 interface add ap0 type __ap), I removed it and have now a working AP+STA on RPI3B+ !

## Bringing the AP down.
Secondly, when the AP (ifidx=1) is stopped while STA (ifidx=0) is associated, the latter always looses it's connection. This is caused by the command ([ref](https://patchwork.kernel.org/patch/5433011/) and [ref](https://github.com/torvalds/linux/blob/a44aa4001a86d46f936ca449e5d6c268446bfae2/drivers/net/wireless/brcm80211/brcmfmac/cfg80211.c#L4125)):
`brcmf_fil_cmd_int_set(ifp, BRCMF_C_DOWN, 1);`
which is related to a patch to support multiple AP's. 
However, let's first try to get 1 AP working, so remove this line, then recompile the brcmfmac kernel module and the STA stays up when AP goes down.

Changes [here] https://github.com/brubbel/linux/commit/7a762f188308642e64fd57d270cbd99b65a0310c)
Module [here](https://drive.google.com/file/d/1S6xMYZw735NgEhaROQbYX4hKomgkmxrm/view?usp=sharing), goes into /lib/modules/$(uname -r)/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko

brubbel, 2018-07-25:
More findings (still related to CYW43455, RPI3B+ as above).

Consider wpa_supplicant configured with (1) ap_scan=1 and (2) a valid network.
When this network goes out of range, wpa_supplicant starts complaining every few seconds:
`wlan0: Failed to initiate sched scan`
and /sys/kernel/debug/brcmfmac/mmc1:0001:1/forensics:
`pfn_gscan_cfg BCME -23 (Unsupported)`

From [this patch](https://github.com/torvalds/linux/commit/f957dd3c8db2781c8a334b166800788feb618625), it seems that pfn_gscan_cfg is disabled for bcm4345* devices (id=17221), however this is not the case, because [brcmf_feat_iovar_int_get(ifp, BRCMF_FEAT_PNO, "pfn");](https://github.com/torvalds/linux/blob/1c8c5a9d38f607c0b6fd12c91cbe1a4418762a21/drivers/net/wireless/broadcom/brcm80211/brcmfmac/feature.c#L218) loads the PNO scan feature anyway.

Including the BRCMF_FEAT_PNO line in the blacklist for bcm4345* fixes this. wpa_supplicant background scans now work properly.
Because powersave is disabled by default on raspbian, PNO doesn't really add value.

brubbel, 2018-07-26:
Question: is it possible to unbind/bind the 3f300000.mmc on the RPI3B?
It results in the following (due to wrong mmc clock? Is the only way a custom dtoverlay?)
```
mmc1: bad CIS tuple 0x20 (0 bytes)
mmc1: error -22 whilst initialising SDIO card
```
Thanks!

gdb-power, 2018-07-26:
The unbind/bind trick should work on all Raspberries.
You just need to find the correct mmc address.

On Pi 3B+ it's 3f300000.mmc
On Pi Zero W it's 20300000.mmc
For other Pis, use this command to find the correct mmc address:
`sudo ls -la /sys/bus/platform/drivers/mmc-bcm2835`

brubbel, 2018-07-26:
Alas, RPI3B does unbind, but no rebind... (3B+ works)

brubbel, 2018-08-02:
Still the same error for RPI3B (rev. a02082).

cyph84, 2018-08-17:
In my case, I have run over 100+ RPi 3 as open access points about a year and they are fine. However, protecting the APs with WPA-PSK, we started seeing this error after a couple of days.

Amit0301, 2018-08-28:
In my case I tested on 3 different RPI3B+ with same SD card, working on 2 showing following error on 3rd. 
 
[   12.779953] fuse init (API version 7.26)
[   19.657576] brcmfmac: brcmf_sdiod_regrw_helper: failed to read data F1@0x08000, err: -110
[   19.657592] brcmfmac: brcmf_chip_recognition: chip backplane type 15 is not supported
[   19.657601] brcmfmac: brcmf_sdio_probe_attach: brcmf_chip_attach failed!
[   19.657610] brcmfmac: brcmf_sdio_probe: brcmf_sdio_probe_attach failed
[   19.657823] brcmfmac: brcmf_ops_sdio_probe: F2 error, probe failed -19...
[   19.658194] usbcore: registered new interface driver brcmfmac

Any idea?

ndarilek, 2018-08-28:
Has anyone started from a known working firmware, then advanced commit by commit until they found a buggy one? Would such an analysis be helpful?

In my case, I don't know if I'm experiencing the same issue as documented here or something new. The latest kernel/wifi drivers fail when used as an access point. Sometimes there are dmesg errors, sometimes not. Stations just get dissociated and can't reconnect unless I either restart hostapd or perform some action that seems to kick the internal wifi back into operation, like scans or listing stations. My solution has been using rpi-update to downgrade to the firmware/kernel combination released on 2018-04-16, which has been rock solid.

I'd be happy to advance through each firmware commit every few days, then report when I've found the one where the errors begin, but only if that would actually be helpful. Otherwise, I'm sticking with 2018-04-16. :)

FWIW, this happens for me on both the 3B and 3B+. So I'm not sure if my issue is in fact this one, but it isn't a misconfiguration on my end since the exact same card behaves the same in both models, then the issues go away entirely if I downgrade firmware. Finding anything on this issue has been tough, since everything out there just tells me to disable power management, which it seems is always off these days anyway (though 2018-04-16 did exhibit some lockups until I disabled power management. It may *technically* always be off, but typing the command did something on the firmware level that made things in that version more stable.)

pelwell, 2018-08-28:
If you can point to a specific firmware (by which I mean an rpi-update start.elf + kernel + DTBs) release that breaks AP then that would be very helpful.

[144 more replies]