raspberrypi/linux #2454: PI3B+: Playback from network share via eth0 results in garbled sound

volumio, 2018-03-20:
With kernel 4.14.26-v7+, playing flac files from SMB shares via ethernet to USB DAC results in garbled sound (dropping packets?).
The very same files, on the very same configuration play spotless via wlan0. 
Probably the same root cause as https://github.com/raspberrypi/linux/issues/2449

I'm available for testing

popcornmix, 2018-03-20:
Does it work on a Pi3 using 4.14.26-v7+?

volumio, 2018-03-20:
Yes, works flawless

popcornmix, 2018-03-20:
@P33M any ideas?

P33M, 2018-03-20:
Can you give a waveform capture?

Make a FLAC file with a low-frequency sinewave - 440Hz or so - play it via the USB DAC - and capture a .wav file using line-in on a PC/laptop?

volumio, 2018-03-20:
Here are the test results:
Created 2 WAV files: 32\192khz and 32\384khz with a sine wave of 440Hz.

Results:
Wifi -> USB DAC = OK
Wifi -> Analog out = OK
LAN -> USB DAC = Issues
LAN -> Analog out = OK

Those are the captures of both files in the LAN to USB DAC situation:
http://repo.volumio.org/Volumio2/tests-pi3b%2B/test-440-192-lan-usb.wav
http://repo.volumio.org/Volumio2/tests-pi3b%2B/test-440-384-lan-usb.wav

With LAN I mean that the files are stored on a NAS and shared via SMB, the PI is conneted to network with gigabit ethernet.

If it can be useful, see the final part of the article for a bit of context:
https://volumio.org/raspberry-pi-3-b-plus-audio-review/

P33M, 2018-03-20:
As a test, what happens if you play the FLAC file from SD storage (i.e. locally) and then run an iperf network bandwidth test? I would expect it to be garbled in the same manner.

volumio, 2018-03-20:
No, running iperf while playing the 384\32 bit wav  from SD Card produces no garble whatsoever.
The command I am launching is 
 iperf -c IPADDR

volumio, 2018-03-23:
A little bit better with 4.19.29 but still get dropouts

P33M, 2018-03-26:
Quite probably related: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=208512&start=50#p1292589

It does not appear to be specific to audio - RX packet drops/line drops can be seen with e.g. rsync.

volumio, 2018-05-16:
@P33M sure, I'll let you know asap

volumio, 2018-05-17:
@P33M Confirm that I do not hear any glitches now, when playing via eth.
Seems to be fixed then

JamesH65, 2018-06-29:
Closing this issue as questions answered/issue resolved.

closed 2018-06-29 by JamesH65