raspberrypi/linux #2457: 4.14 / 3+: oops in lan78xx_set_wol

maxnet, 2018-03-21:
OpenELEC and clones try to enable WoL with an udev rule ( https://github.com/OpenELEC/OpenELEC.tv/blob/master/packages/network/ethtool/udev.d/99-wakeup-eth.rules )

This regulary -but not always- results in an Oops on 3+:

```
[   18.647311] Unable to handle kernel NULL pointer dereference at virtual address 000001b0
[   18.655625] pgd = b1a54000
[   18.658395] [000001b0] *pgd=2efde835, *pte=00000000, *ppte=00000000
[   18.664870] Internal error: Oops: 17 [#1] SMP ARM
[   18.669669] Modules linked in: fuse ipv6 sch_fq_codel brcmfmac brcmutil snd_bcm2835(C) cfg80211 rfkill snd_pcm snd_timer snd joydev uio_pdrv_genirq uio fixed
[   18.684060] CPU: 3 PID: 312 Comm: ethtool Tainted: G         C      4.14.26v7-aufs #1
[   18.684062] Hardware name: BCM2835
[   18.684067] task: b0098f00 task.stack: ae5ee000
[   18.684082] PC is at phy_ethtool_set_wol+0x14/0x3c
[   18.684090] LR is at lan78xx_set_wol+0xd4/0xe4
[   18.684093] pc : [<806c5610>]    lr : [<806cd71c>]    psr: 00070013
[   18.684096] sp : ae5efdd0  ip : ae5efde0  fp : ae5efddc
[   18.684098] r10: 00000000  r9 : 00000040  r8 : 00114809
[   18.684101] r7 : 00000000  r6 : ae5efe28  r5 : aeebe000  r4 : ae461800
[   18.684104] r3 : 00000001  r2 : aee33c00  r1 : ae5efe28  r0 : 00000000
[   18.684110] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   18.684112] Control: 10c5383d  Table: 31a5406a  DAC: 00000055
[   18.684116] Process ethtool (pid: 312, stack limit = 0xae5ee210)
[   18.684122] Stack: (0xae5efdd0 to 0xae5f0000)
[   18.684131] fdc0:                                     ae5efdfc ae5efde0 806cd71c 806c5608
[   18.684137] fde0: 7ebbc400 8095bbe4 7ebbca7c aeebe000 ae5efe84 ae5efe00 807f0ce8 806cd654
[   18.684142] fe00: 7ebbca7c ae5efe90 00114809 00000040 00000006 00000006 7ebbca7c ae5efe90
[   18.684147] fe20: 00000000 00000000 00000006 0000003f 00000020 00000000 00000000 807df494
[   18.684153] fe40: 00000004 80d80540 ae5efe6c 808f4710 ae5efe74 ae5efe60 808f4710 00008946
[   18.684158] fe60: 80d80540 7ebbca7c ae5efe90 00000000 00000000 00000000 ae5efedc ae5efe88
[   18.684167] fe80: 808095b8 807ef3d0 76e8b4d0 b301e540 30687465 00000000 00000000 00000000
[   18.684177] fea0: 7ebbc400 00000000 00000000 00000000 ae5efef4 00008946 fffffdfd 80d80540
[   18.684183] fec0: 7ebbca7c 7ebbca7c b47b48e0 00000000 ae5efefc ae5efee0 807c4dc0 80809080
[   18.684188] fee0: ae41d180 ae41d180 00008946 7ebbca7c ae5eff7c ae5eff00 8029d028 807c4bb0
[   18.684193] ff00: 802a15a4 80d048e0 00000003 ae5eff40 ae5eff3c ae5eff20 8028b3d8 802a9ef0
[   18.684198] ff20: ae5eff64 ae5eff30 802a9ef0 808f278c ae5eff74 ae5eff40 807c49f0 8028b384
[   18.684204] ff40: b4a03110 aead1990 00000000 802a9524 00000006 ae41d180 ae41d180 00008946
[   18.684216] ff60: 7ebbca7c 00000003 ae5ee000 00000000 ae5effa4 ae5eff80 8029d798 8029cf88
[   18.684226] ff80: 00000006 7ebbc400 00000000 00000020 00000036 80108a84 00000000 ae5effa8
[   18.684231] ffa0: 801088e0 8029d760 7ebbc400 00000000 00000003 00008946 7ebbca7c 00000006
[   18.684237] ffc0: 7ebbc400 00000000 00000020 00000036 00000002 7ebbcc30 00000002 7ebbca74
[   18.684242] ffe0: 0004441c 7ebbbe54 0001d2ac 76e8b4dc 60070010 00000003 e594005c e5941060
[   18.684274] [<806c5610>] (phy_ethtool_set_wol) from [<806cd71c>] (lan78xx_set_wol+0xd4/0xe4)
[   18.684289] [<806cd71c>] (lan78xx_set_wol) from [<807f0ce8>] (dev_ethtool+0x1924/0x2358)
[   18.684302] [<807f0ce8>] (dev_ethtool) from [<808095b8>] (dev_ioctl+0x544/0x8a4)
[   18.684315] [<808095b8>] (dev_ioctl) from [<807c4dc0>] (sock_ioctl+0x21c/0x2d8)
[   18.684326] [<807c4dc0>] (sock_ioctl) from [<8029d028>] (do_vfs_ioctl+0xac/0x7d8)
[   18.684337] [<8029d028>] (do_vfs_ioctl) from [<8029d798>] (SyS_ioctl+0x44/0x6c)
[   18.684350] [<8029d798>] (SyS_ioctl) from [<801088e0>] (ret_fast_syscall+0x0/0x28)
[   18.684360] Code: e92dd800 e24cb004 e52de004 e8bd4000 (e59031b0) 
[   18.684482] ---[ end trace 40cd6684fa300f3c ]---
```

https://github.com/maxnet/berryboot/issues/484

MilhouseVH, 2018-03-21:
Yes, we see this too with LibreELEC, although we just disabled [78xx WOL](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux/patches/raspberrypi/linux-999-disable_lan78xx_wol.patch) as a temporary solution (there were other priorities at the time).

maxnet, 2018-03-21:
Problem isn't limited to WOL though, but seems to happen in all functions in which net->phydev is used, without checking if it is not null.

E.g. can also reproduce it with:

```
sudo ifconfig eth0 down
sudo ethtool --show-eee eth0
```

```
[   60.285843] Unable to handle kernel NULL pointer dereference at virtual address 000001b0
[   60.294113] pgd = b1a70000
[   60.296857] [000001b0] *pgd=2fe1e835, *pte=00000000, *ppte=00000000
[   60.303261] Internal error: Oops: 17 [#1] SMP ARM
[   60.308041] Modules linked in: fuse rfcomm cmac bnep hci_uart bluetooth ecdh_generic i2c_dev ip_tables x_tables ipv6 brcmfmac brcmutil snd_bcm2835(C) joydev cfg80211 snd_pcm rfkill snd_timer snd uio_pdrv_genirq uio fixed
[   60.327992] CPU: 3 PID: 966 Comm: ethtool Tainted: G         C      4.14.26v7-aufs #1
[   60.335935] Hardware name: BCM2835
[   60.339385] task: b1af2d00 task.stack: b1a08000
[   60.343990] PC is at phy_ethtool_get_eee+0x14/0x130
[   60.348943] LR is at lan78xx_get_eee+0x40/0xd4
[   60.353450] pc : [<806c5900>]    lr : [<806cd2e8>]    psr: 60000013
[   60.359807] sp : b1a09dc0  ip : b1a09dd8  fp : b1a09dd4
[   60.365106] r10: 00000000  r9 : 00000040  r8 : 00114809
[   60.370408] r7 : aee0c000  r6 : 00000000  r5 : b1a09e28  r4 : aee0c000
[   60.377031] r3 : 806cd2a8  r2 : 00000000  r1 : b1a09e28  r0 : 00000000
[   60.383652] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   60.390884] Control: 10c5383d  Table: 31a7006a  DAC: 00000055
[   60.396706] Process ethtool (pid: 966, stack limit = 0xb1a08210)
[   60.402793] Stack: (0xb1a09dc0 to 0xb1a0a000)
[   60.407207] 9dc0: aee0c000 b1a09e28 b1a09dfc b1a09dd8 806cd2e8 806c58f8 00000000 b1a09e10
[   60.415500] 9de0: 00000004 7ea41bc4 8095bbe4 7ea41c1c b1a09e84 b1a09e00 807efa4c 806cd2b4
[   60.423792] 9e00: 1cd36835 b1a09e40 00114809 00000040 00000044 00000044 b1a09ea4 b1a09e28
[   60.432083] 9e20: 802591e4 8021e3b8 00000044 00000000 00000000 00000000 00000000 00000000
[   60.440375] 9e40: 00000000 00000000 00000000 00000000 b1a09e74 b1a09e60 808f4710 00008946
[   60.448667] 9e60: 80d80540 7ea41c1c b1a09e90 00000000 00000000 00000000 b1a09edc b1a09e88
[   60.456959] 9e80: 808095b8 807ef3d0 00000000 b4a2b780 30687465 00000000 00000000 00000000
[   60.465251] 9ea0: 7ea41bc4 00000000 00000000 00000000 b1a09ef4 00008946 fffffdfd 80d80540
[   60.473543] 9ec0: 7ea41c1c 7ea41c1c 9c827520 00000000 b1a09efc b1a09ee0 807c4dc0 80809080
[   60.481834] 9ee0: 9cd1ea80 9cd1ea80 00008946 7ea41c1c b1a09f7c b1a09f00 8029d028 807c4bb0
[   60.490126] 9f00: 802a15a4 80d048e0 00000003 b1a09f40 b1a09f3c b1a09f20 8028b3d8 802a9ef0
[   60.498418] 9f20: b1a09f64 b1a09f30 802a9ef0 808f278c b1a09f74 b1a09f40 807c49f0 8028b384
[   60.506710] 9f40: b4a03110 9c865330 00000000 802a9524 00000044 9cd1ea80 9cd1ea80 00008946
[   60.515002] 9f60: 7ea41c1c 00000003 b1a08000 00000000 b1a09fa4 b1a09f80 8029d798 8029cf88
[   60.523294] 9f80: 00000044 7ea41c14 7ea41bc4 0005f430 00000036 80108a84 00000000 b1a09fa8
[   60.531585] 9fa0: 801088e0 8029d760 7ea41c14 7ea41bc4 00000003 00008946 7ea41c1c 00000044
[   60.539877] 9fc0: 7ea41c14 7ea41bc4 0005f430 00000036 7ea41dd0 0002dcc8 00031514 00000000
[   60.548168] 9fe0: 76e048a0 7ea41bbc 00018010 76e048ac 60000010 00000003 00000000 00000000
[   60.556469] [<806c5900>] (phy_ethtool_get_eee) from [<806cd2e8>] (lan78xx_get_eee+0x40/0xd4)
[   60.565030] [<806cd2e8>] (lan78xx_get_eee) from [<807efa4c>] (dev_ethtool+0x688/0x2358)
[   60.573152] [<807efa4c>] (dev_ethtool) from [<808095b8>] (dev_ioctl+0x544/0x8a4)
[   60.580653] [<808095b8>] (dev_ioctl) from [<807c4dc0>] (sock_ioctl+0x21c/0x2d8)
[   60.588067] [<807c4dc0>] (sock_ioctl) from [<8029d028>] (do_vfs_ioctl+0xac/0x7d8)
[   60.595656] [<8029d028>] (do_vfs_ioctl) from [<8029d798>] (SyS_ioctl+0x44/0x6c)
[   60.603071] [<8029d798>] (SyS_ioctl) from [<801088e0>] (ret_fast_syscall+0x0/0x28)
[   60.610749] Code: e92dd830 e24cb004 e52de004 e8bd4000 (e59031b0) 
[   60.616957] ---[ end trace fdd7ee79ef0a2e9f ]---
```

macmpi, 2018-03-21:
related? https://github.com/raspberrypi/linux/issues/2442

penfold42, 2018-03-24:
Yep

popcornmix, 2018-04-04:
Latest rpi-update kernel has a potential fix for this issue. Please test.

MilhouseVH, 2018-04-05:
With LibreELEC and latest 4.14.32 b5b6bb9bdb5e62bc7dfc64b40bd833bffd971693, the initial WOL panic and the following eee related panic are both no longer an issue:
```
sudo ifconfig eth0 down
sudo ethtool --show-eee eth0
```

maxnet, 2018-04-05:
Seems to work

closed 2018-04-05 by maxnet