raspberrypi/linux #2459: Issue with latest kernel on RPi3 - undervoltage detected!

nikall, 2018-03-22:
I have a 4ports self powered USB hub connected to the RPi with a weather station connected on it (USB hub). This setup works just fine for the last 3+ years until recently when I performed an kernel update using rpi-update. When I have weather station connected on the HUB i get in the console the undervoltage warning. This shouldn't be happen since the HUB is powered and the weather station gets powered by the HUB not the RPi. Issue is not present when I revert back to stock kernel. I am running raspbian stretch. 

Thank you.

pelwell, 2018-03-22:
How recent was your update? There was a firmware release out for a while which would occasionally fail to update the power management chip, but all releases since the March 13th should be OK in this regard.

nikall, 2018-03-22:
Today I performed another update with the same problem. I detected the problem 4 days ago with the previous update). Now I am running on stock kernel.

nikall, 2018-03-22:
My USB device tree (you may need the info of the USB HUB and Weather station).

Bus 001 Device 006: ID 14cd:6116 Super Top M6116 SATA Bridge
Bus 001 Device 005: ID 1941:8021 Dream Link WH1080 Weather Station / USB Missile Launcher
Bus 001 Device 004: ID 0bda:5401 Realtek Semiconductor Corp. RTL 8153 USB 3.0 hub with gigabit ethernet
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pelwell, 2018-03-22:
The warnings on the console were only added recently, so it is possible there has always been a problem but that you've not been aware of it.

On your "stock" kernel (I presume you mean the one from the "raspberrypi-bootloader" Raspbian package), perform some representative tasks then run "vcgencmd get_throttled" and post the output.

nikall, 2018-03-22:
I am sure the problem wasn't there (I use rpi-update kernels from the very beginning) and I say that because when undervoltage warning is present I can't ssh my RPi and that's how I detected the issue 4 days ago.

pelwell, 2018-03-22:
Humour me - run the test.

nikall, 2018-03-22:
I run this test:

`
nice yes >/dev/null &

for i in `seq 1 4`; do echo reading: $i; sudo dd if=/dev/mmcblk0 of=/dev/null bs=4M; done

for i in `seq 1 4`; do echo writing: $i; dd if=/dev/zero of=deleteme.dat bs=1M count=512; sync; done

killall yes
rm deleteme.dat

echo -n "CPU freq: " ; cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
echo -n "CPU temp: " ; cat /sys/class/thermal/thermal_zone0/temp
dmesg | tail
`
vcgencmd get_throttled output after the test completed 

"throttled=0x50000"

at the same time I have weather station connected and weewx doing its stuff with it. No problem at all.

I reverted back to stock kernel using 

"apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel"

Thank you

pelwell, 2018-03-22:
The output of the get_throttled command is a set of bits:
```
 0: Under-voltage now
 1: ARM frequency capped now
 2: Throttled now
16: Under-voltage previously
17: ARM frequency capped previously
18: Throttled previously
```
So "throttled=0x50000" means that the Pi is OK now but has been throttle because of an under-voltage situation.

What is your power supply rated at? Is it wired directly or via a switch - some switches can be quite lossy.

nikall, 2018-03-22:
It's a LG phone quick charger rated at 5V/2.0A. I use this PS for the last one year with no problem at all. As I said currently I m running on stock kernel (where test was performed). With stock kernel never had this issue. I will update again later on your kernel to test it again and post the results.

pelwell, 2018-03-22:
We recommend a good 2.5A supply for a Pi 3 - the official supply is one example, but there are others.

Undervoltage warnings shouldn't block SSH, so there could be a problem in that area - does the Pi remain responsive when attached to a USB keyboard and monitor?

nikall, 2018-03-22:
Using stock kernel or the newer one? As I said I m pretty sure for the PS, it's not the issue nor with the previous version of kernels installed with rpi-update. Only 4 days ago after the last update using rpi-update I noticed the problem. The only device powered directly from rpi hub is the external hard disk case. 2.0A are more than enough for this. Later I will connect a keyboard/mouse and let you know.

pelwell, 2018-03-22:
You may have got away with your current supply, but that doesn't make it a good choice.

If you've been able to connect via the serial console then you can skip the keyboard test for now. I'm just trying to get a sense for what is stopping your SSH sessions. Possible culprits include the USB controller, just the Ethernet controller, or something else in the software layers. Please run the following diagnostic commands and post the output:
```
$ ifconfig eth0
$ systemctl status ssh
$ dmesg
``

nikall, 2018-03-22:
I am using bonding

ifconfig bond0
bond0: flags=5187<UP,BROADCAST,RUNNING,MASTER,MULTICAST>  mtu 1500
        inet 192.168.2.50  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::ba27:ebff:feec:78df  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:ec:78:df  txqueuelen 1000  (Ethernet)
        RX packets 2968  bytes 349269 (341.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1843  bytes 189800 (185.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

systemctl status ssh
● ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-22 16:17:22 EET; 56min ago
  Process: 975 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
 Main PID: 1135 (sshd)
   CGroup: /system.slice/ssh.service
           └─1135 /usr/sbin/sshd -D

Mar 22 17:09:18 rpi sshd[3568]: Received disconnect from 119.249.54.217 port 50849:11:  [preauth]
Mar 22 17:09:18 rpi sshd[3568]: Disconnected from 119.249.54.217 port 50849 [preauth]
Mar 22 17:12:40 rpi sshd[3848]: rexec line 18: Deprecated option KeyRegenerationInterval
Mar 22 17:12:40 rpi sshd[3848]: rexec line 19: Deprecated option ServerKeyBits
Mar 22 17:12:40 rpi sshd[3848]: rexec line 30: Deprecated option RSAAuthentication
Mar 22 17:12:40 rpi sshd[3848]: rexec line 37: Deprecated option RhostsRSAAuthentication
Mar 22 17:12:40 rpi sshd[3848]: reprocess config line 30: Deprecated option RSAAuthentication
Mar 22 17:12:40 rpi sshd[3848]: reprocess config line 37: Deprecated option RhostsRSAAuthentication
Mar 22 17:12:40 rpi sshd[3848]: Accepted publickey for root from 192.168.2.10 port 52497 ssh2: RSA SHA256:pk1uwBAlabXEMsv+QSSKg0VkieJcK0eVS8OkTcA4+Vw
Mar 22 17:12:40 rpi sshd[3848]: pam_unix(sshd:session): session opened for user root by (uid=0)

nikall, 2018-03-22:
How can I attach the dmesg output on my reply by uploading a txt file ?

nikall, 2018-03-22:
I am using stock kernel...please notice it.

nikall, 2018-03-22:
Also I have to leave for the rest 4 hours so more replies when I get back. Sorry for that.

pelwell, 2018-03-22:
I'm trying to understand why the rpi-update kernel fails to accept ssh connections, so those tests must be run on the rpi-update kernel when it is not responding to ssh.

Pasting the dmesg output into a comment is fine (use ` ``` ` on the lines before and after to make it more readable).

I'm happy to go at your pace, but you might be able to speed things up if you can work out for yourself the immediate cause of SSH not responding, or at least to include the results of other things you have tried - pings, etc, - so I can rule them out as potential causes.

nikall, 2018-03-22:
Hi,

trying to reproduce the problem I saw that kernel has been updated just 5 hours or so ago. Suddenly the problem disappeared. I suppose that this has something to do with the latest commit 5 hours ago.  

anyway, consider the case closed as I am not able to see the undervoltage warning anymore.

pelwell, 2018-03-22:
It could be the kernel version bump, or it might be the USB fifo size patch. Either way, thanks for the feedback.

nikall, 2018-03-23:
Thanks for the support!

closed 2018-03-23 by nikall