raspberrypi/linux #2461: RPI3B+ 64bit not ethernet

derpeter, 2018-03-23:
I'm trying to build a 64bit Kernel für the rpi3b+. As the DTB files seems to be only present in the arm but not in the arm64 tree i did the changes in this patch **https://p.rrbone.net/paste/LbasFZgk#u7DUQBwNAGq03D4d-rydFzmFnH/GJAF46ox9cid4lAx**

So far everthing seems to work except that i dont see the ethernet interface. Wifi is working fine.
Any hint what i have missed would be great.

I tested this with 4.9.80-v8+

lategoodbye, 2018-03-23:
Did you enable the LAN78XX driver?

`CONFIG_USB_LAN78XX=m`

derpeter, 2018-03-23:
ah no, didn't. I assumed defconfig will to the right thing. 
I rebuildet the kernel but cant't test it right now.
I will report if that fixes the issue.

MilhouseVH, 2018-03-23:
Use `CONFIG_USB_LAN78XX= y` if you ever fancy using net boot.

derpeter, 2018-03-24:
OK works :-) thank you very much.

closed 2018-03-24 by derpeter