raspberrypi/linux #2465: Create tag and release matching the official Raspbian image from 2018-03-13

VladimirTechMan, 2018-03-24:
Right now, the latest tag and release published for Raspberry Pi Linux kernel are still "raspberrypi-kernel_1.20171029-1". The ones matching the latest official Raspbian image from 2018-03-13 are missing, somehow. Having them as an official reference point is useful for those who build their own custom kernels – but want those to match the latest Raspbian image, aside from the custom modifications applied.

XECDesign, 2018-03-26:
@popcornmix is HEAD of https://github.com/raspberrypi/linux/commits/rpi-4.9.y-stable the commit to tag?

popcornmix, 2018-03-26:
Yes, HEAD of stable matches the last raspbian image (and apt-get firmware/kernel).

XECDesign, 2018-03-27:
Tagged https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20180313-1

closed 2018-03-27 by VladimirTechMan