raspberrypi/linux #2466: The ondemand governor does not work properly with the 4.14.x kernel vs the 4.9.x kernel

graysky2, 2018-03-25:
I am running Arch ARM (armv7h) on both a Raspberry Pi 3 Model B Plus Rev 1.3 and a Raspberry Pi 2 Model B Rev 1.1.  When I boot either of these using the identical micro SD, I find that under kernel version 4.14.29 (Arch ARM's current version), the ondemand governor has the idle machine throttled up to the maximum frequency for that hardware as judged by running a simple script to query it and a few other parameters (shown at the bottom of the report).  When I boot into kernel version 4.9.80, the idle frequency is the minimum as expected.

### For example, when booted into kernel version 4.9.80 on the RPi3B+ (I get identical results on the RPi2):
``` 
% uptime
 06:31:24 up 14 min,  1 users,  load average: 0.00, 0.00, 0.01

% watch -t ~/bin/temps
CPU temp           :   37°C
CPU freq           :   600 MHz (ondemand)
```
Letting the watch command run for a few minutes, I never saw the frequency increase (as expected since the machine is idle).

### Same hardware after booting into the 4.14.29 kernel:
``` 
% uptime
 06:57:15 up 6 min,  1 user,  load average: 0.00, 0.01, 0.00

% cat /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq 
600000

% watch -t ~/bin/temps
CPU temp           :   43°C
CPU freq           :   1400 MHz (ondemand)
```
Letting the watch command run for a few minutes, I never saw the frequency decrease.  As you see the idle temp is also higher under the this kernel.

The script `~/bin/temps` is simply:
```
#!/bin/sh
freq=$(cat /sys/devices/system/cpu/cpufreq/policy0/scaling_cur_freq)
gov=$(cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor)
temp=$(cat /sys/devices/virtual/thermal/thermal_zone0/temp)

echo "CPU temp           :   $(( temp / 1000 ))°C"
echo "CPU freq           :   $(bc <<< "scale=0; $freq/1000") MHz ($gov)"
```

graysky2, 2018-03-26:
>Can't confirm. CPU runs constantly @ 600MHz when idle (XBian, kernel 4.14.27+)

I will try writing out `XBian_Latest_rpi3.img` to a spare micro SD card to confirm as well.  If it is an Arch-ARM-specific problem, I'm wondering how to diagnose.  Differences in kernel configs (xbian vs Arch ARM)?

pelwell, 2018-03-26:
On-demand CPU scaling is working fine under Raspbian on a 4.14.29 kernel.

On an idle machine, run `vcgencmd measure_clock arm` to confirm.

pelwell, 2018-03-26:
That does look broken. One for the Arch devs, I think, since RPi and XBian kernels don't share the problem.

If the Arch maintainers want to compare notes with us we're happy to chat.

MilhouseVH, 2018-03-26:
Not seeing any issues with ondemand and frequency scaling with LibreELEC builds using 4.14.29 - all seems normal.

graysky2, 2018-03-26:
I can't explain it either.  I am putting Raspbian on a spare card to see what I can learn.  I will close this since it's not general.  Thanks you all for the replies!

closed 2018-03-26 by graysky2