raspberrypi/linux #2467: oops in lan78xx_delayedwork on kernel 4.9 when rebooting

HiassofT, 2018-03-25:
We got a report that sometimes LibreELEC 8.2 wouldn't reboot on RPi3B+.

So far I couldn't reproduce that on Raspbian, but I managed to reproduce it on LibreELEC with an unmodified rpi-4.9 kernel (4.9.80 / commit https://github.com/raspberrypi/linux/commit/be97febf4aa42b1d019ad24e7948739da8557f66) using bcm2709_defconfig with only a minor modification (squashfs built into the kernel).

Here's the LE tree I used for testing: https://github.com/HiassofT/LibreELEC.tv/tree/le82-test-rpi-foundation-kernel

In my tests I needed 5-20 reboots until the issue showed up.

The oops seems to happen when the interface is brought down. This results in the following oops and the system locks up
```
[   31.842306] Internal error: Oops: 5 [#1] SMP ARM
[   31.848199] Modules linked in: uinput 8021q garp stp llc brcmfmac brcmutil cfg80211 evdev joydev hid_logitech_hidpp hid_logitech_dj rfkill bcm2835_gpiomem fixed uio_pdrv_genirq uio sch_fq_codel ipv6
[   31.865886] CPU: 0 PID: 208 Comm: kworker/0:2 Not tainted 4.9.80-v7 #1
[   31.872403] Hardware name: BCM2835
[   31.875802] Workqueue: events lan78xx_delayedwork
[   31.880498] task: ae2c3b00 task.stack: addf0000
[   31.885020] PC is at lan78xx_delayedwork+0xe0/0x4c8
[   31.889888] LR is at lan78xx_delayedwork+0xd8/0x4c8
[   31.894757] pc : [<8055b834>]    lr : [<8055b82c>]    psr: 40080013
[   31.894757] sp : addf1e88  ip : addf1eb8  fp : addf1edc
[   31.906225] r10: 00000000  r9 : af0f54d8  r8 : af0f8d00
[   31.911440] r7 : 00000001  r6 : 00000000  r5 : ae39d500  r4 : ae39d57c
[   31.917957] r3 : 00000000  r2 : 0000001a  r1 : 00000000  r0 : addf1e8c
[   31.924475] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   31.931601] Control: 10c5383d  Table: 2e20006a  DAC: 00000055
[   31.937336] Process kworker/0:2 (pid: 208, stack limit = 0xaddf0210)
[   31.943680] Stack: (0xaddf1e88 to 0xaddf2000)
[   31.948029] 1e80:                   addf1efc 00000001 00000000 00000000 00000000 00000000
[   31.956198] 1ea0: 00000000 00000000 00000000 00000000 00000000 00000000 801363fc ae39d57c
[   31.964367] 1ec0: ae394080 af0f54c0 00000000 af0f8d00 addf1f14 addf1ee0 80136698 8055b760
[   31.972536] 1ee0: 00000008 80e02d00 af0f54d8 ae394080 af0f54c0 ae394098 00000008 80e02d00
[   31.980705] 1f00: af0f54d8 addf0000 addf1f54 addf1f18 80137628 80136480 ae2fec00 80956a98
[   31.988874] 1f20: 80e72677 af0f54c0 00000000 00000000 ae2fec00 addf0000 ae394080 801375c8
[   31.997043] 1f40: 00000000 00000000 addf1fac addf1f58 8013cccc 801375d4 addf1f58 00000000
[   32.005212] 1f60: 2e3d0000 ae394080 00000000 00000000 addf1f70 addf1f70 00000000 00000000
[   32.013381] 1f80: addf1f80 addf1f80 addf1fac ae2fec00 8013cbc4 00000000 00000000 00000000
[   32.021550] 1fa0: 00000000 addf1fb0 801080f4 8013cbd0 00000000 00000000 00000000 00000000
[   32.029719] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   32.037888] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 e92d4070 e2505000
[   32.046065] [<8055b834>] (lan78xx_delayedwork) from [<80136698>] (process_one_work+0x224/0x518)
[   32.054760] [<80136698>] (process_one_work) from [<80137628>] (worker_thread+0x60/0x5fc)
[   32.062844] [<80137628>] (worker_thread) from [<8013cccc>] (kthread+0x108/0x120)
[   32.070236] [<8013cccc>] (kthread) from [<801080f4>] (ret_from_fork+0x14/0x20)
[   32.077451] Code: e59364f0 ebfc061c e3a0201a e50b7050 (e596118c) 
] Started [ 32.083626] ---[ end trace a6cb6f641456d1af ]---
```
The full console log from bootup til system lockup is here: https://gist.github.com/HiassofT/c15c5098c8685af236483f998bd32a70

I then built a kernel with debug symbols and attached gdb to see what's going on:
```
(gdb) target remote /dev/ttyUSB0
Remote debugging using /dev/ttyUSB0
phy_read (regnum=<optimized out>, phydev=<optimized out>) at ./include/linux/phy.h:647
647             return mdiobus_read(phydev->mdio.bus, phydev->mdio.addr, regnum);
(gdb) where
#0  phy_read (regnum=<optimized out>, phydev=<optimized out>) at ./include/linux/phy.h:647
#1  lan78xx_link_reset (dev=<optimized out>) at drivers/net/usb/lan78xx.c:1102
#2  lan78xx_delayedwork (work=0xadc6357c) at drivers/net/usb/lan78xx.c:3244
#3  0x80136698 in process_one_work (worker=0xae88c500, work=0x0) at kernel/workqueue.c:2092
#4  0x80137628 in worker_thread (__worker=0xae88c500) at kernel/workqueue.c:2226
#5  0x8013cccc in kthread (_create=0xae88d680) at kernel/kthread.c:211
#6  0x801080f4 in ret_from_fork () at arch/arm/kernel/entry-common.S:118
(gdb) up
#1  lan78xx_link_reset (dev=<optimized out>) at drivers/net/usb/lan78xx.c:1102
1102            ret = phy_read(phydev, LAN88XX_INT_STS);
(gdb) p phydev
$1 = (struct phy_device *) 0x0
(gdb) up
#2  lan78xx_delayedwork (work=0xadc6357c) at drivers/net/usb/lan78xx.c:3244
3244                    if (lan78xx_link_reset(dev) < 0) {
```
A NULL phydev is a good reason for the kernel getting a bit upset :)

So far we haven't seen this bug in 4.14 kernels. Checking through gitlog a lot of changes to lan78xx where added between 4.9 and 4.14 and the phy_read call where 4.9 oopsed was removed by commit https://github.com/raspberrypi/linux/commit/cc89c323a30e73114a21fc125d2f841658a09987 in 4.10 (not sure if this commit is actually related to the issue, though).

My guess is that some lan78xx backport commits might be needed for 4.9. @pelwell you seem to be fairly familiar with that driver, could you please have a look at it?

pelwell, 2018-03-25:
Our development effort is on 4.14 now, but if there is still demand for 4.9 support we would probably accept some back-port PRs.

HiassofT, 2018-03-26:
I dug a bit into the driver code and it looks like lan78xx_stop is racy:

The phydev is disconnected and net->phydev is set to zero at the beginning: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/usb/lan78xx.c#L2431
but the workqueue is cleared some time after that: https://github.com/torvalds/linux/blob/master/drivers/net/usb/lan78xx.c#L2605

This leaves a window where a deferred link reset (probably scheduled from link_status https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/usb/lan78xx.c#L1206) could be starting / running without net->phydev set.

The relevant code part in lan78xx_stop hasn't been changed in newer kernels, so probably it's racy in all kernel versions and I just had been "lucky" to catch such a race in 4.9

@MilhouseVH pointed me to this proposed patch https://patchwork.kernel.org/patch/10282365/ which was mentioned in #2442 It looks like this patch (or it's later incarnations, as changes were requested) should fix the issue as it ties the phydev lifecycle to the driver lifecycle (instead of interface).

JamesH65, 2018-04-20:
@HiassofT @pelwell ISTR this patch got included? Can we close?

pelwell, 2018-04-20:
Yes - a combination of the proposed patch mentioned above and a followup, both now merged upstream, should have fixed the problem.

closed 2018-04-20 by pelwell