raspberrypi/linux #2469: Source archive don't unpack correctly

kennyliveforever, 2018-03-26:
---- VOLUMIO RASPBERRY PI KERNEL SOURCE DOWNLOADER ----

This process might take a long time

Checking if build essential is installed
Firmware revision is 5c80565c5c0c7f820258c792a98b56f22db2dd03
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   169  100   169    0     0     26      0  0:00:06  0:00:06 --:--:--    44
100    41  100    41    0     0      4      0  0:00:10  0:00:08  0:00:02    28
Kernel revision is a2f34d45809d8685bca1e91989e35746499ac400
Getting modules symvers for V6 kernel
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   175  100   175    0     0     65      0  0:00:02  0:00:02 --:--:--    65
100  720k  100  720k    0     0  47107      0  0:00:15  0:00:15 --:--:-- 49220
Downloading Kernel source tarball from  https://github.com/raspberrypi/linux/archive/a2f34d45809d8685bca1e91989e35746499ac400.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   159    0   159    0     0     53      0 --:--:--  0:00:02 --:--:--    53
  2  138M    2 4123k    0     0  14972      0  2:41:38  0:04:41  2:36:57     0a2  2  138M    2 4123k    0     0   5270      0  7:39:12  0:13:21  7:25:51     0curl: (56) SSL read: error:00000000:lib(0):func(0):reason(0), errno 104
creating /usr/src/rpi-linux folder
Extracting Kernel

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
Cloning current config
Copying modules symverse
make: *** No rule to make target 'modules_prepare'.  Stop.

pelwell, 2018-03-26:
Errno 104 is ECONNRESET - the server reset the connection.

We've found downloading archives from GitHub to be unreliable - we recommend using `git clone --depth 1 ...` instead. It could also be that your network connection glitched, but either way I don't think this is the correct place to report the problem unless you are seeing errors from the kernel.

kennyliveforever, 2018-03-26:
Thanks!
I'll try to do this

popcornmix, 2018-03-26:
Another thing suggested here https://github.com/Hexxeh/rpi-update/issues/174
is to create a file ~root/.curlrc containing the following line
--keepalive-time 2

JamesH65, 2018-04-20:
Closing this issue as questions answered/issue resolved.

closed 2018-04-20 by JamesH65