raspberrypi/linux #2470: arch/arm/lib/memset_rpi.S:65: Error: garbage following instruction -- `orr DAT0,DAT0,lsl#8'

jeffersonchua, 2018-03-26:
Problem cross-compiling linux kernel 4.16-rc6 for pi0 (armv6).

Kernel: Linux 4.16-rc6
crosstool-ng-1.23.0
CT_CC_GCC_VERSION="6.3.0"
CT_LIBC_VERSION="2.25"
CT_LIBC_GLIBC_V_2_25=y
CT_ARCH_SUPPORTS_WITH_ARCH=y
CT_ARCH_SUPPORTS_WITH_CPU=y
CT_ARCH_SUPPORTS_WITH_TUNE=y
CT_ARCH_SUPPORTS_WITH_FLOAT=y
CT_ARCH_SUPPORTS_WITH_FPU=y
CT_ARCH_SUPPORTS_SOFTFP=y
CT_ARCH_EXCLUSIVE_WITH_CPU=y
CT_ARCH_ARCH="armv6"
CT_ARCH_CPU=""
CT_ARCH_TUNE=""
CT_ARCH_FPU="vfp"
CT_ARCH_FLOAT_HW=y
CT_TARGET_CFLAGS="-mfpu=vfp -mfloat-abi=hard"
CT_TARGET_LDFLAGS=""
CT_ARCH_FLOAT="hard"
CT_TARGET_CFLAGS="-mfpu=vfp -mfloat-abi=hard"
CT_BINUTILS_VERSION="2.28"

Encountered the following error ...
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  AS      arch/arm/lib/memset_rpi.o
arch/arm/lib/memset_rpi.S: Assembler messages:
arch/arm/lib/memset_rpi.S:65: Error: garbage following instruction -- `orr DAT0,DAT0,lsl#8'
arch/arm/lib/memset_rpi.S:67: Error: garbage following instruction -- `orr DAT0,DAT0,lsl#16'
make[1]: *** [scripts/Makefile.build:425: arch/arm/lib/memset_rpi.o] Error 1
make: *** [Makefile:1051: arch/arm/lib] Error 2

No problem compiling linux-4.15

Thanks,
Jeff

popcornmix, 2018-03-27:
No changes in that file between rpi-4.15.y and rpi-4.16.y.
But I can reproduce the error - I'm guessing there are some changes to assembler flags in 4.16.
I've pushed a fix. Okay for you now?

jeffersonchua, 2018-03-27:
> No changes in that file between rpi-4.15.y and rpi-4.16.y.

Yes. I see that as well. I reported it because this looks strange to me that nobody reported it.

> I've pushed a fix. Okay for you now?

Fixed! Compiled fine. Now will need to try to boot it up for both pi0w and pi3.  I had problem booting up with same config ok for 4.15, but 4.16 on rpi3 get stucked at a single underscore.

Thank you!

Jeff.

popcornmix, 2018-03-27:
There's a problem with the USB fiq on 4.16.
Adding ``dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0`` to cmdline.txt may help.

jeffersonchua, 2018-03-27:
>  Adding dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 to cmdline.txt may help.

This works. Without it, won't work.

Now both pi0w and pi3 works on 4.16 :)

Thank you once again!

Jeff

closed 2018-03-27 by jeffersonchua