raspberrypi/linux #2475: lan78xx on 3B+ does not detect network link while 3B does just fine

gtrainavicius, 2018-03-27:
I have a lengthy 100Mbps cable at my home (30+ meters) and it works well with Raspberry Pi 3B, as well as other network devices like laptops, however, Raspberry Pi 3B+ is unable to automatically recognize and negotiate the network link.

`dmesg` on 3B gives:

```
 IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
 IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
 IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
 smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC3E1
```

the green and amber LEDs blink.

and on 3B+:

```
 libphy: lan78xx-mdiobus: probed
...
 IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
 IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
 IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

If 100Mbps full duplex mode is forced on 3B+, the network connection starts working and the green LED starts blinking (note the 'Link detected'):

```
modep@modep:~ $ sudo ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        Supports Wake-on: pumbag
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: no
modep@modep:~ $ sudo ethtool -s eth0 speed 100 duplex full autoneg off
modep@modep:~ $ sudo ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  100baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: No
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: off
        Supports Wake-on: pumbag
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

gtrainavicius, 2018-03-27:
I noticed this workaround for long cables here, might be relevant: https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L1827

6by9, 2018-03-28:
Daft question - have you checked the cable is good across all 4 pairs? 100Mbit and below only uses two of the pairs, whilst 1000Mbit uses all 4. You may have enough connectivity for the Pi and switch to negotiate 1000 but not enough to actually use it.
See comments on threads such as https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=208520#p1290412 where PoE adapters have been wired with only 2 pairs through, yet the devices are negotiating gigabit.

pelwell, 2018-03-28:
Can I assume that links are established correctly with shorter cables?

Try disabling the EEE support (`dtparam=eee=off`) - it's unlikely to make a difference, but it's worth a go.

gtrainavicius, 2018-03-28:
Yes, it does work with shorter cables. `dtparam=eee=off` didn't fix the issue.

I will test my cable later today. Still it's weird that other devices, including the RPi3, work just fine with identical setup.

6by9, 2018-03-28:
RPi3 is 100Mbit only. It doesn't have the option of negotiating gigabit and needing all 4 pairs.
You don't say what the other devices are so I can't say what speed they will try to negotiate.

rodizio1, 2018-03-28:
Ethtool says the other side is also 100mbit:

```
Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
```

gtrainavicius, 2018-03-28:
I have checked the cable connection - the cable test tool does report that it only has 2 pairs connected.

> what the other devices are

The router is Technicolor MediaAccess TG389ac, ethernet ports are 1Gbps capable.
My laptop has I219-LM ethernet card, 1Gbps capable, and it automatically negotiates 100Mbps link using the cable.

> It doesn't have the option of negotiating gigabit and needing all 4 pairs.

The issue here is not about not being able to achieve 1Gbps, but about getting 100Mbps automatically set up with a 100Mbps cable. I would expect 3B+ to negotiate 100Mbps automatically in case 1Gbps is not possible.

6by9, 2018-03-28:
No, it is doing the correct thing according to the base 1000base-T spec.
https://en.wikipedia.org/wiki/Gigabit_Ethernet#1000BASE-T
> Since negotiation takes place on only two pairs, if two gigabit devices are connected through a cable with only two pairs, the devices will successfully choose 'gigabit' as the highest common denominator (HCD), but the link will never come up. Most gigabit physical devices have a specific register to diagnose this behaviour. Some drivers offer an "Ethernet@Wirespeed" option where this situation leads to a slower yet functional connection.[17]

We can go back to Microchip (or at least the datasheets) to see if the chip has such a register/feature, but it's likely to take a while to sort out and implement reliably.

gtrainavicius, 2018-03-28:
Ok, I see, thank you for sharing this information. I didn't expect this was not going to be part of the standard, even though it seems it's possible to handle this situation. It'd be great to have this implemented.

andrum99, 2018-04-04:
How many pairs does the cable actually have? What are you testing it with? If there are only 2 pairs present wired to pins 1,2,3 and 6 then this should work fine. If the cable actually has 2 working pairs and 2 faulty pairs then your cable is the problem, and you can't expect it to work properly with all devices.

What speed does your laptop connect to the TG389ac at using this cable? Is it reporting gigabit?

gtrainavicius, 2018-04-05:
I have fixed up the 2 previously missing cable pairs, so all 4 are good, and now auto link detection does work fine.

You can treat this issue more like a feature request. :)

```
pi@raspberrypi:~ $ sudo ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Half 1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        Supports Wake-on: pumbag
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

boatbodger, 2018-08-29:
Thanks for this post - I was tearing my hair out after a change from 3B to 3B+*
In my situation, I only have two pairs available, so forcing the speed down to 100Mbit/s (which is oodles for my application) solves the problem, although I had to have autoneg on (not off) with a Cisco Small Business switch.
I have to say that many devices (e.g. my laptop) which have Gig Eth *do* negotiate down to 100Mbit/s when there are only two pairs available.  To me, this seems more sensible than the "standard".
I would 'vote up' the feature request to auto-detect 2-pair cables and drop to 100Mbit/s.

Oh.  BTW the dtparam=eee=off didn't make any difference.
Is there a way to force 100Mbit/s via config.txt or any other way rather than resorting to rc.local?

*and as you can see, I don't have much left

6by9, 2018-08-29:
@pelwell I'd forgotten about this one. Can we enquire of Microchip if a configuration is possible where it will automatically back off down to 100Mbit/s should it only partially negotiate 1Gbit/s?

The datasheet is pretty impenetrable unless you know everything there is to know about ethernet PHYs  It appears the driver sets the MAC_CR_AUTO_SPEED_ bit and lets the adapter get on with everything, so it may well not be possible.

pelwell, 2018-08-29:
I've asked the question, and will report back any answers.

jozef, 2018-10-20:
100Mbit auto-negotiation also did not work on my 1Gbit POE switch with POE splitter (only 2 pairs used for ethernet in that case so 100Mbit only). used `apt-get install ethtool` and pre-up configuration for that interface in `/etc/network/interfaces`:

```
auto eth0
iface eth0 inet dhcp
    pre-up ethtool -s eth0 speed 100 duplex full autoneg on
```

only then the status leds on 3B+ eth port went on and network communication was possible.

pelwell, 2018-11-27:
See https://github.com/raspberrypi/linux/pull/2771 for a potential solution to this issue.

pelwell, 2018-11-29:
The latest rpi-update kernel has a fix for this - auto-downshift (the mechanism that enables the PHY to give up on a 1000Mbs link and try 100Mbs instead) is enabled by default. You can tune the failover time using the `eth_downshift_after` parameter (e.g. `dtparam=eth_downshift_after=3`). The default value is 2 (i.e. after 2 failed auto-negotiations) - the minimum supported by the hardware. The value `0` disables the feature.

If you know that the downshift is always required you can reduce the time to get a link by limiting the maximum speed - `dtparam=eth_max_speed=100`.

N.B. This patch won't help network booting, but it will allow an NFS root partition to be used.

JamesH65, 2019-07-31:
Closing this issue as questions answered/issue resolved.

closed 2019-07-31 by JamesH65