raspberrypi/linux #2478: Device Tree Overlay for MAX98089 on the Raspberry Pi

Gadgetoid, 2018-03-28:
Feel free to close or disregard this issue if you think it's out of scope for her, but I could not think of a better place to ask.

Usually I can muddle my way through things, but Device Tree just doesn't seem to be one of them.

Story so far:

I'm trying to get a Maxim max98089 tqfp evaluation kit working with the Raspberry Pi. I can get it up and running by using the `hifiberry-dac` overlay, and the USB control features to configure the audio output stages. But I'm trying to get it working with the proper codec.

I have obtained the source for the max9808x coded from https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/codecs/max98088.c and used DKMS to build a package that lets me compile it into a loadable module for my running kernel (Raspbian's stock, at time of writing, 4.9.80).

I can load the module, see the max98089 i2c address blocked out with UU indicating that it is reserved.

However, I can not get a usable audio IO device to appear in ALSA, so this is where I'm stuck.

My dtoverlay - cobbled together from Googling and decompiling various dtbo files - currently looks like this:

```
/dts-v1/;

/ {
        compatible = "brcm,bcm2708";

        fragment@0 {
                target = <0xdeadbeef>;

                __overlay__ {
                        status = "okay";
                };
        };

        fragment@1 {
                //target-path = [2f 00];
                target = <0xdeadbeef>;

                __overlay__ {
                        #address-cells = <0x1>;
                        #size-cells = <0x0>;
                        status = "okay";

                        max98088@10 {
                                #sound-dai-cells = <0x0>;
                                compatible = "maxim,max98089", "maxim,max98088";
                                reg = <0x10>;
                                //mic-left-digital;
                                //mic-right-digital;
                                //PVDD-supply = <0xdeadbeef>;
                                //DVDD-supply = <0xdeadbeef>;
                                status = "okay";
                        };
                };
        };

        fragment@2 {
                target = <0xdeadbeef>;

                __overlay__ {
                        compatible = "maxim98088";
                        i2s-controller = <0xdeadbeef>;
                        status = "okay";
        //              clock44-gpio = <0xdeadbeef 0x5 0x0>;
        //              clock48-gpio = <0xdeadbeef 0x6 0x0>;
                };
        };

        fragment@3 {
                target-path = [2f 00];

                __overlay__ {

                        max98088-codec {
                                #sound-dai-cells = <0x0>;
                                compatible = "maxim,max98088", "maxim,max98089";
                                status = "okay";
                        };
                };
        };

        __fixups__ {
                i2s = "/fragment@0:target:0", "/fragment@2/__overlay__:i2s-controller:0";
                i2c1 = "/fragment@1:target:0";
                //vdd_3v3_reg = "/fragment@1/__overlay__/wm8804@10:PVDD-supply:0", "/fragment@1/__overlay__/wm8804@310:DVDD-supply:0";
                sound = "/fragment@2:target:0";
                //gpio = "/fragment@2/__overlay__:clock44-gpio:0", "/fragment@2/__overlay__:clock48-gpio:0";
        };
};
```

pelwell, 2018-03-28:
Don't write overlays by decompiling others - compilation is a lossy process, and once compiled the overlays become fragile. The source to all the overlays is availabe in our kernel trees - the `hifiberry-dac` source is [here](https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/overlays/hifiberry-dac-overlay.dts) (it's unchanged from the 4.9 kernel).

Regenerate your overlay from the source, then start a topic in the [Device Tree section of the RPi Forums](https://www.raspberrypi.org/forums/viewforum.php?f=107). I usually get notified of posts there pretty promptly, but feel free to send me (PhilE) a PM to alert me to it.

closed 2018-03-28 by pelwell

Gadgetoid, 2018-03-28:
Thank you- forcing me back to square one with the dtoverlay was crucial to my success! It's not perfect, but it works.

I suppose this is an appropriate place to ask- but how do we go about cleaning this up and submitting it for integration here? We'd likely want maxim9808x support integrated into Raspbian if we ever happen to build a product around it, albeit right now my experiments are just speculative.

pelwell, 2018-03-28:
The max98088 driver has been in the kernel for years, and it supports the max98089. It's lacking the compatible strings, but thanks to a bit of Raspbian udev magic you shouldn't need them - if you ask for "maxim,98089" it will also try the i2c alias "98089".

If you are up for building your own kernel, add "CONFIG_SND_SOC_MAX98088=m" to your .config and give it a go.