raspberrypi/linux #2479: 4.14.x: CONFIG_BT_HCIUART_BCM is not enabled in arch/arm/configs/bcmrpi_defconfig

amak79, 2018-03-29:
I have a RPi Zero W running Void Linux (4.14.30 armv6l), which uses the default [config](https://github.com/voidlinux/void-packages/blob/master/srcpkgs/rpi-kernel/template#L71) `arch/arm/configs/bcmrpi_defconfig`.

I haven't been able to get Bluetooth working. When running `btattach` (`hciattach` is deprecated and unavailable on Void Linux), I get the following:

```
# btattach -N -B /dev/ttyAMA0 -P bcm -S 3000000
Attaching Primary controller to /dev/ttyAMA0
Switched line discipline from 0 to 15
Failed to set protocol: Protocol not supported
No controller attached
```
When comparing `dmesg` output with Raspbian, I noticed that on Void Linux the Broadcom UART protocol wasn't being registered. While on Raspbian `dmesg` shows the following:
```
...
Bluetooth: HCI UART protocol Broadcom registered
...
```
This has led me to suspect that Broadcom UART protocol support was missing from the kernel.

Inspecting the 4.14.x config `arch/arm/configs/bcmrpi_defconfig` shows that `CONFIG_BT_HCIUART_BCM` is not enabled, but is in 4.9.x. I believe this is caused by [this](https://github.com/torvalds/linux/commit/33cd149e767be9afbab9fcd3d5165a2de62313c8) commit introduced in 4.14, which added a `CONFIG_BT_HCIUART_SERDEV` dependency to `CONFIG_BT_HCIUART_BCM`. `CONFIG_BT_HCIUART_SERDEV` needs `CONFIG_SERIAL_DEV_BUS`, which is also not enabled. Without these new dependencies, the `hci_uart` module is not built with Broadcom UART protocol support.

amak79, 2018-03-31:
After upgrading Raspbian to 4.14.x which doesn't have `CONFIG_BT_HCIUART_BCM` enabled, it appears that Bluetooth works without it. I can only assume that the issue on Void Linux lies with `btattach`. All other distros are using a patched `hciattach`. Is it even possible to use plain `btattach`?

amak79, 2018-03-31:
I was able to get `btattach` to work on Raspbian after disabling the `hciuart` service, and using the `h4` protocol instead of `bcm`:
```
# btattach -B /dev/ttyAMA0 -P h4
Attaching Primary controller to /dev/ttyAMA0
Switched line discipline from 0 to 15
Device index 0 attached
```
Using the `-S` option with `3000000`, `924600` or `460800` prevents the controller from working, so I dropped it. The controller MAC address is set to `AA:AA:AA:AA:AA:AA` instead of the serial number derived one. I'm not sure what effect that has. I confirmed it works by paring a game controller and testing it with `jstest`.

Although `btattach` now works with Void Linux and the `hci0` device is visible under `/sys/class/bluetooth`, Bluetooth still doesn't work. For some reason `bluetoothctl` doesn't list any controllers. I have the `BCM43430A1.hcd` firmware file placed in `/lib/firmware/brcm`, but `btattach` doesn't appear to load it, even on Raspbian.

amak79, 2018-04-01:
I finally got Bluetooth working on Void Linux using `btattach`. I'm not sure what it was, maybe the Bluetooth service or RPi needed a restart. I can pair the game controller, even without the `BCM43430A1.hcd` firmware file:
```
# dmesg
...
[ 1279.244366] hid-generic 0005:2DC8:3820.0003: unknown main item tag 0x0
[ 1279.246446] input: 8Bitdo NES30 Pro as /devices/platform/soc/20201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:2DC8:3820.0003/input/input2
[ 1279.254289] hid-generic 0005:2DC8:3820.0003: input,hidraw0: BLUETOOTH HID v1.00 Gamepad [8Bitdo NES30 Pro] on aa:aa:aa:aa:aa:aa
...
```
The `AA:AA:AA:AA:AA:AA` MAC address issue remains. `btattach` doesn't offer a way to set the MAC address like `hciattatch` does. Is there any other way to set the controller MAC address?

Is the use of the HCI UART H4 protocol correct? I assumed the BCM protocol would be the correct one, but `CONFIG_BT_HCIUART_BCM` is not enabled in the 4.14 kernel and Bluetooth still works.

amak79, 2018-04-04:
I reverted to Raspbian 4.9.x which has `CONFIG_BT_HCIUART_BCM` enabled. I ran `btattach` with `-P bcm` when I noticed the BCM protocol was attempting to load the `brcm/BCM.hcd` firmware. I then copied `BCM43430A1.hcd` to `BCM.hcd` and ran `btattach` again with the following results:
```
# dmesg
...
[  110.648332] uart-pl011 20201000.serial: no DMA platform data
[  110.699913] Bluetooth: HCI UART driver ver 2.3
[  110.699938] Bluetooth: HCI UART protocol H4 registered
[  110.699944] Bluetooth: HCI UART protocol Three-wire (H5) registered
[  110.700199] Bluetooth: HCI UART protocol Broadcom registered
[  110.726961] Bluetooth: hci0: BCM: failed to write clock (-56)
[  110.734896] Bluetooth: hci0: BCM: chip id 94
[  110.734920] Bluetooth: hci0: BCM (001.002.009) build 0000
[  111.317099] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  111.317113] Bluetooth: BNEP filters: protocol multicast
[  111.317136] Bluetooth: BNEP socket layer initialized
[  114.416341] Bluetooth: hci0: BCM: failed to write clock (-56)
[  114.422338] Bluetooth: hci0: BCM (001.002.009) build 0360
```
I appears that the firmware is loaded (build changes from `0000` to `0360`), but an error occurs. The `hci0` device is visible under `/sys/class/bluetooth`, but no controller is available in `bluetoothctl`. Does the BCM protocol require a different firmware file, or is there an issue with `BCM43430A1.hcd`?

I fairly certain that my original conclusion was correct. The `CONFIG_BT_HCIUART_BCM` option should be enabled. It would be great to have someone from the RPi kernel team state their opinion on this issue.

pelwell, 2018-04-04:
As you have said, Raspbian is happy without CONFIG_BT_HCIUART_BCM. If you think that some other operating modes require it, [build your own kernel](https://www.raspberrypi.org/documentation/linux/kernel/building.md) with the option enabled, verify that it solves the problem, then report back.

amak79, 2018-04-04:
Thank you for the reply.

As I already mentioned, Void Linux uses the same kernel config as Raspbian. Building my own kernel wouldn't help. I can use Raspbian with kernel 4.9 instead. The lack of `CONFIG_BT_HCIUART_BCM` affects distributions that only have `btattach` available, instead of the deprecated `hciattach` that Raspbian uses.

There is also the additional issue that I mentioned in my previous post, where the firmware is loaded, but no controller is available when using `btattach`.

pelwell, 2018-04-04:
> As I already mentioned, Void Linux uses the same kernel config as Raspbian. Building my own kernel wouldn't help.

It would help because you would enable CONFIG_BT_HCIUART_BCM before building the kernel (as I said - "build your own kernel _with the option enabled_"). This would then allow you to confirm that it makes a positive difference.

We don't have the resources to support all the different versions of Linux out there - if there are changes we can make to the standards kernels to better support them then we will consider doing so, but others will have to diagnosis. The Void Linux maintainers (and other users) may be able to help you.

amak79, 2018-04-04:
I built my own kernel with the these additional options added to the default `arch/arm/configs/bcmrpi_defconfig`:
```
CONFIG_SERIAL_DEV_BUS=m
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
```
Running `btattach`:
```
btattach -B /dev/ttyAMA0 -P bcm -S 3000000
Attaching Primary controller to /dev/ttyAMA0
Switched line discipline from 0 to 15
Device index 0 attached
```
`dmesg` output:
```
...
[    8.058194] Bluetooth: Core ver 2.22
[    8.058367] NET: Registered protocol family 31
[    8.058376] Bluetooth: HCI device and connection manager initialized
[    8.060419] Bluetooth: HCI socket layer initialized
[    8.060446] Bluetooth: L2CAP socket layer initialized
[    8.060539] Bluetooth: SCO socket layer initialized
[    8.226164] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.226182] Bluetooth: BNEP filters: protocol multicast
[    8.226215] Bluetooth: BNEP socket layer initialized
[   24.342851] uart-pl011 20201000.serial: no DMA platform data
[   24.408126] Bluetooth: HCI UART driver ver 2.3
[   24.408143] Bluetooth: HCI UART protocol H4 registered
[   24.408150] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   24.408575] Bluetooth: HCI UART protocol Broadcom registered
[   24.542235] Bluetooth: hci0: BCM: chip id 94
[   24.544510] Bluetooth: hci0: BCM: features 0x2e
[   24.567958] Bluetooth: hci0: BCM43430A1
[   24.567990] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[   28.349397] Bluetooth: hci0: BCM (001.002.009) build 0360
[  238.722183] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  238.722228] Bluetooth: HIDP socket layer initialized
[  240.884171] hid-generic 0005:2DC8:3820.0001: unknown main item tag 0x0
[  240.886488] input: 8Bitdo NES30 Pro as /devices/platform/soc/20201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:2DC8:3820.0001/input/input0
[  240.893146] hid-generic 0005:2DC8:3820.0001: input,hidraw0: BLUETOOTH HID v1.00 Gamepad [8Bitdo NES30 Pro] on b8:27:eb:f0:c9:98
```
I can report that Bluetooth works. I was able to pair my game controller. The correct firmware was loaded without error, and a proper MAC address assigned to the `hci0` device i.e. not `AA:AA:AA:AA:AA:AA`. I can also set the baud rate, which previously prevented the controller from working.

pelwell, 2018-04-04:
Excellent - thank you. I'll make sure we enable the option in future builds.

amak79, 2018-04-05:
Thank you.

Is there a public repository where the Bluetooth firmware files `BCM43430A1.hcd` and `BCM4345C0.hcd` are kept? I couldn't find one, so I grabbed the files from a Raspbian installation. Having a public repository would make it possible to have the files available via the Void Linux `rpi-firmware` package.

XECDesign, 2018-04-05:
@amak79 Just added a repo you could grab them from directly: https://github.com/RPi-Distro/bluez-firmware/tree/master/broadcom

pelwell, 2018-04-05:
Enabling those two settings (only `CONFIG_SERIAL_DEV_BUS=m` and `CONFIG_BT_HCIUART_BCM=y` are required - `CONFIG_BT_HCIUART_SERDEV=y` is implied) causes `serdev.ko` to be built (an 18kB module) and `hci_uart.ko` to increase in size by another 18kB - the static kernel is unchanged. I think this is acceptable, especially since `CONFIG_BT_HCIUART_BCM=y` was present in 4.9.

The defconfigs have now been updated accordingly.

amak79, 2018-04-05:
@pelwell I think you may have overlooked `CONFIG_SERIAL_DEV_BUS=m` in `arch/arm64/configs/bcmrpi3_defconfig`. It's still not enabled.

JamesH65, 2018-04-20:
I believe this issue can now be closed?

closed 2018-04-21 by pelwell

woqidaideshi, 2020-04-13:
> CONFIG_BT_HCIUART_BCM

I am using rpi-4.19.y(4.19.108). 
Are you sure you can add "CONFIG_BT_HCIUART_BCM" to the default config before building the kernel? I add "CONFIG_BT_HCIUART_BCM=y", but it disappeared after building kernel. So it didn't work.  When I btattached /dev/ttyAMA0 with bcm, I got these：
# btattach -B /dev/ttyAMA0 -P bcm -S 3000000
Attaching Primary controller to /dev/ttyAMA0
Switched line discipline from 0 to 15
Failed to set protocol: Protocol not supported

Can you tell me how you solved this? Thanks.