raspberrypi/linux #2481: Kernel oops running usb storage on PI3B+

comdata, 2018-03-29:
Hello,

I am running a PI 3B+ with hard disks attached via USB. All hard disks are self powered. When running with 4.14.30 v7 I am getting regularly kernel oops under high load (e.g. compiling something)

Here is a screenshot:

[image]

Reverting back to 4.9.80 seems to fix the issue.

Kind Regards

P33M, 2018-03-29:
The top of the stacktrace is cut off, unfortunately.

Can you add `framebuffer_height=2160` to /boot/config.txt, or alternatively `enable_uart=1` and record the serial console output by dumping it on another device?

zerwes, 2018-03-29:
crashes appear after:
```
Start-Date: 2018-03-29  00:36:21
Commandline: apt-get -y dist-upgrade
Upgrade: libraspberrypi-bin:armhf (1.20180313-1, 1.20180328-1), libraspberrypi-dev:armhf (1.20180313-1, 1.20180328-1), raspberrypi-kernel:armhf (1.20180313-1, 1.20180328-1), raspberrypi-bootloader:armhf (1.20180313-1, 1.20180328-1), libraspberrypi0:armhf (1.20180313-1, 1.20180328-1), raspberrypi-sys-mods:armhf (20180315, 20180328+1)
End-Date: 2018-03-29  00:38:42
```

comdata, 2018-03-30:
Ok, so I will wait for the 4.14.32 then. Thanks.

horgh, 2018-03-31:
I think I'm experiencing this too. I've been posting about it in this thread on the forum: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=208821

With the latest firmware, I'm still seeing crashes:

```
pi@rpi5:~ $ cat /boot/.firmware_revision 
2148253d340d7c646583cb8f0be6403511522cc4
```

```
Mar 30 22:49:01 rpi5 kernel: [33987.220493] Alignment trap: not handling instruction e1903f9f at [<8079ef70>]
Mar 30 22:49:01 rpi5 kernel: [33987.220510] Unhandled fault: alignment exception (0x001) at 0xba4429ed
Mar 30 22:49:01 rpi5 kernel: [33987.222504] pgd = b9f7c000
Mar 30 22:49:01 rpi5 kernel: [33987.224400] [ba4429ed] *pgd=3a41141e(bad)
Mar 30 22:49:01 rpi5 kernel: [33987.226328] Internal error: : 1 [#1] SMP ARM
Mar 30 22:49:01 rpi5 kernel: [33987.228248] Modules linked in: ip6table_filter ip6_tables xt_tcpudp nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack iptable_filter sg brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
Mar 30 22:49:01 rpi5 kernel: [33987.237083] CPU: 1 PID: 7877 Comm: sshd Tainted: G         C      4.14.31-v7+ #1104
Mar 30 22:49:01 rpi5 kernel: [33987.241838] Hardware name: BCM2835
Mar 30 22:49:01 rpi5 kernel: [33987.244212] task: b72e8f00 task.stack: b9c68000
Mar 30 22:49:01 rpi5 kernel: [33987.246637] PC is at _raw_read_lock+0x1c/0x38
Mar 30 22:49:01 rpi5 kernel: [33987.249041] LR is at ext4_es_lookup_extent+0x30/0x244
Mar 30 22:49:01 rpi5 kernel: [33987.251491] pc : [<8079ef74>]    lr : [<803265ac>]    psr: 00000013
Mar 30 22:49:01 rpi5 kernel: [33987.254016] sp : b9c69cd0  ip : b9c69ce0  fp : b9c69cdc
Mar 30 22:49:01 rpi5 kernel: [33987.256529] r10: 00000000  r9 : 00001000  r8 : ba4429ed
Mar 30 22:49:01 rpi5 kernel: [33987.259030] r7 : 00000000  r6 : 427e0000  r5 : b9c69d28  r4 : ba442861
Mar 30 22:49:01 rpi5 kernel: [33987.261558] r3 : 00000000  r2 : b9c69d28  r1 : 427e0000  r0 : ba4429ed
Mar 30 22:49:01 rpi5 kernel: [33987.264064] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Mar 30 22:49:01 rpi5 kernel: [33987.266663] Control: 10c5383d  Table: 39f7c06a  DAC: 00000055
Mar 30 22:49:01 rpi5 kernel: [33987.269221] Process sshd (pid: 7877, stack limit = 0xb9c68210)
Mar 30 22:49:01 rpi5 kernel: [33987.271769] Stack: (0xb9c69cd0 to 0xb9c6a000)
Mar 30 22:49:01 rpi5 kernel: [33987.274317] 9cc0:                                     b9c69d04 b9c69ce0 803265ac 8079ef64
Mar 30 22:49:01 rpi5 kernel: [33987.279447] 9ce0: 00000000 b9c69dc0 ba442861 00000000 00000000 ba442860 b9c69d74 b9c69d08
Mar 30 22:49:01 rpi5 kernel: [33987.284516] 9d00: 80335f60 80326588 40000013 ffffffff b9c69d5c ba442860 b9c69e44 b9c69d28
Mar 30 22:49:01 rpi5 kernel: [33987.289597] 9d20: 8079f680 801eda18 80ffffec ba442861 00001000 00000000 10000022 b2000001
Mar 30 22:49:01 rpi5 kernel: [33987.294868] 9d40: 427e0000 00000ba4 ba442860 00000000 00001000 427e0000 00000ba4 ba442860
Mar 30 22:49:01 rpi5 kernel: [33987.300470] 9d60: 00001000 00000000 b9c69e44 b9c69d78 80355cf4 80335f14 80355868 80d162ac
Mar 30 22:49:01 rpi5 kernel: [33987.306363] 9d80: 00000001 00000000 00000000 00000000 78daaef1 ba442860 7f000000 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.312502] 9da0: 00000000 00000001 00001000 ba442861 00000001 00000ba4 427e1000 00000ba4
Mar 30 22:49:01 rpi5 kernel: [33987.318979] 9dc0: 00000000 00000000 427e0000 00001000 00000000 00000000 00000000 b640e820
Mar 30 22:49:01 rpi5 kernel: [33987.325631] 9de0: b9c69e04 b9c69df0 8078b708 8078b644 b9c69e1c ba442864 b9c69e2c b9c69e08
Mar 30 22:49:01 rpi5 kernel: [33987.332468] 9e00: 8021d0d8 8078b6f8 00000000 00000000 00000000 00000000 b640e81c 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.339453] 9e20: ba442860 ba442861 b9d94f00 b9c69ef0 b9c69f08 ba442860 b9c69e64 b9c69e48
Mar 30 22:49:01 rpi5 kernel: [33987.346512] 9e40: 80334270 80355914 80334204 00000000 ba442860 b640e81c b9c69ecc b9c69e68
Mar 30 22:49:01 rpi5 kernel: [33987.353571] 9e60: 8022036c 80334210 b97ad6e8 b9c69e78 8029a528 8029a178 00080001 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.360629] 9e80: 00000001 ffffffff b9d94f68 00000fff b640e730 b9c68000 00000000 b9c69ea8
Mar 30 22:49:01 rpi5 kernel: [33987.367679] 9ea0: 8029afa0 b9d94f00 b9c69f80 00000000 00000000 00000200 00000000 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.374733] 9ec0: b9c69edc b9c69ed0 80327bc4 8021fe9c b9c69f4c b9c69ee0 8028a6cc 80327b8c
Mar 30 22:49:01 rpi5 kernel: [33987.381303] 9ee0: 00000200 b9c69ee8 7ea75670 00000200 00000000 00000000 00000200 b9c69ee8
Mar 30 22:49:01 rpi5 kernel: [33987.384329] 9f00: 00000001 8079a468 b9d94f00 00000000 00000000 00000000 00000000 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.387358] 9f20: 00000000 00000000 00000200 b9d94f00 7ea75670 b9c69f80 80108224 b9c68000
Mar 30 22:49:01 rpi5 kernel: [33987.390380] 9f40: b9c69f7c b9c69f50 8028a7b8 8028a5f8 00000000 802aadbc b9c69f7c b9d94f00
Mar 30 22:49:01 rpi5 kernel: [33987.393401] 9f60: b9d94f00 00000200 7ea75670 80108224 b9c69fa4 b9c69f80 8028ad68 8028a728
Mar 30 22:49:01 rpi5 kernel: [33987.396429] 9f80: 00000000 00000000 00012e57 00000008 76fd3960 00000003 00000000 b9c69fa8
Mar 30 22:49:01 rpi5 kernel: [33987.399457] 9fa0: 80108080 8028ad20 00012e57 00000008 00000003 7ea75670 00000200 00000000
Mar 30 22:49:01 rpi5 kernel: [33987.402486] 9fc0: 00012e57 00000008 76fd3960 00000003 00000003 7ea75670 76fd3b48 7ea75624
Mar 30 22:49:01 rpi5 kernel: [33987.405515] 9fe0: 76fd2cf0 7ea755cc 76fa92c0 76fbc4ec 00000010 00000003 00000000 00000000
[... cut ...]
Mar 30 23:01:46 rpi5 kernel: [34753.000106] [<80327bc4>] (ext4_file_read_iter) from [<8028a6cc>] (__vfs_read+0xe0/0x130)
Mar 30 23:01:46 rpi5 kernel: [34753.003028] [<8028a6cc>] (__vfs_read) from [<8028a7b8>] (vfs_read+0x9c/0x168)
Mar 30 23:01:46 rpi5 kernel: [34753.004543] [<8028a7b8>] (vfs_read) from [<8028ad68>] (SyS_read+0x54/0xb0)
Mar 30 23:01:46 rpi5 kernel: [34753.006047] [<8028ad68>] (SyS_read) from [<80108080>] (ret_fast_syscall+0x0/0x28)
```

I have a USB harddrive connected to a powered hub. The Pi is connected via wifi only.

andrum99, 2018-04-04:
+1 I'm seeing this as well. I'm running stock Raspbian Stretch fully up-to-date. Just updated kernel using rpi-update so hopefully that will have fixed it - will report back.

andrum99, 2018-04-04:
@pelwell Is the fix for this bug the same as for #2446?

pelwell, 2018-04-04:
I think this is a different issue - the TCP bug caused failures in the TCP stack, whereas this looks like the result of a corruption of some kind (memory or filing system).

andrum99, 2018-04-04:
@pelwell

> A patch for that bug was applied to our current branch this afternoon. The latest firmware build, available via rpi-update, includes the patch.

So is that (https://github.com/raspberrypi/linux/commit/661230038a8736ba9023978c53cd21cd65739406)?

comdata, 2018-04-04:
I could add that the system is running as NAS with cable based ethernet connection and multiple hard discs connected via USB running btrfs on top of it.

Moved back to 4.9.80 and reduced the clock speed to 1200 and it seems stable for now.

andrum99, 2018-04-04:
@pelwell So to try and unmuddle things: *this* bug is unrelated to the TCP bug (#2446) and has a different, and as yet unknown cause - is that right?

(I thought the fix for the TCP bug (#2446) was https://github.com/raspberrypi/linux/commit/9d2ad143e40c38d34be86578840499a976c0a5b0?

https://github.com/raspberrypi/linux/commit/661230038a8736ba9023978c53cd21cd65739406 looks different)

comdata, 2018-04-04:
It seems better with 4.14.32 and arm_freq=1200 set in the config.txt.

Without the arm_freq set it crashes again.

bcutter, 2018-04-05:
My system also suffers these kernel crashes, they either start with
`Internal error: Oops: 5 [#1] SMP ARM` or
`Internal error: Oops - undefined instruction: 0 [#1] SMP ARM`.
After a certain time watchdog kicks in and reboots the system (Raspbian Stretch):
```
Message from watchdog:
The system will be rebooted because of error 253 = 'load average too high'
```
It´s on a brand new Pi 3 B+ (`Linux  4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux`), didn´t have that on the same SD card with Raspbian Jessie on a Pi 2 B.
Power voltage should be good (5V 3A), overclocking not available for Pi 3 B+, SD card might be - will rule this out by switching the sd card back into the previous Pi 2 B.

**Update:**
- So far System running stable back on the Pi 2 B, so microSD card and power seem to be okay.
- CPU stresstests (stressberry, 5 minutes test) were also fine on the Pi 3 B+. There is no "crash pattern", but because of the watchdog doing it´s job it´s obviously in cpu/memory load intensive situations.
**--> I´m wondering if it´s a hardware or software issue, but definitely something with the Pi 3 B+.** Sending it back, trying with another one? Waiting for software updates?

Anything else I can test / provide the experts here?

pelwell, 2018-04-05:
Please try with `sdram_freq=450` in `config.txt`. Don't bother putting the same card in a 2B - this doesn't sound like a faulty card.

bcutter, 2018-04-05:
THX, found that information [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1297394#p1297394) a few minutes ago and set "arm_freq=1200" & "sdram_freq=450", will now switch hardware back to 3 B+ and take a close look.

You ruled out it must be the sdram_freq, so "arm_freq=1200" is not necessary?

bcutter, 2018-04-06:
Okay, after some really stable hours I‘ll switch now to only cut sdram frequency.

I don’t know what’s the usual sdram freq („vcgencmd get_config int“ should show), but what we buy here for stability is performance. I don’t know how much, but... **is it a kernel/Software thing which can be fixed or is it a hardware issue, so not all Pi 3 B + suffer this?**

Depending on this I‘m thinking of sending my current hardware back, trying my luck with a new/other one...

popcornmix, 2018-04-06:
The non-turbo sdram frequency is 400MHz on all devices.
On Pi3, the turbo sdram frequency is 450MHz
On Pi3 B+, the turbo sdram frequency is 500MHz
It would be very useful to determine if this frequency is causing unreliability on your Pi.
There are other settings that can be set in software that could help (voltage, delays, schmoo) but we need to know if sdram is actually the problem.

May also be worth running memtester (sudo apt-get install memtester). That tends to be a very effective way of testing if the sdram is not reliable at a given frequency.

bcutter, 2018-04-06:
Well, even it´s just a temperature benchmark, the sdram frequency (at the end resulting in temperature, right?) seems to have no direct impact to the performance (core frequency of course has (a huge one)):

[image]

(https://ibb.co/ekvScH)

System running smooth with only set sdram freq to 450 MHz, core back on default (1.4 GHz).

Ferroin, 2018-04-06:
As a general rule, RAM speed will have near zero impact on the temperature reported by the thermal sensor because the thermal sensor is in the CPU, not the RAM.  Additionally though, 50MHz isn't likely to make much of a difference in terms of temperature difference even if you were looking at the RAM.

There's unfortunately no particularly quick and easy method to test memory performance on a Pi.  A quick and dirty option (but not very accurate) is to create a tmpfs mount, put a moderately large file on it (128M seems to work well) and then use `dd` with a 4k block size (to match the page size) to copy that to another location on the tmpfs mount and see how long it takes.  This involves at least two memcpy() operations for each block, and therefore should do a decent job of providing comparison points for the memory performance.  Even here though, I doubt that 50MHz will make much difference.

bcutter, 2018-04-06:
Update: already had my first (watchdog initiated) reboot since changed config to only lower sdram frequency.

Message from watchdog:
The system will be rebooted because of error 253 = 'load average too high'

This is „the“ typical behavior. Seems like 1.4 GHz is too much.

Ok speaking clear: I spent something between **11 to 13 hours** during the last week for investigating and fixing (e. g. almost 10 external fsck’s putting the SD card to another pi) all the issues coming from this unstable system on the Pi 3 B+. I‘m therefore **very frustrated** (first time in 5 years of Pi experience) and I think I‘ll return that piece of faulty Pi hardware and get another one. I want stability AND performance - and spend my time for more important things -.-

JamesH65, 2018-04-06:
@bcutter Would it be possible for you to send the Pi showing these symptoms to us? We would send you back a new 3B+ plus some goodies to cover the return postage? That way we could do the testing inhouse and save you some time. Please send email to info@raspberrypi.org with a FAO to James Hughes so its gets to me. Thanks.

bcutter, 2018-04-06:
@JamesH65 : sounds good, basically. Shipping costs and times might be a barrier (Germany - UK; with 9 € it´s a fourth of the Pi 3 B+ value)... but first thank you for this offer, I´ll think about it.

JamesH65, 2018-04-06:
We'll send you something on return that compensates for the postage costs. 9 € is insignificant to us to try and help figure out what the problem is, but I want to ensure you are not out of pocket.

uglymagoo, 2018-04-10:
Is there something new to report? Many Plex users have the some problem with a Raspberry Pi 3 Model B (not+). A downgrade to 4.9.x immediately fixes this problem. Has 4.14.x been pushed to the official raspbian repo? Is is possible to remove this kernel upgrade form the official repos till this issue has been fixed?

https://forums.plex.tv/discussion/315991/after-update-to-latest-movie-playback-stops-after-2-3-minutes-and-requires-reboot-of-pms

pelwell, 2018-04-10:
This issue is specific to the 3B+. If you are seeing problems on a 3B then create another issue, including the whole of any stack traces you have, a description of your setup (hardware and software) and what the Pi was doing when it crashed.

pelwell, 2018-04-11:
@bcutter We would really like to get our hands on your 3B+. Email me - phil@raspberrypi.org - and I'll explain the deal.

comdata, 2018-04-11:
@pelwell if it helps I could send you my B+ I can revert to a 3B till I get a replacement

pelwell, 2018-04-11:
Sure - use the email above.

bcutter, 2018-04-11:
@pelwell:
Since today I‘m fighting with what I expected in the first place: heavy SD card corruption. My Pi‘s eat them regularly, about one card per year (current one: 9 months...) - which’s is pretty cost intensive meanwhile running on 128 GB SDXCs. I think I‘ll outsource data storage and swap file to USB flash drives in the near future to minimize read/write access for the SD card.

So at the moment I‘m restoring from a backup to a fresh SD card. After that I need to retest the Pi 3B+ Hardware with that healthy SD card (don’t want you to send a proper working Pi 3 B+ without very good reasons), at the moment I can not ensure it‘s the SD card causing all my troubles, even it started at the time when upgrading from Jessie to Stretch and switching to Pi 3 B+. I‘ll come back to you in the next days.

To fasten my testing process: there‘s a guide on how to stress the memory which I didn’t try yet in the raspberrypi forums. I also know how to stress test CPU and storage, if I find time I will try that with different config.txt settings and check system stability during all runs - I think that‘s what you‘d also do in the first place. And the best cross check test is to run all that tests with my stable Pi 2 B of course.

bcutter, 2018-04-15:
SD card now is fine. Tried several as ram overvoltage settings, nothing worked. Only stable setting is armfreq=1200 with sdram freq=450. Will send it back, it’s just unacceptable.

pelwell, 2018-04-15:
The link between the SoC and the SDRAM has a number of adjustable parameters, and the hope is that with the right settings there will be no need to reduce the clocks.

JamesH65, 2018-04-23:
@comdata Are you and any others still seeing the USB issue? I suspect this was a 3B+ SoC/SDRAM timing issue rather than anything related to the USB.

comdata, 2018-04-23:
no the issue seems to be related to the specific unit. with the replacement
unit I can not see this issue and it survived the stress I put on it so far.

Mit freundlichen Grüßen
Christoph Mertins

comdata, 2018-04-29:
haven't seen this issue on the replacement units so will close this. Thanks everyone.

closed 2018-04-29 by comdata

JamesH65, 2018-04-29:
Great, thanks for the update, glad it's all working now.