raspberrypi/linux #2482: Network driver on RPi3 BPlus causing hung tasks when working on an NFS mount

graysky2, 2018-03-30:
**Platform/Distro:** RPi 3B+ running Arch ARM (armv7h).  
**Kernel version:** 4.14.31 (https://github.com/raspberrypi/linux/commit/b36f4e9e198477803d29861e02d3ea00fe5e09ab)
**Firmware version:** latest as I write this (https://github.com/raspberrypi/firmware/commit/c14a90333c13f507ab219d583b74a998ec11a6e7)

**Bug:** Frequent kernel oops due to blocked tasks when writing files to NFS mount.

**Details**:  When compiling,  dmesg is full of kernel oops like the below when doing so on an NFS mount.  Compiling to the micro SD card is fine.  I believe that the software (disto) on the micro SD card is NOT to blame...  if I put the same micro SD card into a RPi3 or RPi2, I can compile without error.

Again, I am using an NFS mounted partition (/scratch) on which to compile, so I'm hypothesizing that these problems are related to the network driver.

```% dmesg
...
[ 2455.534291] INFO: task ld:24879 blocked for more than 120 seconds.
[ 2455.538489]       Tainted: G         C      4.14.31-1-ARCH #1
[ 2455.542688] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2455.550990] ld              D    0 24879  24804 0x00000000
[ 2455.555379] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 2455.559662] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 2455.563990] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 2455.572326] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 2455.580865] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 2455.589272] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 2455.597837] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 2455.606295] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 2455.610675] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 2455.614999] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[ 2547.695051] nfs: server ease not responding, still trying
[ 2548.735626] nfs: server ease not responding, still trying
[ 2548.768826] nfs: server ease OK
[ 2548.796748] nfs: server ease OK
[ 2701.296329] INFO: task ld:24879 blocked for more than 120 seconds.
[ 2701.300214]       Tainted: G         C      4.14.31-1-ARCH #1
[ 2701.304061] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2701.311642] ld              D    0 24879  24804 0x00000000
[ 2701.315536] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 2701.319458] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 2701.323355] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 2701.330878] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 2701.338447] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 2701.345916] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 2701.353469] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 2701.360953] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 2701.364740] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 2701.368593] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[ 2772.976750] nfs: server ease not responding, still trying
[ 2774.331264] nfs: server ease OK
[ 2947.057892] INFO: task ld:24879 blocked for more than 120 seconds.
[ 2947.061907]       Tainted: G         C      4.14.31-1-ARCH #1
[ 2947.066031] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2947.074107] ld              D    0 24879  24804 0x00000000
[ 2947.078244] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 2947.081483] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 2947.084348] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 2947.090033] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 2947.095898] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 2947.101751] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 2947.107513] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 2947.113352] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 2947.116350] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 2947.119289] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[ 2998.258064] nfs: server ease not responding, still trying
[ 2999.352463] nfs: server ease OK
[ 3192.819075] INFO: task ld:24879 blocked for more than 120 seconds.
[ 3192.823185]       Tainted: G         C      4.14.31-1-ARCH #1
[ 3192.827330] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 3192.835447] ld              D    0 24879  24804 0x00000000
[ 3192.839604] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 3192.842832] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 3192.845750] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 3192.851476] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[... 101 lines cut ...]
[ 4790.282558] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 4790.285531] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 4790.288488] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 4790.294136] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 4790.299855] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 4790.305556] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 4790.311366] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 4790.317112] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 4790.320380] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 4790.323699] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[ 4790.330500] INFO: task ld:7636 blocked for more than 120 seconds.
[ 4790.334181]       Tainted: G         C      4.14.31-1-ARCH #1
[ 4790.338097] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 4790.346223] ld              D    0  7636   7633 0x00000000
[ 4790.350304] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 4790.354463] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 4790.358593] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 4790.366494] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 4790.374744] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 4790.383021] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 4790.391236] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 4790.399371] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 4790.403607] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 4790.407831] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
```

graysky2, 2018-03-30:
An easy way to trigger this bug (if you don't want to try compiling the kernel package) is to simply use dd to write out from `/dev/zero` to the NFS mount.  For example on my RPi3 B+:

```
# mount ease:/scratch /scratch-nfs
% dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=1000 status=progress
964689920 bytes (965 MB, 920 MiB) copied, 149 s, 6.5 MB/s

<<< it froze up after about 965 MB written >>>
<<< In dmesg I get anohter server not responding error >>>

[ 5112.824818] nfs: server ease not responding, still trying
[ 5149.707808] nfs: server ease OK
```

Now, if I swap out the micro SD and boot into a RPi 2 I have lying around, same network cable, same power supply, and repeat the commands, everything works as expected.  I think that helps to rule out the NFS server, network hardware etc. as potentially to blame.

```
# mount ease:/scratch /scratch-nfs
% dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=1000 status=progress
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 346 s, 12.1 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 357.595 s, 11.7 MB/s
dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=1000 status=progress  0.00s user 24.47s system 5% cpu 8:03.99 total
```

pelwell, 2018-03-30:
Does disabling Energy Efficient Ethernet make a difference? Add `dtparam=eee=off` to config.txt and reboot.

But before trying that you can confirm whether EEE is active using `ethtool --show-eee eth0`.

graysky2, 2018-03-30:
Great suggestion, @pelwell.  I got some very encouraging results using the faster `dd` test which passed (ran it 3x only sharing 2 of them below).  I am compiling the same package that gave the errors initially and will post back with those results.

Before:
```
# ethtool --show-eee eth0
EEE Settings for eth0:
	EEE status: enabled - active
	Tx LPI: 600 (us)
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  100baseT/Full 
	                            1000baseT/Full 
	Link partner advertised EEE link modes:  100baseT/Full 
	                                         1000baseT/Full 
```

After:
```
# ethtool --show-eee eth0
EEE Settings for eth0:
	EEE status: disabled
	Tx LPI: disabled
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  100baseT/Full 
	                                         1000baseT/Full 
```

The test with dd:
```
# mount ease:/scratch /scratch-nfs

% dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=1000 status=progress && rm fill
4169138176 bytes (4.2 GB, 3.9 GiB) copied, 97 s, 42.9 MB/s 
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 100.665 s, 41.7 MB/s
dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=1000 status=progress  0.00s user 13.79s system 13% cpu 1:40.68 total

% dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=2000 status=progress && rm fill
8380219392 bytes (8.4 GB, 7.8 GiB) copied, 198 s, 42.3 MB/s
2000+0 records in
2000+0 records out
8388608000 bytes (8.4 GB, 7.8 GiB) copied, 201.245 s, 41.7 MB/s
dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=2000 status=progress  0.00s user 27.98s system 13% cpu 3:21.25 total

% dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=2000 status=progress && rm fill
8380219392 bytes (8.4 GB, 7.8 GiB) copied, 198 s, 42.3 MB/s
2000+0 records in
2000+0 records out
8388608000 bytes (8.4 GB, 7.8 GiB) copied, 201.052 s, 41.7 MB/s
dd if=/dev/zero of=/scratch-nfs/fill bs=4M count=2000 status=progress  0.00s user 28.23s system 13% cpu 3:22.19 total
```

graysky2, 2018-03-30:
> I am compiling the same package that consistently gives the errors now and will post back with those results.

Grrrr, same error.  The package I am building is the linux kernel for raspberry pi.  It is still giving errors:
```
[ 3315.685473] INFO: task gzip:29769 blocked for more than 120 seconds.
[ 3315.685636]       Tainted: G         C      4.14.31-1-ARCH #1
[ 3315.685767] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 3315.685955] gzip            D    0 29769  29767 0x00000000
[ 3315.686127] [<80a87848>] (__schedule) from [<80a88018>] (schedule+0x3c/0xa0)
[ 3315.686299] [<80a88018>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[ 3315.686473] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[ 3315.686663] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[ 3315.686875] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[ 3315.687121] [<80233190>] (filemap_write_and_wait_range) from [<803db1c4>] (nfs_file_fsync+0x30/0x280)
[ 3315.687349] [<803db1c4>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[ 3315.687529] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[ 3315.691478] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[ 3315.695540] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[ 3402.725251] nfs: server ease not responding, still trying
[ 3403.765783] nfs: server ease not responding, still trying
[ 3404.089089] nfs: server ease OK
[ 3404.089297] nfs: server ease OK
```

graysky2, 2018-03-31:
I combined a few replies into one and tried to make it it bit more concise.  TL;DR version is that disabling EEE does not help.

pelwell, 2018-03-31:
If possible, and if it isn't already on, can you enable flow control on the switch port connected to the Pi?

graysky2, 2018-03-31:
@pelwell - All the wired connections go through an unmanaged switch.  No settings to tweak :/

mkreisl, 2018-03-31:
> Bug: Frequent kernel oops due to blocked tasks when writing files to NFS mount.

I had similar issues on SAMBA mount. But currently I can not run tests again, because I sent back my Pi3B+.

IMO current revision of Pi3B+ has serious hardware issues and I don't believe that they can be solved via software (Finally, I never was able to play a video longer than 15mins without a Kodi crash, kernel Oops, or freeze)

@pelwell and co:
Which Pi3B+ (revision) are you currently using? Parts of 0-series or parts from current production line, which customers are using now.

I still can't believe that you guys never had such issues before

JamesH65, 2018-03-31:
Current production I believe. Although I don't think there have been
many/any changes since the prototypes.

It appears that the issues are erratic, and depend on the capabilities  of
the network the device is attached to. We are trying to figure out the
exact circumstances. I suspect there are a number of issues being seen, as
often happens when a previously working  driver suddenly because used by
250k extra people over a weekend, in all sorts of new and unpredictable
ways. I have high hopes there is a software solution to this, we've always
been able to find them in the past.

Knoppix1, 2018-04-01:
Perso I back on pi 2b ...
(He boot faster with the same SD card...)

@mkreisl when your pi is back if work normally I send my pi too

zmartell, 2018-04-04:
+1 I am noticing this issue as well when reading off samba mount. Brand new RPI 3B+.

graysky2, 2018-04-04:
@pelwell - From @popcornmix's advice in https://github.com/raspberrypi/linux/issues/2442, I built:
* raspberrypi-firmware from https://github.com/raspberrypi/firmware/commit/0dff9ecc00d8114164bb9a73799793dad24d8593
* linux-raspberrypi from https://github.com/raspberrypi/linux/commit/b5b6bb9bdb5e62bc7dfc64b40bd833bffd971693

I automated that dd test I described above in a [simple script](https://gist.github.com/graysky2/e13e1ed42520b62f7994bb4eca8ac6a8) that repeats the writing out of 1G worth of zero filled file over an NFS share 32 times.  I then used [histogram.py](https://github.com/bitly/data_hacks) to compute the stats.  

With the `dtparam=eee=off` parameter set in `/boot/config.txt`  I got some consistent results:
```
# NumSamples = 32; Min = 25.46; Max = 25.75
# Mean = 25.687864; Variance = 0.002705; SD = 0.052009; Median 25.693114
```

When I removed that line (reverting to the default state of it being on, 1 of the 32 runs was really long:
```
% histogram.py -p < results.csv
# NumSamples = 36; Min = 25.34; Max = 139.44
# Mean = 28.763650; Variance = 350.005030; SD = 18.708421; Median 25.599488

% cat results.csv
25.445376403
25.720758864
25.721237277
25.635027395
25.343302226
139.443197034
25.833022001
25.557960071
25.542359739
25.582804051
25.656525006
25.655809019
25.593429933
25.632081580
25.595172430
25.585557826
25.632576300
25.588884800
25.709275077
25.626382722
25.642235963
25.603803083
25.610397819
25.559838127
25.561999023
25.569138242
25.561247940
25.548115207
25.624705979
25.563953483
25.531985345
25.632131435
25.587854852
25.605524076
25.575880103
25.611836975
```

graysky2, 2018-04-05:
OK... still experiencing the timeouts when compiling to the NFS share with eee enabled.  I am currently building https://github.com/raspberrypi/linux/commit/c2eb30683b43b13b931bd9cfef6a2a09ac7b7c1e and will test it for the compiling to NFS with eee enabled and with it disabled.

graysky2, 2018-04-06:
@pelwell - I am still getting network timeouts...

```
[11786.758187] nfs: server ease not responding, still trying
[11786.758192] nfs: server ease not responding, still trying
[11786.758206] nfs: server ease not responding, still trying
[11786.758225] nfs: server ease not responding, still trying
[11794.438353] INFO: task ld:25967 blocked for more than 120 seconds.
[11794.441599]       Tainted: G         C      4.14.32-2-ARCH #1
[11794.444867] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[11794.451496] ld              D    0 25967  25966 0x00000000
[11794.454918] [<80a87c48>] (__schedule) from [<80a88418>] (schedule+0x3c/0xa0)
[11794.458408] [<80a88418>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[11794.461670] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[11794.468043] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[11794.474533] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[11794.481165] [<80233190>] (filemap_write_and_wait_range) from [<803db254>] (nfs_file_fsync+0x30/0x280)
[11794.487571] [<803db254>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[11794.494001] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[11794.497339] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[11794.500698] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[11831.579322] nfs: server ease OK
[11831.579326] nfs: server ease OK
[11831.583067] nfs: server ease OK
[11831.583118] nfs: server ease OK
[12040.199240] INFO: task ld:27693 blocked for more than 120 seconds.
[12040.202836]       Tainted: G         C      4.14.32-2-ARCH #1
[12040.206449] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[12040.213627] ld              D    0 27693  27692 0x00000000
[12040.217311] [<80a87c48>] (__schedule) from [<80a88418>] (schedule+0x3c/0xa0)
[12040.220971] [<80a88418>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[12040.223568] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[12040.228677] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[12040.233740] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[12040.238903] [<80233190>] (filemap_write_and_wait_range) from [<803db254>] (nfs_file_fsync+0x30/0x280)
[12040.244189] [<803db254>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[12040.249445] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[12040.252070] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[12040.254713] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
[12101.639302] nfs: server ease not responding, still trying
[12101.639311] nfs: server ease not responding, still trying
[12101.639328] nfs: server ease not responding, still trying
[12142.599536] nfs: server ease not responding, still trying
[12143.639966] nfs: server ease not responding, still trying
[12143.900616] nfs: server ease OK
[12143.900633] nfs: server ease OK
[12143.909707] nfs: server ease OK
[12143.917548] nfs: server ease OK
[12143.917848] nfs: server ease OK
[12408.840196] nfs: server ease not responding, still trying
[12408.840200] nfs: server ease not responding, still trying
[12408.840228] nfs: server ease not responding, still trying
[12408.840248] nfs: server ease not responding, still trying
[12408.840274] nfs: server ease not responding, still trying
[12408.840412] INFO: task ld:29538 blocked for more than 120 seconds.
[12408.840421]       Tainted: G         C      4.14.32-2-ARCH #1
[12408.840424] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[12408.840430] ld              D    0 29538  29537 0x00000000
[12408.840493] [<80a87c48>] (__schedule) from [<80a88418>] (schedule+0x3c/0xa0)
[12408.840514] [<80a88418>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[12408.840541] [<8015c138>] (io_schedule) from [<80230be4>] (wait_on_page_bit+0x110/0x15c)
[12408.840559] [<80230be4>] (wait_on_page_bit) from [<80230d0c>] (__filemap_fdatawait_range+0xdc/0x128)
[12408.840574] [<80230d0c>] (__filemap_fdatawait_range) from [<80233190>] (filemap_write_and_wait_range+0x54/0x88)
[12408.840596] [<80233190>] (filemap_write_and_wait_range) from [<803db254>] (nfs_file_fsync+0x30/0x280)
[12408.840618] [<803db254>] (nfs_file_fsync) from [<802d5dcc>] (vfs_fsync+0x24/0x2c)
[12408.840635] [<802d5dcc>] (vfs_fsync) from [<8029d758>] (filp_close+0x2c/0x80)
[12408.840648] [<8029d758>] (filp_close) from [<8029d7cc>] (SyS_close+0x20/0x48)
[12408.840665] [<8029d7cc>] (SyS_close) from [<80107ce0>] (ret_fast_syscall+0x0/0x4c)
```

mkreisl, 2018-04-06:
Same here, nothing changed. Still absolutely unstable, unreliable and completely unusable, the Pi3B+

JamesH65, 2018-04-06:
Odd, got one on my desk that is working fine. I think you forget to add "In the circumstances I am using it".

Anyway, issues still being looked at both here and at Microchip. There was a patch on the linux netdev list today for this chips driver (lan78xxx) for EEE which may well help, that will need to be tried. It's not like we are just sitting here twiddling our thumbs.

mkreisl, 2018-04-06:
> Anyway, issues still being looked at both here and at Microchip. There was a patch on the linux netdev list today for this chips driver (lan78xxx) for EEE which may well help, that will need to be tried. It's not like we are just sitting here twiddling our thumbs.

Seems you're getting fire under your a.. now :smile: 

IMO you're looking on the wrong place. LAN issues are only the top of the iceberg

I was already reporting, that system is still unstable after that dump microchip is powered off and all traffic is going over wlan device. System still freezing randomly. So, before I'm better informed, I would say the hole Pi3B+ design is a huge issue

pelwell, 2018-04-06:
Some users who reported problems (and there honestly haven't been that many, but they are shouting loudly) have had success with adding `sdram_freq=450` to config.txt. I would recommend anybody with stability problems (anything not obviously network related) to do the same.

mkreisl, 2018-04-06:
> Some users who reported problems (and there honestly haven't been that many, but they are shouting loudly) have had success with adding sdram_freq=450 to config.txt. I would recommend anybody with stability problems (anything not obviously network related) to do the same.

What's the default for Pi3B+. Cant find it [here](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md)

pelwell, 2018-04-06:
500 turbo, 400 normal

graysky2, 2018-04-06:
> For reference, here is the script to automate the replicate compile jobs.

@pelwell - I have some hard data now.  I ran the make benchmark writing out to the NFS share under 2 conditions, once with eee disabled and once with it enabled.  There is a clear trend: eee is causing problems.

Here are 9 or 10 replicates running `make zImage` with all times reported in minutes.
Several trends from these data:
* Average time to compile is nearly double with eee enabled.
* Standard deviation and variance is much worse with eee enabled (more unpredictable compile times).
```
% histogram.py < eee_on_zimage 
# NumSamples = 9; Min = 9.77; Max = 29.07
# Mean = 18.573025; Variance = 86.102905; SD = 9.279165; Median 10.764777
```
vs
```
% histogram.py < eee_off_zimage
# NumSamples = 10; Min = 9.91; Max = 10.87
# Mean = 10.178291; Variance = 0.067048; SD = 0.258936; Median 10.166035
```

Here are 9 or 10 replicates running `make modules` with all times reported in minutes.
```
% histogram.py < eee_on_modules 
# NumSamples = 9; Min = 25.21; Max = 67.19
# Mean = 51.765753; Variance = 218.212739; SD = 14.772026; Median 46.494882
```
vs
```
% histogram.py < eee_off_modules
# NumSamples = 9; Min = 26.33; Max = 49.60
# Mean = 33.328529; Variance = 42.429103; SD = 6.513763; Median 32.126122
```
The same trends from these data:
* Average time to compile is about 1.5x longer with eee enabled.
* Standard deviation and variance is much worse with eee enabled (more unpredictable compile times).

graysky2, 2018-04-06:
@mkreisl - Please keep this issue on task... it's scoped for network writes not for general stability.  Open a new task for that.

mkreisl, 2018-04-06:
@graysky2 Oops, sorry for tainting your thread

graysky2, 2018-04-06:
A potential work-around: don't totally disable EEE, but set `dtparam=tx_lpi_timer=10000` in `/boot/config.txt` which I did and found nearly identical results in the `make zImage` benchmark to having EEE totally disabled:
```
# NumSamples = 12; Min = 9.90; Max = 10.19
# Mean = 10.089245; Variance = 0.007412; SD = 0.086094; Median 10.119596
```

mkreisl, 2018-04-06:
All those EEE settings doesn't help for me, because my router/switch does not support EEE (most router with integrated switch does not support it) and I'm still getting nfs timeouts even if EEE is completely disabled, or I'm getting

```
Apr  6 16:11:14 kmxbilr2 kernel: [  837.345227] CIFS VFS: sends on sock aa2921c0 stuck for 15 seconds
Apr  6 16:11:14 kmxbilr2 kernel: [  837.345261] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:11:30 kmxbilr2 kernel: [  852.705497] CIFS VFS: sends on sock aa2921c0 stuck for 15 seconds
Apr  6 16:11:30 kmxbilr2 kernel: [  852.705532] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:11:30 kmxbilr2 kernel: [  852.833704] CIFS VFS: Free previous auth_key.response = 99685c00
Apr  6 16:11:55 kmxbilr2 kernel: [  878.305932] CIFS VFS: sends on sock aa29c380 stuck for 15 seconds
Apr  6 16:11:55 kmxbilr2 kernel: [  878.305972] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:12:11 kmxbilr2 kernel: [  893.666123] CIFS VFS: sends on sock aa29c380 stuck for 15 seconds
Apr  6 16:12:11 kmxbilr2 kernel: [  893.666156] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:12:26 kmxbilr2 kernel: [  909.026351] CIFS VFS: sends on sock aa29c380 stuck for 15 seconds
Apr  6 16:12:26 kmxbilr2 kernel: [  909.026382] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:12:41 kmxbilr2 kernel: [  924.386541] CIFS VFS: sends on sock aa29c380 stuck for 15 seconds
Apr  6 16:12:41 kmxbilr2 kernel: [  924.386573] CIFS VFS: Error -11 sending data on socket to server
Apr  6 16:12:41 kmxbilr2 kernel: [  924.484318] CIFS VFS: Free previous auth_key.response = a7910f00
```
if using SAMBA mount instead of NFS mount and after some time process that writes to share stucks and becomes uninterruptable 'D' state forever

graysky2, 2018-04-06:
@mkreisl - Are you booted into the same kernel and are you using the same firmware commit that I am?
Kernel: https://github.com/raspberrypi/linux/commit/c2eb30683b43b13b931bd9cfef6a2a09ac7b7c1e
Firmware: https://github.com/raspberrypi/firmware/commit/0dff9ecc00d8114164bb9a73799793dad24d8593

mkreisl, 2018-04-06:
@graysky2 
Kernel: yes (XBian built based on bcm2709_defconfig)
Firmware: yes, excatly the same version

graysky2, 2018-04-06:
@mkreisl - not sure what to say then.... perhaps you have a different issue.  As a control, have you tried the same stuff with another older RPi?  Like a 2 or 3?

mkreisl, 2018-04-07:
@graysky2 Sure, I'm running same procedure on Pi1, 2 and 3 (without +) since years without any problem.

mkreisl, 2018-04-07:
@graysky2 In short words, I can explain what it does

1) mount network share (sshfs, nfs or samba)
2) create image on this share, big enough to backup data from root/boot fs into it
3) create partition in image (vfat for boot, btrfs for root)
4) copy boot partition into mounted image (loop device) 
5) copy  all subvolumes into mounted image (using btrfs send/receive or tar, both tested)
6) close everything and umount share

From 1 to 4 it always works, and within 5 it stucks always, but not on the same subvolume
And, it does not matter if source fs is on sd, usb disk or iSCSI target

graysky2, 2018-04-07:
I can complete these steps on mine without error except I don't do much with BTRFS.  Can you complete the steps with ext4?  I can:

```
# mount nfs share
mount -t nfs4 ease:/scratch /opt/armv7h/scratch
mount | grep opt
ease:/scratch on /opt/armv7/scratch type nfs4 
(rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=10.9.8.108,local_lock=none,addr=10.9.8.128)

# make image and format it
cd /opt/armv7h/scratch
truncate -s 5G image.img
mkfs.ext4 -F image.img
mkdir mp

# mount it
mount -o loop image.img mp
chown facade:users mp

# add 1.0 GB of data to image
dd if=/dev/zero of=/opt/armv7/scratch/mp/zerofill bs=4M count=250 status=progress

ls -l mp
total 1024020
drwx------ 2 facade users      16384 Apr  7 12:30 lost+found
-rw-r--r-- 1 facade users 1048576000 Apr  7 12:32 zerofill

umount mp
```

mkreisl, 2018-04-07:
> What is your NFS or samba server? How are these configured? I can complete these steps on mine without error except I don't do much with BTRFS. Can you complete the steps with ext4?

Running Debian Stretch as NFS and SAMBA server. Nothing special configured AFAIR
Currently it's not possible to run tests with ext4, don't have any configuation based on ext4. Will do more testing next days, actually I'm testing wlan only configuration.

pelwell, 2018-04-07:
You haven't said if you have changed sdram_freq or arm_freq yet.

mkreisl, 2018-04-07:
@pelwell 

> You haven't said if you have changed sdram_freq or arm_freq yet.

Ehm @graysky2 told me that I should not post such topics here :disappointed: 

Yes, changing sdram clock to 450 did it, setting arm clock to 1200 I tested 2 weeks ago without success

But, reducing sdram clock did not solve ethernet network issues discussed here

graysky2, 2018-04-07:
> Ehm @graysky2 told me that I should not post such topics here disappointed

No, just reminded you that this task is about network write issues, if one of the developers asks you to try something, do it, just don't add a bunch of unrelated issues on top of this one.

pelwell, 2018-04-07:
Exactly. There appear to be two classes of errors - network problems and memory corruption. I'm satisfied this is purely a networking problem.

EEE does seem to worsen network throughout, enabling flow control improves it, and we haven't ruled out further driver problems.

graysky2, 2018-04-07:
Based on @mkreisl's observation about samba shares, I did the same `make zImage` benchmark on a samba share (samba version 4.7.6) and found inconsistent performance (nothing in dmesg) though:

```
# NumSamples = 4; Min = 462.65; Max = 1251.20
# Mean = 895.113710; Variance = 124647.496919; SD = 353.054524; Median 933.305226

1,zImage,631.770020346
2,zImage,462.645222405
3,zImage,1251.199164634
4,zImage,1234.840431102
```

mkreisl, 2018-04-07:
@graysky2 modified your script as follows and let it run

```
#!/bin/bash

cd /srv/backup
mount | grep /srv/backup

truncate -s 8G image.img
mkfs.ext4 -F image.img
mkdir -p img

mount -o loop image.img img

to='--warning=none --acls --xattrs'
tar c --one-file-system $to -C / . | tar x $to -v -C /srv/backup/img

ls -l img

umount img && rmdir img && rm -f image.img
```

Destination is ext4, but source is still BTRFS

It ends up always with `nfs: server kmcubie not responding, timed out`

```
[ 1692.649145] INFO: task loop0:9479 blocked for more than 120 seconds.
[ 1692.649159]       Tainted: G         C      4.14.32+ #1
[ 1692.649167] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1692.649175] loop0           D    0  9479      2 0x00000000
[ 1692.649219] [<808c4124>] (__schedule) from [<808c47d4>] (schedule+0x58/0xb8)
[ 1692.649240] [<808c47d4>] (schedule) from [<8014be98>] (io_schedule+0x20/0x40)
[ 1692.649262] [<8014be98>] (io_schedule) from [<80221fb8>] (wait_on_page_bit+0x108/0x128)
[ 1692.649279] [<80221fb8>] (wait_on_page_bit) from [<802220bc>] (__filemap_fdatawait_range+0xe4/0x114)
[ 1692.649297] [<802220bc>] (__filemap_fdatawait_range) from [<80224614>] (filemap_write_and_wait_range+0x68/0xa4)
[ 1692.649316] [<80224614>] (filemap_write_and_wait_range) from [<803a0184>] (nfs_file_fsync+0x3c/0x308)
[ 1692.649335] [<803a0184>] (nfs_file_fsync) from [<802c7644>] (vfs_fsync_range+0x68/0xbc)
[ 1692.649356] [<802c7644>] (vfs_fsync_range) from [<802c76cc>] (vfs_fsync+0x34/0x3c)
[ 1692.649383] [<802c76cc>] (vfs_fsync) from [<7f007e80>] (loop_queue_work+0x574/0xb98 [loop])
[ 1692.649438] [<7f007e80>] (loop_queue_work [loop]) from [<8013dff4>] (kthread_worker_fn+0xbc/0x1e0)
[ 1692.649459] [<8013dff4>] (kthread_worker_fn) from [<7f005738>] (loop_kthread_worker_fn+0x34/0x194 [loop])
[ 1692.649480] [<7f005738>] (loop_kthread_worker_fn [loop]) from [<8013dec4>] (kthread+0x13c/0x16c)
[ 1692.649502] [<8013dec4>] (kthread) from [<8010834c>] (ret_from_fork+0x14/0x28)
[ 1784.809482] nfs: server kmcubie not responding, timed out
[ 1784.809540] nfs: server kmcubie not responding, timed out
[ 1784.809563] nfs: server kmcubie not responding, timed out
[ 1813.481779] INFO: task kworker/u8:2:7543 blocked for more than 120 seconds.
[ 1813.481794]       Tainted: G         C      4.14.32+ #1
[ 1813.481801] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1813.481810] kworker/u8:2    D    0  7543      2 0x00000000
[ 1813.481844] Workqueue: writeback wb_workfn (flush-7:0)
[ 1813.481885] [<808c4124>] (__schedule) from [<808c47d4>] (schedule+0x58/0xb8)
[ 1813.481903] [<808c47d4>] (schedule) from [<8014be98>] (io_schedule+0x20/0x40)
[ 1813.481923] [<8014be98>] (io_schedule) from [<805688c8>] (blk_mq_get_tag+0x1a4/0x2fc)
[ 1813.481944] [<805688c8>] (blk_mq_get_tag) from [<805634e8>] (blk_mq_get_request+0x134/0x43c)
[ 1813.481961] [<805634e8>] (blk_mq_get_request) from [<805661a8>] (blk_mq_make_request+0xec/0x758)
[ 1813.481980] [<805661a8>] (blk_mq_make_request) from [<80558d50>] (generic_make_request+0xe8/0x260)
[ 1813.481999] [<80558d50>] (generic_make_request) from [<80558f80>] (submit_bio+0xb8/0x1a8)
[ 1813.482018] [<80558f80>] (submit_bio) from [<8035dec8>] (ext4_io_submit+0x58/0x64)
[ 1813.482038] [<8035dec8>] (ext4_io_submit) from [<80342b6c>] (ext4_writepages+0x718/0x998)
[ 1813.482056] [<80342b6c>] (ext4_writepages) from [<80234668>] (do_writepages+0x30/0x9c)
[ 1813.482074] [<80234668>] (do_writepages) from [<802c20d8>] (__writeback_single_inode+0x44/0x554)
[ 1813.482092] [<802c20d8>] (__writeback_single_inode) from [<802c2b74>] (writeback_sb_inodes+0x238/0x4c0)
[ 1813.482111] [<802c2b74>] (writeback_sb_inodes) from [<802c2e8c>] (__writeback_inodes_wb+0x90/0xd0)
[ 1813.482128] [<802c2e8c>] (__writeback_inodes_wb) from [<802c31b8>] (wb_writeback+0x2ec/0x3bc)
[ 1813.482144] [<802c31b8>] (wb_writeback) from [<802c3c24>] (wb_workfn+0x1d4/0x570)
[ 1813.482167] [<802c3c24>] (wb_workfn) from [<80137b18>] (process_one_work+0x140/0x4c4)
[ 1813.482187] [<80137b18>] (process_one_work) from [<80137ef8>] (worker_thread+0x5c/0x568)
[ 1813.482206] [<80137ef8>] (worker_thread) from [<8013dec4>] (kthread+0x13c/0x16c)
[... 44 lines cut ...]

But it does not always block everything infinitely bc mounted loop device will randomly become ro (after a looong time doing nothing) and tar does not continue copying, just throws a lot of error messages to the screen or tar continues copying until nfs is blocked again or third case is blocking infinitely

```
root@kmxbilr2 ~/bin # ps ax | grep tar\ 
13829 pts/3    S+     0:18 tar c --one-file-system --warning=none --acls --xattrs -C / .
13830 pts/3    D+     0:20 tar x --warning=none --acls --xattrs -C /srv/backup/img
15679 pts/1    S+     0:00 grep --color=auto tar 
root@kmxbilr2 ~/bin # 
```

This is my eee setting:

```
root@kmxbilr2 ~/bin # ethtool --show-eee eth0
EEE Settings for eth0:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported

root@kmxbilr2 ~/bin # 
```

mkreisl, 2018-04-07:
So, I run above test three times:

1st run: loop device becomes r/o
2nd run: test finished
3rd run: test blocked infinitely

but getting always messages like this: `NFO: task loop0:9479 blocked for more than 120 seconds.`

graysky2, 2018-04-08:
> Destination is ext4, but source is still BTRFS
> It ends up always with nfs: server kmcubie not responding, timed out

I don't use BTRFS nor do I have access to your specific filesystem, so let's come up with a standard script anyone can use provided that they have an NFS server and can mount to it.

[Here](https://gist.github.com/graysky2/c97f47b8f1a0f5d8ae3d664780ddf20d) is a such a script.

When I run it on my 3B+, I see that the original bug I for which I opened this ticket is STILL present:
From dmsg:
```
[   90.991309] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[  699.983054] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[  858.079796] INFO: task jbd2/loop0-8:633 blocked for more than 120 seconds.
[  858.079980]       Tainted: G         C      4.14.32-2-ARCH #1
[  858.080112] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  858.080290] jbd2/loop0-8    D    0   633      2 0x00000000
[  858.080451] [<80a87c48>] (__schedule) from [<80a88418>] (schedule+0x3c/0xa0)
[  858.080627] [<80a88418>] (schedule) from [<8015c138>] (io_schedule+0x14/0x3c)
[  858.080801] [<8015c138>] (io_schedule) from [<80a88d28>] (bit_wait_io+0xc/0x60)
[  858.080975] [<80a88d28>] (bit_wait_io) from [<80a8899c>] (__wait_on_bit+0x60/0xb8)
[  858.081158] [<80a8899c>] (__wait_on_bit) from [<80a88a68>] (out_of_line_wait_on_bit+0x74/0xa0)
[  858.085451] [<80a88a68>] (out_of_line_wait_on_bit) from [<803be5b4>] (jbd2_journal_commit_transaction+0xf6c/0x17d8)
[  858.094108] [<803be5b4>] (jbd2_journal_commit_transaction) from [<803c3ea4>] (kjournald2+0xe0/0x2e4)
[  858.102800] [<803c3ea4>] (kjournald2) from [<8014e708>] (kthread+0x138/0x150)
[  858.107318] [<8014e708>] (kthread) from [<80107db8>] (ret_from_fork+0x14/0x3c)
[ 1633.827852] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
```

Output of the script:
```
# /root/bin/nfstest 
Writting round 1
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 605 s, 6.9 MB/s 
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 605.129 s, 6.9 MB/s
total 4096020
drwx------ 2 root root      16384 Apr  8 12:54 lost+found
-rw-r--r-- 1 root root 4194304000 Apr  8 13:04 rnd.1
Round 1 done
Writting round 2
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 930 s, 4.5 MB/s 
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 929.514 s, 4.5 MB/s
total 4096020
drwx------ 2 root root      16384 Apr  8 13:05 lost+found
-rw-r--r-- 1 root root 4194304000 Apr  8 13:20 rnd.2
Round 2 done
Writting round 3
2990538752 bytes (3.0 GB, 2.8 GiB) copied, 679 s, 4.4 MB/s 
```

It is currently stuck on round 3.  System is still up but not very responsive.

graysky2, 2018-04-08:
Well, the important thing is that you are independently able to reproduce this bug.

> ... and your network performance is ridiculous slow
Just on the 3B+ and I believe it is due to the locks ... the network is all cat6 going through an SG100-16.  Here is an x86_64 box doing the same 4G write:
```
% dd if=/dev/zero of=/mnt/media/test/fill bs=4M count=1000 status=progress
4177526784 bytes (4.2 GB, 3.9 GiB) copied, 31 s, 135 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 35.8983 s, 117 MB/s
```

lategoodbye, 2018-04-09:
@graysky2 In order to identify a locking issue you could try to recompile your kernel with the following options:
```
Kernel hacking  --->
  Lock Debugging (spinlocks, mutexes, etc...)  --->
     Lock debugging: prove locking correctness (CONFIG_PROVE_LOCKING=y)
     Sleep inside atomic section checking (CONFIG_DEBUG_ATOMIC_SLEEP=y)
```

graysky2, 2018-04-09:
You bet.  Building now.

```
┌── Lock Debugging (spinlocks, mutexes, etc...) ─────────────────────────────────────────────────────────────────────────────┐
 │                                                                                                                            │
 │                               -*- RT Mutex debugging, deadlock detection                                                   │
 │                               -*- Spinlock and rw-lock debugging: basic checks                                             │
 │                               -*- Mutex debugging: basic checks                                                            │
 │                               [ ] Wait/wound mutex debugging: Slowpath testing                                             │
 │                               -*- Lock debugging: detect incorrect freeing of live locks                                   │
 │                               [*] Lock debugging: prove locking correctness                                                │
 │                               [ ] Lock usage statistics                                                                    │
 │                               [ ] Lock dependency engine debugging (NEW)                                                   │
 │                               [*] Sleep inside atomic section checking                                                     │
 │                               [ ] Locking API boot-time self-tests                                                         │
 │                               < > torture tests for locking                                                                │
 │                               < > Wait/wound mutex selftests                                                               │
```

graysky2, 2018-04-09:
@lategoodbye @pelwell - I never thought to see what what going on the nfs server's side of things.  The dmesg log is spammed with hundreds of lines like this:
```
[ 3624.508074] nfsd4_validate_stateid: 119 callbacks suppressed
[ 3624.508074] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.508938] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.509754] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.510577] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.511396] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.512229] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.513059] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.513872] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.514688] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3624.515514] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 3767.535787] nfsd4_validate_stateid: 119 callbacks suppressed
[ 3767.535788] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.423284] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.423887] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.424487] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.425385] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.425979] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.426585] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.427501] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.428322] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4105.429125] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4218.046249] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
[ 4218.047093] NFSD: client 10.9.8.108 testing state ID with incorrect client ID
```

graysky2, 2018-04-09:
@lategoodbye - I don't think the kernel has what you need... I booted into it but get this on lock up:
```
[  532.399675] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[  858.099808] INFO: task jbd2/loop0-8:859 blocked for more than 120 seconds.
[  858.100037]       Tainted: G        WC      4.14.32-3-ARCH #1
[  858.100215] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  858.100453] jbd2/loop0-8    D    0   859      2 0x00000000
[  858.100665] [<80baebf0>] (__schedule) from [<80baf72c>] (schedule+0x50/0xc0)
[  858.100892] [<80baf72c>] (schedule) from [<80168230>] (io_schedule+0x14/0x3c)
[  858.101124] [<80168230>] (io_schedule) from [<80bb00f4>] (bit_wait_io+0xc/0x60)
[  858.101354] [<80bb00f4>] (bit_wait_io) from [<80bafd68>] (__wait_on_bit+0x60/0xb8)
[  858.101594] [<80bafd68>] (__wait_on_bit) from [<80bafe34>] (out_of_line_wait_on_bit+0x74/0xa0)
[  858.101868] [<80bafe34>] (out_of_line_wait_on_bit) from [<80420fe0>] (jbd2_journal_commit_transaction+0x1470/0x1ef8)
[  858.102191] [<80420fe0>] (jbd2_journal_commit_transaction) from [<80426e70>] (kjournald2+0xdc/0x2d0)
[  858.102472] [<80426e70>] (kjournald2) from [<80157430>] (kthread+0x148/0x164)
[  858.102697] [<80157430>] (kthread) from [<8010824c>] (ret_from_fork+0x14/0x28)
[  858.102923] INFO: lockdep is turned off.
```

lategoodbye, 2018-04-09:
@graysky2 This means you build the kernel with these options, but it didn't trigger any warning except of the hung? I tought it would be worth to try it.

graysky2, 2018-04-09:
@lategoodbye - Perhaps I didn't compile with enough of these options enabled?

lategoodbye, 2018-04-09:
@graysky2 If you want to doublecheck it, search for "Lock dependency validator" within dmesg. The default RPI kernel doesn't have it.

graysky2, 2018-04-09:
>If you want to doublecheck it, search for "Lock dependency validator" within dmesg. The default RPI kernel doesn't have it.

It's there:
```
% dmesg | grep -i validator
[    0.004391] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
```

> Edit: The config above looks good.
Edit2: lockdep is disabled because of the hung.

OK... so no need to recompile.

lategoodbye, 2018-04-10:
@graysky2 I plan to submit RPi 3+ support to Linux Mainline soon. Any chance for you to give it a try?

graysky2, 2018-04-10:
@lategoodbye -  Longterm, I need the vendor libraries supplied by the raspberrypi-firmware package  which I believe are tied to the 4.14.x kernel.  I might have some time over the weekend to compile/use a mainline version of the kernel.  I believe that would be aarch64 in Arch ARM.

lategoodbye, 2018-04-10:
@graysky2 Sure. It would help to narrow down, if this is a downstream specific issue. It would be the best to use the same rootfs, which currently causes this issue and only the replace the kernel and device tree BLOB. Mainline kernel support both 32 and 64 bit.

Thanks in advance
Stefan

graysky2, 2018-04-12:
> [Here](https://gist.github.com/graysky2/c97f47b8f1a0f5d8ae3d664780ddf20d) is a such a script.

@pelwell @mkreisl  - More evidences that the bug we are seeking is in the wired LAN driver: if I use the on board Wi-Fi, I am able to successfully complete this without kernel oops or lockup.

5 GHz connection:
```
% cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   56.  -54.  -256        0      0      0     19      0        0
```

Running the script:
```
# /root/bin/nfstest
Writting round 1
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 358 s, 11.7 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 358.232 s, 11.7 MB/s
total 4096020
drwx------ 2 root root      16384 Apr 12 15:03 lost+found
-rw-r--r-- 1 root root 4194304000 Apr 12 15:09 rnd.1
Round 1 done
Writting round 2
4185915392 bytes (4.2 GB, 3.9 GiB) copied, 358 s, 11.7 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 357.618 s, 11.7 MB/s
total 4096020
drwx------ 2 root root      16384 Apr 12 15:09 lost+found
-rw-r--r-- 1 root root 4194304000 Apr 12 15:15 rnd.2
Round 2 done
Writting round 3
4181721088 bytes (4.2 GB, 3.9 GiB) copied, 357 s, 11.7 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 357.173 s, 11.7 MB/s
total 4096020
drwx------ 2 root root      16384 Apr 12 15:15 lost+found
-rw-r--r-- 1 root root 4194304000 Apr 12 15:21 rnd.3
Round 3 done
Writting round 4
4181721088 bytes (4.2 GB, 3.9 GiB) copied, 357 s, 11.7 MB/s
1000+0 records in
1000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 357.512 s, 11.7 MB/s
total 4096020
drwx------ 2 root root      16384 Apr 12 15:21 lost+found
-rw-r--r-- 1 root root 4194304000 Apr 12 15:27 rnd.4
Round 4 done
/root/bin/nfstest  0.35s user 105.39s system 7% cpu 24:31.40 total
```

And dmesg:
```
[  +2.033113] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[  +0.001800] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28 
[  +0.081575] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  +0.000902] brcmfmac: power management disabled
[  +3.238507] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[Apr12 15:03] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[Apr12 15:08] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000132] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000100] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000093] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
[  +0.000117] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000106] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000090] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000090] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000089] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.000086] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[Apr12 15:09] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[Apr12 15:10] net_ratelimit: 73 callbacks suppressed
[  +0.000008] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.013994] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
[  +0.009320] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xff
[  +0.010984] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.009323] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008868] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008751] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008961] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[Apr12 15:14] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.009775] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008504] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008744] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008674] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008638] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0x0
[  +0.008662] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
[  +0.008604] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
[  +0.008401] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
[  +0.008542] brcmfmac: brcmf_proto_bcdc_hdrpull: wlan0: non-BCDC packet received, flags 0xfb
```

mkreisl, 2018-04-12:
@graysky2 Thanks for confirming that wlan is ok
I was posting this multiple times already :smile:

mkreisl, 2018-04-15:
No news about this annoying issue. Hmmm,  seems that you guys from Raspberry Pi foundation having absolutely no idea how to fix it :angry: 

Extremely disappointing

JamesH65, 2018-04-15:
You do know it's the weekend, right? Even RPI employees have the weekend off. We have multiple engineers working on this. I'm sure there will be progress in the near future. Meanwhile, perhaps laying off the sarcasm might help.

pelwell, 2018-04-15:
Really? I find it highly motivating.

pelwell, 2018-05-04:
Microchip pointed us at a patch that should prevent link drops when using EEE with long cables. Testing here has shown that it completely fixes the throughput EEE-related throughput issues (caused by link drops) I have observed (the cable in question is about 40 metres long). Since the patch tweaks a lot of PHY registers it may also make a difference with shorter cables. The patch (and a few pre-requisites) is in the rpi-4.14.y branch and will be picked up by the next firmware build.

pelwell, 2018-05-04:
Yes, that's correct - the kernel commit you reference is the significant one.

graysky2, 2018-05-04:
@pelwell - Great, building now.  Will let you know.

[46 more replies]

closed 2018-06-26 by JamesH65