raspberrypi/linux #2484: Unable to handle kernel NULL pointer dereference at virtual address 00000028

Darren, 2018-04-03:
```bash
uname -a
Linux rpi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```
```
Apr  2 21:31:03 rpi kernel: [40400.681540] Unable to handle kernel NULL pointer dereference at virtual address 00000028
Apr  2 21:31:03 rpi kernel: [40400.686375] pgd = b6b14000
Apr  2 21:31:03 rpi kernel: [40400.688685] [00000028] *pgd=330cd835, *pte=00000000, *ppte=00000000
Apr  2 21:31:03 rpi kernel: [40400.691134] Internal error: Oops: 17 [#1] SMP ARM
Apr  2 21:31:03 rpi kernel: [40400.693456] Modules linked in: iptable_filter tun xt_REDIRECT nf_nat_redirect xt_set xt_multiport ip_set_hash_net ip_set nfnetlink ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack cmac bnep hci_uart bluetooth ecdh_generic brcmfmac brcmutil cfg80211 cdc_ether rfkill r8152 snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
Apr  2 21:31:03 rpi kernel: [40400.709277] CPU: 0 PID: 5292 Comm: udpxy Tainted: G         C      4.14.30-v7+ #1102
Apr  2 21:31:03 rpi kernel: [40400.714799] Hardware name: BCM2835
Apr  2 21:31:03 rpi kernel: [40400.717514] task: b8e43c00 task.stack: b30a6000
Apr  2 21:31:03 rpi kernel: [40400.720270] PC is at tcp_push+0x44/0x148
Apr  2 21:31:03 rpi kernel: [40400.723050] LR is at tcp_sendmsg_locked+0x308/0xdfc
Apr  2 21:31:03 rpi kernel: [40400.725791] pc : [<806ce384>]    lr : [<806d1950>]    psr: 40000013
Apr  2 21:31:03 rpi kernel: [40400.728559] sp : b30a7d00  ip : b731873c  fp : b30a7d24
Apr  2 21:31:03 rpi kernel: [40400.731300] r10: b69d9500  r9 : ffffffe0  r8 : b731873c
Apr  2 21:31:03 rpi kernel: [40400.734059] r7 : 00000000  r6 : 000005a8  r5 : 00000000  r4 : b7318640
Apr  2 21:31:03 rpi kernel: [40400.736785] r3 : 00000001  r2 : 000005a8  r1 : 00000000  r0 : b6acc300
Apr  2 21:31:03 rpi kernel: [40400.739398] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr  2 21:31:03 rpi kernel: [40400.742102] Control: 10c5383d  Table: 36b1406a  DAC: 00000055
Apr  2 21:31:03 rpi kernel: [40400.744792] Process udpxy (pid: 5292, stack limit = 0xb30a6210)
Apr  2 21:31:03 rpi kernel: [40400.747474] Stack: (0xb30a7d00 to 0xb30a8000)
Apr  2 21:31:03 rpi kernel: [40400.750092] 7d00: 00003e38 806d1930 00000058 b30a7e10 b7318640 00000058 b30a7dbc b30a7d28
Apr  2 21:31:03 rpi kernel: [40400.755247] 7d20: 806d1950 806ce34c 00003e38 b30a7d38 8048bda0 80789524 80c069c4 80d0a08c
Apr  2 21:31:03 rpi kernel: [40400.760389] 7d40: 80c049c0 b30a6028 00000000 b8e4425c 00000001 00000000 00000000 00000000
Apr  2 21:31:03 rpi kernel: [40400.765546] 7d60: b30a7d8c 00000200 00000001 b7318790 000005a8 00000000 00000000 00003e38
Apr  2 21:31:03 rpi kernel: [40400.770865] 7d80: 000001a1 b30a7d90 801235c4 801e0000 b7318640 b7318640 00000524 b30a7e10
Apr  2 21:31:03 rpi kernel: [40400.776500] 7da0: 00000000 b30a7e68 b9736d80 00000000 b30a7ddc b30a7dc0 806d247c 806d1654
Apr  2 21:31:03 rpi kernel: [40400.782448] 7dc0: 806d2444 b7318640 00000000 00000000 b30a7dfc b30a7de0 806fe570 806d2450
Apr  2 21:31:03 rpi kernel: [40400.788623] 7de0: 806f1f4c 80666e4c b30a7ee8 00000000 b30a7e0c b30a7e00 80657c84 806fe534
Apr  2 21:31:03 rpi kernel: [40400.795142] 7e00: b30a7e64 b30a7e10 80657d2c 80657c6c 00000000 00000000 00000001 00000058
Apr  2 21:31:03 rpi kernel: [40400.801845] 7e20: 000004cc b30a7f00 00000001 b973f248 00000000 00000000 00000000 b30a7e68
Apr  2 21:31:03 rpi kernel: [40400.808707] 7e40: 00000000 b30a7f68 00000001 b30a7f68 01af13b8 b30a6000 b30a7ea4 b30a7e68
Apr  2 21:31:03 rpi kernel: [40400.815649] 7e60: 802883a4 80657ca0 b9736d80 00000000 00000000 00000000 00000000 00000000
Apr  2 21:31:03 rpi kernel: [40400.822404] 7e80: 00000000 00000000 b9736d80 b9736d80 00000000 b30a7ee8 b30a7ed4 b30a7ea8
Apr  2 21:31:03 rpi kernel: [40400.825912] 7ea0: 802899f8 8028827c 00000000 b30a7f00 b9736d80 00000000 b30a7ee8 b30a7f68
Apr  2 21:31:03 rpi kernel: [40400.829405] 7ec0: 01af13b8 b30a6000 b30a7f5c b30a7ed8 80289bc0 80289974 b30a7ee4 b30a7ee8
Apr  2 21:31:03 rpi kernel: [40400.832905] 7ee0: 000005dc 00000000 00000001 00000000 00000524 b30a7f00 00000001 b973f248
Apr  2 21:31:03 rpi kernel: [40400.836401] 7f00: 01aee544 00000524 b973f240 00000000 00000000 00000000 00000000 00000000
Apr  2 21:31:03 rpi kernel: [40400.839890] 7f20: 00000000 00000530 b973f248 00000000 b86a31a0 802aad44 b30a7f5c b30a7f48
Apr  2 21:31:03 rpi kernel: [40400.843380] 7f40: b9736d80 b9736d80 00000000 00000001 b30a7f94 b30a7f60 80289c7c 80289b58
Apr  2 21:31:03 rpi kernel: [40400.846870] 7f60: 00000000 000005dc 00000000 00000000 b30a7fa4 01aee008 00000000 00011860
Apr  2 21:31:03 rpi kernel: [40400.850367] 7f80: 00000092 80108224 b30a7fa4 b30a7f98 8028b344 80289c1c 00000000 b30a7fa8
Apr  2 21:31:03 rpi kernel: [40400.853855] 7fa0: 80108080 8028b334 01aee008 00000000 00000009 01af13b8 00000001 01af13b8
Apr  2 21:31:03 rpi kernel: [40400.857348] 7fc0: 01aee008 00000000 00011860 00000092 00000000 00000000 76f57000 7eff3e74
Apr  2 21:31:03 rpi kernel: [40400.860847] 7fe0: 00000000 7eff3e44 0001c46c 76e8b8ec 60000010 00000009 00000000 00000000
Apr  2 21:31:03 rpi kernel: [40400.864357] [<806ce384>] (tcp_push) from [<806d1950>] (tcp_sendmsg_locked+0x308/0xdfc)
Apr  2 21:31:03 rpi kernel: [40400.867858] [<806d1950>] (tcp_sendmsg_locked) from [<806d247c>] (tcp_sendmsg+0x38/0x4c)
Apr  2 21:31:03 rpi kernel: [40400.871359] [<806d247c>] (tcp_sendmsg) from [<806fe570>] (inet_sendmsg+0x48/0xe4)
Apr  2 21:31:03 rpi kernel: [40400.874869] [<806fe570>] (inet_sendmsg) from [<80657c84>] (sock_sendmsg+0x24/0x34)
Apr  2 21:31:03 rpi kernel: [40400.878395] [<80657c84>] (sock_sendmsg) from [<80657d2c>] (sock_write_iter+0x98/0xcc)
Apr  2 21:31:03 rpi kernel: [40400.881933] [<80657d2c>] (sock_write_iter) from [<802883a4>] (do_iter_readv_writev+0x134/0x158)
Apr  2 21:31:03 rpi kernel: [40400.885489] [<802883a4>] (do_iter_readv_writev) from [<802899f8>] (do_iter_write+0x90/0x1b0)
Apr  2 21:31:03 rpi kernel: [40400.889037] [<802899f8>] (do_iter_write) from [<80289bc0>] (vfs_writev+0x74/0xc4)
Apr  2 21:31:03 rpi kernel: [40400.892592] [<80289bc0>] (vfs_writev) from [<80289c7c>] (do_writev+0x6c/0xfc)
Apr  2 21:31:03 rpi kernel: [40400.894419] [<80289c7c>] (do_writev) from [<8028b344>] (SyS_writev+0x1c/0x20)
Apr  2 21:31:03 rpi kernel: [40400.896214] [<8028b344>] (SyS_writev) from [<80108080>] (ret_fast_syscall+0x0/0x28)
Apr  2 21:31:03 rpi kernel: [40400.899674] Code: e155000c 03a05000 e2117902 1a000039 (e5d50028) 
```

pelwell, 2018-04-03:
Your kernel predates a fix to the Linux TCP stack that went into our rpi-4.14.y branch on the 28th (later kernel version are unaffected). If you rpi-update now you won't get that crash again.

Darren, 2018-04-03:
@pelwell
Thanks!

closed 2018-04-03 by Darren

andrum99, 2018-04-07:
That would be https://github.com/raspberrypi/linux/commit/9d2ad143e40c38d34be86578840499a976c0a5b0.

@pelwell Do you have an ETA on this patch landing on the "release" Pi kernel, i.e. available through apt-get? I was seeing crashes due to the bug this fixes on my Pi 3B+, and am now running the rpi-update kernel to avoid them. (Apologies if this *has* already been made available via apt-get - I'm not sure where to check what has been included in that kernel).

andrum99, 2018-04-07:
@JamesH65 Thanks - I wouldn't normally expect a new release kernel for around 3 months, but thought one might get released slightly sooner given there are a few fixes for the 3B+. By slightly I mean something like a month from now. Running an rpi-updated kernel makes me a bit nervous, more so as it diverges from the release kernel.

pelwell, 2018-04-07:
Each kernel and firmware update carries a risk, but I don't think the risk is cumulative.

I'm not going to make any promises about the timing of the next release, but with a bit of luck a month wouldn't be out of the question.