raspberrypi/linux #2485: where is Network driver on RPi3 B BCM43438 scorce code ?

ZhaoXuDongTJ, 2018-04-03:
in recently, I want to gather CSI data by RPi3 B, so I must modify the driver code,  which file is used to drive it, thanks!

pelwell, 2018-04-03:
The [brcmfmac driver](https://github.com/raspberrypi/linux/tree/rpi-4.14.y/drivers/net/wireless/broadcom/brcm80211/brcmfmac) drives the BCM43438. Note that some low-level functionality is provided by a closed-source (even to use) firmware blob.

JamesH65, 2018-04-20:
Closing this issue as questions answered/issue resolved.

closed 2018-04-20 by JamesH65

zeoneo, 2019-08-08:
Hi @JamesH65 can you point me to linux source where it loads this binary firmware in BCM.CYW chip?
Also how do I use binary firmware, any documentation or linux source code link?

pelwell, 2019-08-08:
This is not the forum for questions (I won't be giving further answers), but the driver is called brcmfmac - I'm sure you can find it.