raspberrypi/linux #2489: Is it possible to disable VC4 sound?

vanfanel, 2018-04-05:
Hi,

Using the BRCM drivers, there's the "dtparam=audio=off" option in config.txt to disable loading the audio modules, but what's the way to do the same with the VC4 drivers?
I am having problems with my HifiBerry DAC+ and I would like to disable the onboard audio.
I suppose I could build a kernel without VC4 audio support, but I'm sure there must be a way to archieve the same without rebuilding the kernel, right?

Thanks

JamesH65, 2018-05-11:
@popcornmix @pelwell Not quite sure what is wanted here. Is there such a thing as VC4 audio support? Nothing in the kernel config about it.

@vanfanel What do you mean by VC4 audio with regard to the Linux kernel?

JamesH65, 2018-06-29:
@popcornmix Any ideas?

JamesH65, 2019-07-31:
The audio driver is bcm2835-audio, it's a module, so can you simply remove the module to disable it?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

pelwell, 2019-07-31:
I'm pretty sure he's referring to the HDMI audio support available from the VC4 driver, but why he couldn't reply to that effect beats me.

Audio is part of the regular vc4_hdmi driver, which is itself bundled into the overall vc4 module, so blacklisting won't work. It will skip the audio initialisation without failing to create everything else if the Device Tree node has no `dmas` property, but deleting a property isn't possible with an overlay. Fortunately a present-but-empty `dmas` doesn't make any sense, so we can modify the driver to treat an empty dmas the same as an absent one.

If @vanfanel shows an interest I can make that change and add an overlay parameter to control it, otherwise I have more important things to do.

vanfanel, 2019-07-31:
@JamesH65 and @pelwell : I am VERY sorry I did nor reply earlier. I thought I had! Sorry.

I am referring to the VC4-related vc4-hdmi ALSA card, yes, not the bcm2835 ALSA card. The bcm2835 card works fine, and I know it can be disabled on config.txt very easily.
But the vc4-hdmi audio card provided by the VC4 stack only seems to support one format and frequency:

https://github.com/raspberrypi/linux/issues/3043

So, yes, disabling the vc4-hdmi in config.txt card via an overlay parameter would be desirable.
But even better would be having the vc4-hdmi audio card working instead of depending on the (deprecated?) bmc2835 interface.

pelwell, 2019-07-31:
Tweaking the driver and adding the overlay parameter is something I can do in a few minutes, getting the driver to work could take days/weeks for me.

vanfanel, 2019-07-31:
@pelwell: I perfectly understand.
Do you have an idea on why the vc4-hdmi driver only supports one audio format? Is it still being developed?

pelwell, 2019-07-31:
> Is it still being developed?

Yes and no - it isn't finished (in that it isn't fully featured), but the developer isn't actively working on it.

vanfanel, 2019-07-31:
@pelwell : This is shocking! The vc4 development is abandoned? What will be the graphics standard for the Pi then?

pelwell, 2019-07-31:
Other's are being moved onto the project, but progress will be slow for a while.

popcornmix, 2019-07-31:
Note: you can still use the legacy alsa driver with vc4-fkms-v3d (and vc4-kms-v3d).
That is how LibreELEC is configured on Pi4.

vanfanel, 2019-07-31:
@popcornmix : full vc4-kms-v3d is what I use on my projects involving small Pi systems. I know the legacy alsa driver (bmc2835) works fine with that, but having two ALSA drivers (legacy bmc2835 and vc4-hdmi) is causing problems, thats why I was asking if the non-working vc4-hdmi alsa driver could be somehow disabled.

pelwell, 2019-07-31:
I'll add the ability to disable it when I get a spare moment.

vanfanel, 2019-07-31:
@pelwell : Thanks!

pelwell, 2019-07-31:
I've pushed patches that should resolve this issue for you. They'll get picked up by future kernel builds.

vanfanel, 2019-07-31:
@pelwell: Great! I hope one day we can rely on the vc4-hdmi ALSA driver... but for now, this will do!

6by9, 2019-08-01:
The hardware only supports one format, and that is SNDRV_PCM_FMTBIT_IEC958_SUBFRAME_LE.
Questions were asked on the relevant kernel mailing lists as to who or what should be responsible 
for adding that encapsulation when the client wants to provide something else, but no answers were forthcoming.

Duplication in all relevant drivers seems wrong and alsa should in theory support it, but getting the magic runes in asoundrc correct is non-trivial.  There is some reference in https://anholt.livejournal.com/49474.html?nojs=1, but I can't find the kernel mailing list references at present.
Memory says Pulseaudio will also deal with it all for you, and allows mixing of the streams prior to encapsulation which AIUI alsalib won't allow you to do.

popcornmix, 2019-08-01:
@vanfanel this is in latest rpi-update kernel. Can you test? I think you want:
```
dtoverlay=vc4-kms-v3d,audio=off
```

vanfanel, 2019-08-03:
@popcornmix : Yes! It works! :+1: 

@6by9 : I put all the information I could find regarding a possible asoundrc here: https://github.com/raspberrypi/linux/issues/3043
There is a supposedly working asoundrc in: https://raw.githubusercontent.com/takaswie/alsa-lib/master/src/conf/cards/vc4-hdmi.conf
...but sadly it does not work for me.
As you can see, I tried different combinations with no success... I believe ALSA can take care of format conversions.

AndreasHeinrichEmteria, 2020-05-17:
Hey @popcornmix @vanfanel , with audio=off I just deactivate the brcm sound card, what do I have to deactivate vc4-hdmi soundcard? How can I add a present but empty dmas?

What are the entries in config.txt?

6by9, 2020-05-18:
@AndistormEm As stated earlier in this thread, 
```dtoverlay=vc4-kms-v3d,audio=off```

Also documented in the overlays README - https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/README#L2721

AndistormEm, 2020-05-18:
@6by9 Yes this is as I tried, but I deactivate brcm soundcard and not the vc4-hdmi

6by9, 2020-05-18:
More information required from you. Kernel version, and Pi version for a start.

Pi 4, kernel 5.4.20
```dtoverlay=vc4-kms-v3d```
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi1], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```dtoverlay=vc4-kms-v3d,audio=off```
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi1 [vc4-hdmi1], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```dtoverlay=vc4-kms-v3d,audio=off,audio1=off```
```
pi@raspberrypi:~ $ aplay -l
aplay: device_list:272: no soundcards found...
```

```dtparam=audio=on``` would be the normal line for controlling the direct audio driver.

pelwell, 2020-05-18:
@AndistormEm Are you running a 5.4 kernel and not using a Pi 4? If so, I understand the problem. There were some recent changes to the way CMA is configured - it now uses Device Tree rather than kernel command line parameters - and the `audio` parameter in the 5.4 (and later) version of the `vc4-kms-v3d` overlay got broken.

There's a fixed version in the kernel source tree which will be in subsequent releases, but you can test it now.

1. Download the source file: https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/vc4-kms-v3d-overlay.dts
2. Compile it: `dtc -@ -I dts -O dtb -o vc4-kms-v3d.dtbo vc4-kms-v3d-overlay.dts`
3. Install it: `sudo cp vc4-kms-v3d.dtbo /boot/overlays`

AndistormEm, 2020-05-18:
I am running a 4.9 kernel, and now I understand. The 'audio' parameter hasn't been implemented in 4.9. So I have to look for another solution or update the kernel. Thanks @pelwell and @6by9 !!

mightyoakbob, 2022-02-03:
Thank you to everyone who asked for or who added the command:
dtoverlay=vc4-kms-v3d,audio=off

I have spent days trying to get a HiFiBerry Dac+ to work on a raspberry pie built on Bullseye and that command fixed the issue.
For some reason not needed with Buster OS.

Thanks again.

aaronk6, 2022-02-03:
I'd also like to disable VC4 audio, so my HiFiBerry becomes card 0.

However, when I add `dtoverlay=vc4-kms-v3d,audio=off` to config.txt as suggested above, it will only remove one of the vc4  devices (vc4hdmi0). vc4hdmi1 will still be present.

Any idea why that is?

With `dtoverlay=vc4-kms-v3d`:

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

With `dtoverlay=vc4-kms-v3d,audio=off`:

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```

pelwell, 2022-02-04:
It works for me - compare this:
```
pi@raspberrypi:~ $ grep kms /boot/config.txt 
dtoverlay=vc4-kms-v3d
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
with this:
```
pi@raspberrypi:~ $ grep kms /boot/config.txt 
dtoverlay=vc4-kms-v3d,noaudio
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```
What does `uname -a` report on your system?

mightyoakbob, 2022-02-04:
Please forgive me, I'm sure you know this but I didn't until yesterday afternoon.
Yesterday morning I found out about the command : dtoverlay=vc4-kms-v3d,audio=off
And on my old pi with one HDMI port it fixed the issue.
But, it made me look for vc4-kms-v3d and I found it in /boot/config.txt
So now I go to /boot/config.txt
and remove:
dtparam=audio=on              (on board audio)
dtoverlay=vc4-kms-v3d       (HDMI)

and reboot.
No sound cards at all and no need for the new command either.
This is before I add the driver for the HIFIBerry DAC.

pi@LinnPi4:~ $ aplay -l
aplay: device_list:274: no soundcards found...

I'm sorry, I'm sure you already know about this but I had to be sure, folks helped me.

Bob.

aaronk6, 2022-02-04:
@pelwell Thanks! I see you are using `noaudio` instead of `audio=off`. I’ve updated my config accordingly and this did the trick!

Since I have also disabled onboard audio by removing `dtparam=audio` from `config.txt`, this leaves me with:

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Perfect!

For the record, `uname -a` is reporting the following:

```
Linux info-pi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

pelwell, 2022-02-04:
>  see you are using `noaudio` instead of `audio=off`.

I didn't even notice the difference, otherwise I would have commented on it. `audio` is not a parameter of the `vc4-kms-v3d` overlay. However, it is a parameter of the base .dtb file, and:
```
dtoverlay=vc4-kms-v3d,audio=off
```
has the same effect as:
```
dtoverlay=vc4-kms-v3d
dtparam=audio=off
```
which more obviously disables the on-board audio without affecting the HDMI audio.

ArleCamille, 2022-09-22:
Sorry for necromancing the issue, but it seems that passing the parameter audio=off does NOT disable HDMI2's audio output for Raspberry Pi 400. Could you check it out?

DennisFury, 2023-04-22:
just wanted to say I also ran into this issue.  Mine was a bit different, my USB speaker and one of the HDMI audios (vc4hdmi#) kept switching positions as the card # (result of aplay -l )  after each reboot.  I use this as a personal kisok display and my script would fail to run after 1/2 the reboots because no audio device found (wrong default device set).  

Using this command in /boot/config.txt and now I only see 1 HDMI audio in aplay -l
dtoverlay=vc4-kms-v3d,audio=off

so far so good, I would expect both HDMI audio devices to be gone ( I still see vc4hdmi1, but at least vc4hdmi0 is gone).  

4 reboots and he card numbers for the devices appear to be sticking.

thanks

uxigene, 2023-05-17:
> just wanted to say I also ran into this issue. Mine was a bit different, my USB speaker and one of the HDMI audios (vc4hdmi#) kept switching positions as the card # (result of aplay -l ) after each [...]

Have same issue. Can't disable vc4hdmi1 in my rasberrypi 4

strilov, 2023-05-23:
Hi. In case this is still an issue, there seems to be a pi4 specific overlay, which has audio options for hdmi0 and hdmi1: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/vc4-kms-v3d-pi4-overlay.dts#L181

mightyoakbob, 2023-05-26:
Interesting, thanks for the info, just wish i understood it.
Bob.

vanfanel, 2023-05-27:
> Hi. In case this is still an issue, there seems to be a pi4 specific overlay, which has audio options for hdmi0 and hdmi1: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/vc4-kms-v3d-pi4-overlay.dts#L181

Can you please give an example on how to use that overlay to get video & audio on HDMI-1 only?

Thanks

pelwell, 2023-05-27:
There is help available: `$ dtoverlay -h vc4-kms-v3d` (or browse the README directly).
The answer you are looking for is:
```
dtoverlay=vc4-kms-v3d,nohdmi0
```
Unless you are labelling the connectors 1 & 2, in which case you want:
```
dtoverlay=vc4-kms-v3d,nohdmi1
```

vanfanel, 2023-05-27:
> There is help available: `$ dtoverlay -h vc4-kms-v3d` (or browse the README directly). The answer you are looking for is:
> 
> ```
> dtoverlay=vc4-kms-v3d,nohdmi0
> ```
> 
> Unless you are labelling the connectors 1 & 2, in which case you want:
> 
> ```
> dtoverlay=vc4-kms-v3d,nohdmi1
> ```

Ah, yes, thanks for that :)