raspberrypi/linux #2490: after updating to 4.14.32, OpenCV 3.4 open USB camera timeout

imkow, 2018-04-05:
4.14.31 is OK.

detailed info:

select timeout
select timeout
OpenCV Error: Assertion failed (total() == 0 || data != __null) in Mat, file /home/pi/projects/opencv-3.4.0/modules/core/include/opencv2/core/mat.inl.hpp, line 500
terminate called after throwing an instance of 'cv::Exception'
what(): /home/pi/projects/opencv-3.4.0/modules/core/include/opencv2/core/mat.inl.hpp:500: error: (-215) total() == 0 || data != __null in function Mat

popcornmix, 2018-04-05:
Please report output of ``lsusb`` and ``lsmod`` after booting with camera connected.

JamesH65, 2018-05-11:
@imkow Is this still an issue with later kernels? (We are currently on 4.14.39 if you use rpi-update)

jruizaranguren, 2018-05-21:
I am getting the same error with last kernel version 4.14.41-v7+.

```
select timeout
select timeout
OpenCV(3.4.1) Error: Assertion failed (total() == 0 || data != __null) in Mat, file /home/pi/opencv-3.4.1/modules/core/include/opencv2/core/mat.inl.hpp, line 500
```

6by9, 2018-05-21:
Further debug required to have any hope of tracking this down.
```sudo apt install v4l-utils```
```v4l2-ctl -D```
```v4l2-ctl --list-formats-ext```
```v4l2-ctl -V```
```v4l2-ctl -stream-mmap=3 --stream-count=1000 --stream-to=/dev/null```
That last one should stream 1000 frames from the webcam (ditching the resulting data). If that works then it implies OpenCV is doing something wrong.

There's no obvious changes in the history for USB or V4L2 between 4.14.31 and 4.14.32, other than https://github.com/raspberrypi/linux/commit/8960ba68b681a62a523cc593773d5d03a413a075, but that should only be fixing an initialisation error.

jruizaranguren, 2018-05-21:
In my case, this did the trick:

```
sudo rmmod uvcvideo
sudo modprobe uvcvideo nodrop=1 timeout=5000
```

JamesH65, 2018-06-26:
@6by9 Do we need to increase a default timeout somewhere, as per the previous comment?

6by9, 2018-06-26:
```timeout``` defaults to 5000 anyway.
https://elixir.bootlin.com/linux/v4.14.52/source/drivers/media/usb/uvc/uvcvideo.h#L171
https://elixir.bootlin.com/linux/v4.14.52/source/drivers/media/usb/uvc/uvc_driver.c#L39

```nodrop``` has the description of "Don't drop incomplete frames".
If nodrop!=0, then the uvc queue gets a ```UVC_QUEUE_DROP_CORRUPTED``` flag applied, and if we get errors reported then the buffer is reused rather than returned. 
Getting errors would imply the USB camera is misbehaving at some level, so it should NOT be on by default.

Looking back at the original report, there's the outside chance of a USB regression within the 4.14.31 to 4.14.32 window, but unlikely.
If @imkow reports back with the actual model information of his webcam, and whether ```nodrop=1``` works for him, then it gives something that can be investigated.

JamesH65, 2018-06-29:
@imkow Can you provide some more information on the HW being used and try the nodrop=1 option outlined above? We won't be able to make any progress unless we get this information.

nguyen13, 2018-10-17:
I'm facing the same problem. Did you fix this problem?

pelwell, 2018-10-17:
Which kernel are you running? Can you provide the information and follow the suggestion that James mentions above? (@imkow seems to have lost interest)

6by9, 2018-10-17:
Have you tried the steps using v4l2-ctl in https://github.com/raspberrypi/linux/issues/2490#issuecomment-390604528 ? 
Has this device every worked on any Linux platform?

nguyen13, 2018-10-18:
I am using Raspberry Pi 3 B+. I already uninstall OpenCv 3.4.1 and install 3.3.0 instead. It seems working normally now. Thanks guys