raspberrypi/linux #2491: Getting Under-voltage warnings when /etc/cron.daily runs

picameratk, 2018-04-05:
# Top of syslog:
Apr  5 03:25:37 motionpi3 liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.24.0" x-pid="355" x-info="http://www.rsysl$
Apr  5 03:27:29 motionpi3 kernel: [44909.410295] **Under-voltage detected!** (0x00050005)
Apr  5 03:27:35 motionpi3 kernel: [44915.650294] Voltage normalised (0x00000000)
Apr  5 03:29:29 motionpi3 kernel: [45030.060770] **Under-voltage detected!** (0x00050005)
Apr  5 03:29:33 motionpi3 kernel: [45034.210692] Voltage normalised (0x00000000)
Apr  5 03:30:01 motionpi3 CRON[15736]: (root) CMD (   PATH="$PATH:/usr/local/bin/" pihole updatechecker local)
Apr  5 03:39:01 motionpi3 CRON[15794]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /u$
Apr  5 03:39:01 motionpi3 CRON[15796]: (root) CMD (  [ -x /usr/lib/php5/sessionclean ] && /usr/lib/php5/sessionclean)
Apr  5 03:39:06 motionpi3 systemd[1]: Starting Clean php session files...
Apr  5 03:39:06 motionpi3 systemd[1]: Started Clean php session files.
Apr  5 03:40:01 motionpi3 CRON[15880]: (root) CMD (   PATH="$PATH:/usr/local/bin/" pihole updatechecker local)
Apr  5 03:50:01 motionpi3 CRON[15939]: (root) CMD (   PATH="$PATH:/usr/local/bin/" pihole updatechecker local)
Apr  5 03:57:28 motionpi3 kernel: [46708.615260] **Under-voltage detected!** (0x00050005)
Apr  5 03:57:34 motionpi3 kernel: [46714.855234] Voltage normalised (0x00000000)

**At no other time daily do I get Under-voltage messages.**   

# /etc/crontab:
SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin
 m h dom mon dow user  command
17   *  * * *   root    cd / && run-parts --report /etc/cron.hourly
**25   3  * * *   root    cd / && run-parts --report /etc/cron.daily**
47   4  * * 7   root    cd / && run-parts --report /etc/cron.weekly
52   4  1 * *   root    cd / && run-parts --report /etc/cron.monthly
47   1  * * 7   root    ionice -c 3 fstrim -v /
------------------------------------------------------------------------
Other notes:
Raspberry Pi 3 with Raspian Stretch
uname -a: Linux motionpi3 **4.14.32-v7+** #1106 SMP Wed Apr 4 18:19:23 BST 2018 armv7l GNU/Linux
Replaced CanaKit 2.5 A Pi adapter with a new 3A capable Pi power adapter.
Pi used for a combination of Home Automation devices through hostapd, motion detection and Pihole.  Average CPU utilization 1.92 cores as per htop
-----------------------------------------------------------------------
Power usage:
Running headless with one usb based Wifi adapter:
Hawking Technologies HAWNU1 Hi-Gain Wireless-150N Network Adapter with Range Amplifier [Ralink RT3070]

No overclocking.

root@motionpi3:/etc/cron.daily# ls -l
total 64
-rwxr-xr-x 1 root root   268 Nov  3  2014 00logwatch
-rwxr-xr-x 1 root root  1474 Sep 13  2017 apt-compat
-rwxr-xr-x 1 root root 15001 Nov 21  2016 apt.dpkg-bak
-rwxr-xr-x 1 root root   314 Jun 15  2014 aptitude
-rwxr-xr-x 1 root root   355 Oct 28  2014 bsdmainutils
-rwxrwxrwx 1 root root   264 Sep 12  2016 ClearMotionLog
-rwxr-xr-x 1 root root  1597 Nov 26  2015 dpkg
-rwxr-xr-x 1 root root   338 Aug  6  2016 lighttpd
-rwxr-xr-x 1 root root  2211 Jul 11  2014 locate
-rwxr-xr-x 1 root root    90 Apr  5 06:49 logrotate
-rwxr-xr-x 1 root root  1065 Dec 13  2016 man-db
-rwxr-xr-x 1 root root  1386 Aug  8  2017 ntp
-rwxr-xr-x 1 root root   249 Jan 23  2016 passwd

simonk83, 2018-04-06:
I was getting the same, but replacing my power supply appears to have fixed it

JamesH65, 2018-04-06:
These warning simply pass on the results of the low voltage detection circuitry. So you have a low voltage situation, usually caused by a inadequate power supply or power cable.

picameratk, 2018-04-06:
As per my original discussion I bought a new power supply that is rated for 5v and 3A which is more powerful that the old 2.5A power adapters.  I am getting the following messages using the new power adapter.

Also, I receive many more Voltage normalised messages that I do Under-voltage messages which I find curious as well.

**grep Under /var/log/syslog**
Apr  6 03:25:31 motionpi3 kernel: [131192.232834] Under-voltage detected! (0x00050005)

**grep Voltage /var/log/messages**
Apr  6 03:25:35 motionpi3 kernel: [131196.392873] Voltage normalised (0x00000000)
Apr  6 04:19:36 motionpi3 kernel: [134437.044976] Voltage normalised (0x00000000)
Apr  6 04:51:36 motionpi3 kernel: [136356.891041] Voltage normalised (0x00000000)
Apr  6 05:01:14 motionpi3 kernel: [136935.132919] Voltage normalised (0x00000000)
Apr  6 05:43:55 motionpi3 kernel: [139495.620991] Voltage normalised (0x00000000)
Apr  6 06:47:54 motionpi3 kernel: [143335.315353] Voltage normalised (0x00000000)
Apr  6 07:09:55 motionpi3 kernel: [144656.115042] Voltage normalised (0x00000000)
Apr  6 07:37:54 motionpi3 kernel: [146334.681245] Voltage normalised (0x00000000)
Apr  6 09:01:53 motionpi3 kernel: [151374.537690] Voltage normalised (0x00000000)
Apr  6 09:41:54 motionpi3 kernel: [153774.864817] Voltage normalised (0x00000000)
Apr  6 10:17:55 motionpi3 kernel: [155935.994672] Voltage normalised (0x00000000)
Apr  6 10:27:54 motionpi3 kernel: [156535.036185] Voltage normalised (0x00000000)
Apr  6 13:05:55 motionpi3 kernel: [166015.704445] Voltage normalised (0x00000000)
Apr  6 13:41:54 motionpi3 kernel: [168174.749814] Voltage normalised (0x00000000)
Apr  6 13:53:55 motionpi3 kernel: [168896.512111] Voltage normalised (0x00000000)
Apr  6 13:57:55 motionpi3 kernel: [169135.712821] Voltage normalised (0x00000000)
Apr  6 14:21:54 motionpi3 kernel: [170575.077509] Voltage normalised (0x00000000)

simonk83, 2018-04-06:
Yep, I suspect your new power supply is no better.    I bought one that was rated fine for the Pi as well, and it was still shitty.   I ended up with an official one and it's fine.

picameratk, 2018-04-06:
I appears that the Under-voltage issue was caused by a ## **case fan** that was power off of the GPIO.

closed 2018-04-06 by picameratk