raspberrypi/linux #2493: Will there be a official 4.9 release with 3B+ support or is it all 3.14 from now on?

jorisvergeer, 2018-04-06:
Like the title said:
Will there be a official 4.9 release with 3B+ support or is it all 3.14 from now on?

I have production pies running on various locaties on rasbian jessie with 3.9 and I prefer not to update to 3.14 and prefer using the same distribution for new instalation on the 3B+

jorisvergeer, 2018-04-09:
Yes, i meant 4.14 ;)

In raspbian, it looked like the firmware packages and kernel packages depend on each other. (Maybe I saw that wrong...)

In your opinion, is installing the 4.14 kernel on raspbian jessie stable? 
We use a customized raspbian image, and creating a stretch image would require some time. (And we have some 3B+ stock at the moment we cannot use)

monetny, 2018-04-16:
for install 4.9 kernel with 3B+ support run command
sudo rpi-update 936a8dc3a605c20058fbb23672d6b47bca77b0d5

E3V3A, 2018-04-18:
@monetny 
> sudo rpi-update 936a8dc3a605c20058fbb23672d6b47bca77b0d5

So can I use the same to downgrade from 4.14 to 4.9.80 on RPi3B?

Ferroin, 2018-04-18:
You can prevent the package manager from screwing with the kernel by putting a hold on the kernel package using `apt-mark hold` (you can later remove the hold with `apt-mark unhold`).

E3V3A, 2018-04-19:
> putting a hold on the kernel package using apt-mark hold
Sorry for the OT, but I use this all the time for single packages, but I would guess for kernel, you'd need to include the bootloader, firmware and libraries as well?

lurch, 2018-04-26:
> I have production pies running on various locaties on rasbian jessie with 3.9 and I prefer not to update to 3.14 and prefer using the same distribution for new instalation on the 3B+

If you've got systems in production and you don't want to modify your customized raspbian image, another option would be to keep buying the old model 3B and not the 3B+ ? :slightly_smiling_face: 
(The RPF make a big thing about not EOLing older models, for precisely this reason)

JamesH65, 2018-05-11:
I think the question has been answered, any objections to this issue being closed?

JamesH65, 2018-06-26:
Closing this issue as questions answered/issue resolved.

closed 2018-06-26 by JamesH65