raspberrypi/linux #2495: please update the 4.9 branch

stefson, 2018-04-07:
hey there, 

it would be great to have the patches from kernel.org integrated into the 4.9 branch, it is still at 4.9.80 at the moment. All patches apply but for this one, which already got merged into the rpi 4.9 branch: 

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/diff/drivers/usb/gadget/function/f_hid.c?id=v4.9.92&id2=v4.9.89

the conflict can be solved easily, it shouldn't require any troublesome backporting at all.

popcornmix, 2018-04-09:
The problem is we are no longer testing the 4.9 kernel as it is no longer used by raspbian.
We'd recommend users switch to the 4.14 kernel. Any reason not to do this?

smp79, 2018-04-10:
I'm also have to use kernel 4.9 for my LibreELEC builds. With kernel 4.14 there are audio dropouts during heavy sdcard i/o activity. It is impossible to watch and record the TV channel at the same time. No such issue then building with kernel 4.9.

stefson, 2018-04-11:
There are more OS out there but raspian, Gentoo for that matter or flavours of Ubuntu. I don't really trust 4.14 yet, it had terrible bugs for weeks. I can always maintain my own kernel tree, but it would be great to see the patches from kernel.org added.

JamesH65, 2018-04-20:
Closing as a Won't Fix. This means that although the issue is acknowledged, it is felt that no further action is likely. This may be down to a bad benefit vs cost analysis, or it may actually not be possible to fix.

closed 2018-04-20 by JamesH65