raspberrypi/linux #2496: Legacy OSS drivers.

ED6E0F17, 2018-04-08:
Noticed in the forums:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=207875

OSS sound has been obsolete for a very long time, but is used in a lot of legacy code, mostly games. It seems that upstream will remove it after 4.14, but it may still work if you enable "BROKEN"

https://github.com/raspberrypi/linux/commit/31cbee6a5611f07d2d66f55bb6f8648db5947e32#diff-ecedcdca1f8e6c93a5b538cef26256b3

ED6E0F17, 2018-04-08:
It rather depends on how broken it actually is. I do not like breaking userspace, but you control the userspace, so you can fix it in Raspbian. Enabling a broken kernel driver would stop osspd from running, but if the kernel driver builds and runs then it would be an easy fix to re-enable it.

ED6E0F17, 2018-04-08:
Closing as an upstream issue, since linux OSS emulation was killed in October last year. This makes me sad.

ED6E0F17, 2018-04-08:
I got that totally wrong. OSS drivers were removed, but OSS emulation is still in the kernel. I think it is just a configuration issue caused by: https://github.com/raspberrypi/linux/commit/eb3b705aaed9cab7efef8c905eda919b5fd902b9#diff-1927138f57398ea2053f25d80040aab4

pelwell, 2018-04-10:
Which is it going to be - `osspd-alsa` or `CONFIG_SND_OSSEMUL=y`? I would prefer the user-space solution since it doesn't increase the kernel footprint for all the users who don't need OSS sound.

Ferroin, 2018-04-10:
FWIW, based on my own experience, the kernel option is a bit more reliable, but I've only ever used it on an x86 system, so that may not be the case on the Pi.  It does 'just work' without needing any user intervention though, and when not actively used it doesn't have any overhead other than the increased memory footprint.

ED6E0F17, 2018-04-11:
There was another patch that should allow OSS emulation to be built as a module now, and since sound is built as a module you should find that SND_MIXER_OSS, SND_PCM_OSS and SND_OSSEMUL do not increase the size of the kernel image.

https://github.com/raspberrypi/linux/commit/3d774d5ef066971aee1839150bf47d100d8e43d3#diff-1927138f57398ea2053f25d80040aab4

JamesH65, 2018-06-29:
@ED6E0F17 @Ferroin Have either of you anything to add to this, or requests for kernel changes or suggestions?

ED6E0F17, 2018-06-30:
Closing as @pelwell has expressed a preference for a userspace solution. I prefer "no regressions" for the sake of that one person who is going to hit this issue and lose hours of their life fixing it, but there is a balance and 4.14 has been rolled out to Raspbian for some time now.

closed 2018-06-30 by ED6E0F17