raspberrypi/linux #2499: python-* packages segfault on install...

tomidotomicode, 2018-04-08:
I'm running Raspbian 9.4 (stretch) on Raspberry Pi 3B+, tried installing some package that required python dependencies. 

python-idna segfaults, python-numpy segfaults, python-gobject-2 segfaults, python-gi segfaults, python-pyasn1 segfaults,... you get the idea.

Here is a list of packages that could not be installed 
 python-idna
 python-crypto
 python-pyasn1
 python-wheel
 python-pkg-resources
 python-numpy
 python-keyrings.alt
 python-gobject-2
 python-gi
 python-six
 python-enum34
 python-cryptography
 python-cairo
 python-dbus
 python-ipaddress
 python-pip
 python-xdg
 python-secretstorage
 python-setuptools
 python-keyring

Even python itself segfaults when run. Should be easy to reproduce... unfortunetaly I can not even remove these packages (even uninstall segfaults). Any ideas how to fix?

pelwell, 2018-04-09:
It sounds like your installation is corrupt - all those packages are present and up-to-date in the current Stretch release. Are you running Raspbian Lite?

You could try removing and reinstalling python, but then you will have to reinstall all of its dependencies, and you still wouldn't be sure that something else wasn't corrupted. A clean start might be a better option.

tomidotomicode, 2018-04-09:
Yes, thought so, this is Raspbian Lite stretch with latest upgrades. After the reboot whole thing couldn't boot properly anymore, so I simply started from the scratch.

pelwell, 2018-04-09:
I'm closing this issue now. So far this doesn't look like it was a kernel issue, but I won't rule it out yet. Ensure your power supply can deliver 2.5A at 5V because under-voltage can be a cause of corruption. If the command `vcgencmd get_throttled` returns a non-zero value then your system has either been overheating or underpowered.

If you see a similar issue again, drop me an email (phil@raspberrypi.org) as you may have a faulty board that we would be interested in.

closed 2018-04-09 by pelwell