raspberrypi/linux #2500: Writing to the sdcard hangs suddently

carlonluca, 2018-04-09:
Hello! I'm using a modified version of noobs for a project and I noticed that sometimes the process that writes to the sdcard hangs indefinitely. The process is a wget | gz | bsdtar. When it hangs, I can log in via ssh and whatever I try to run on the sdcard hangs. I added strace to the image and I get this:

```
strace dd if=/dev/zero of=/mnt2/test count=1024 bs=1024
execve("/bin/dd", ["dd", "if=/dev/zero", "of=/mnt2/test", "count=1024", "bs=1024"], [/* 20 vars */]) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0x76fd7000
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=298280, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0x76fd6000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\20\241\0\0004\0\0\0"..., 4096) = 4096
mmap2(NULL, 348160, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f76000
mmap2(0x76f76000, 293688, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x76f76000
mmap2(0x76fc5000, 4832, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x47000) = 0x76fc5000
mmap2(0x76fc7000, 14184, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76fc7000
close(3)                                = 0
munmap(0x76fd6000, 4096)                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=25312, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_UNINITIALIZED, -1, 0) = 0x76fd6000
set_tls(0x76fd6480, 0x76fd6b20, 0x76fd6b28, 0x76fd6480, 0x76fd8f74) = 0
mprotect(0x76fc5000, 4096, PROT_READ)   = 0
mprotect(0x76fd8000, 4096, PROT_READ)   = 0
ioctl(0, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, {B38400 opost isig icanon echo ...}) = 0
getuid32()                              = 0
brk(0)                                  = 0x1da4000
brk(0x1da5000)                          = 0x1da5000
rt_sigaction(SIGUSR1, {0x5d410, [], SA_RESTORER|SA_RESTART, 0x76f80d74}, NULL, 8) = 0
open("/dev/zero", O_RDONLY|O_LARGEFILE) = 3
dup2(3, 0)                              = 0
close(3)                                = 0
open("/mnt2/test", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666
```

This is the status:

```
# cat /proc/242/status
Name:   dd
Umask:  0077
State:  D (disk sleep)
Tgid:   242
Ngid:   0
Pid:    242
PPid:   239
TracerPid:      239
Uid:    0       0       0       0
Gid:    0       0       0       0
FDSize: 32
Groups: 0 10 
VmPeak:     1176 kB
VmSize:     1176 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:       184 kB
VmRSS:       184 kB
RssAnon:              32 kB
RssFile:             152 kB
RssShmem:              0 kB
VmData:       44 kB
VmStk:       132 kB
VmExe:       640 kB
VmLib:       320 kB
VmPTE:         8 kB
VmPMD:         0 kB
VmSwap:        0 kB
Threads:        1
SigQ:   8/1546
SigPnd: 0000000000000000
ShdPnd: 0000000000020002
SigBlk: 0000000000000000
[... 231 lines cut ...]
[    2.962052] usb 1-1.4: new low-speed USB device number 5 using dwc_otg
[    3.105795] usb 1-1.4: New USB device found, idVendor=1c4f, idProduct=0048
[    3.105803] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.105807] usb 1-1.4: Product: Usb Mouse
[    3.105811] usb 1-1.4: Manufacturer: SIGMACHIP
[    3.105893] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[    3.115805] input: SIGMACHIP Usb Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:1C4F:0048.0002/input/input1
[    3.118748] EXT4-fs (mmcblk0p5): re-mounted. Opts: data=ordered
[    3.125885] EXT4-fs (mmcblk0p5): re-mounted. Opts: data=ordered
[    3.182281] hid-generic 0003:1C4F:0048.0002: input,hidraw1: USB HID v1.10 Mouse [SIGMACHIP Usb Mouse] on usb-3f980000.usb-1.4/input0
[    5.233500] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[    5.373018] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.863495] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xDDE1
[    6.886543] random: crng init done
[    8.774934] EXT4-fs (mmcblk0p5): re-mounted. Opts: data=ordered
[   11.449557] EXT4-fs (mmcblk0p5): re-mounted. Opts: data=ordered
[   11.574721]  mmcblk0: p1 p2 < p5 p6 p7 p8 >
[   11.718720]  mmcblk0: p1 p2 < p5 p6 p7 p8 >
[   12.806468] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[   19.694824] EXT4-fs (mmcblk0p7): recovery complete
[   19.697769] EXT4-fs (mmcblk0p7): mounted filesystem with ordered data mode. Opts: (null)
[   26.801034] EXT4-fs (mmcblk0p7): mounted filesystem with ordered data mode. Opts: (null)
```

Rebooting fixes the problem and everything is back to normal. I have never reproduced anything similar in raspbian. Any advice on what I could do to debug further? Thanks!

jens-maus, 2018-04-13:
This is very interesting. I am having similar issues here with one of my projects where also on heavy write operations to the SD card (mmc device) operations suddenly stop with the kernel returning "hung process" warnings after 120s. I am also able to perform a remote login to the system and every process not read/writing to the sd card returns immediately. but as soon as I try to read from the sd card the process in question hangs.

Compared to your setup I am using the latest 4.14.33 kernel. After some investigation I could slightly improve the probability that the sd card hang occurs by adding `sdhci_bcm2708.enable_llm=0` to the kernel command-line disabling the low-latency mode of the sdhci. However, as said, this only seem to reduce the probability that this hang occurs as I can still see it happing from time to time.

I will try to grab some more debug information on that matter and post it here accordingly. But if a raspberrypi developer could already give some hints in how or where to investigate that issue, this would be great.

stschake, 2018-04-16:
Could you post your kernel configs? I've had this happen too, with the exact same symptoms (everything that needs sdcard IO just hangs), sometimes without the kernel hung task detector complaining. I can't imagine it happens under Raspbian, so maybe there is some common variable in our configs that might be causing it.

jens-maus, 2018-04-16:
That's the kernel defconfig file my project is using and which exhibits the issue (especially if `sdhci_bcm2708.enable_llm=0` is omitted):

https://github.com/jens-maus/RaspberryMatic/blob/master/buildroot-external/board/raspberrypi3/kernel_defconfig

carlonluca, 2018-04-17:
@stschake in fact I was never able to reproduce in Raspbian.

jens-maus, 2018-04-17:
As @stchake is using the NOOBS provided kernel which actually is also created by buildroot like my environment/kernel I also suspect that this might be somewhat related to some kernel option or toolchain build related things causing that raspberrypi kernels compiled within buildroot might exhibit the problem. Perhaps indeed comparing kernel defconfigs between Raspbian and buildroot/NOOBS and also perhaps toolchain related differences might help in trying to pin down the potential issue. However, would also be great if some raspberrypi developer could comment on that issue and give some hints in how to debug this situation.

pelwell, 2018-04-17:
@jens-maus Which kernel are you building? sdhci_bcm2708 hasn't existed for very long time now (the 3.9 kernel, from what I can tell).

The sdhost driver interacts with the firmware to manage the clock divisor, but only when the MMC framework is expecting the clock to change, i.e. when the card is being probed and brought up to speed. Apart from that the only dependency should be on the DMA driver/hardware and on the SD hardware itself.

You may learn something by enabling the driver's logging feature which will record activity the kernel message log. Add `dtparam=sd_debug=on` to config.txt and reboot. You can also eliminate a DMA problem as being the cause (at the cost of some performance) by adding `dtparam=sd_force_pio=on`.

monetny, 2018-04-17:
I change root to f2fs for fix this problem (not fix, but minimaze)

JamesH65, 2018-05-11:
@jens-maus Did you have a chance to try out @pelwell suggestions above?

JamesH65, 2018-06-29:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

carlonluca, 2018-06-30:
I tested with the dtparam=sd_debug=on and I was able to reproduce the issue a couple of times. No error appeared in dmesg. Logs indicated that data was being written and then gradually stopped. The rest of the behavior is as described in the first post. I'm now using the watchdog to handle this issue, I couldn't find another way.