raspberrypi/linux #2503: Kernel 4.14 behaves poorly in LibreELEC

smp79, 2018-04-10:
I have to use kernel 4.9 for my LibreELEC 9.0 builds because with the current default kernel 4.14 there are audio dropouts during heavy sdcard i/o activity. 
Among other things, this makes impossible to watch and record the TV channel at the same time. The issue is present in current Milhouse builds. No such issue then building with kernel 4.9.

popcornmix, 2018-04-11:
You'll need to provide some details. What model of Pi? What audio interface?
Any custom config.txt settings or hardware attached?
Mediainfo of a file that has problems?
Can you reproduce without PVR? (e.g. play a test file and copy a file on sdcard at same time)

smp79, 2018-04-11:
The hardware is a Pi3, connected directly to a TV via HDMI. DVB tuner connected to the same Pi3.
I don't know how to reproduce without PVR but 1 user reported that my build with kernel 4.9 fixed a [similar issue for him](https://forum.kodi.tv/showthread.php?tid=298461&pid=2703880#pid2703880).

The easiest way for me to reproduce is to start recording when watching a TV channel. That's when the audio dropouts occur with tons of ActiveAE errors in kodi.log. Also any heavy network activity like copying a file over LAN will cause the short audio dropouts / ActiveAE errors.

I also tried to record to a USB drive instead of the sdcard - same issues. 

The recorded stream is perfectly fine, only the playback is affected.

mkreisl, 2018-04-13:
Similar issue reported from an XBian user, see [this post](http://forum.xbian.org/thread-2877-post-35146.html#pid35146)

Reverting kernel 4.14.32+ to 4.9.91+ solves his issue

smp79, 2018-04-13:
I will also add that I tested with different kernel versions - the issue was introduced in kernel 4.10.

smp79, 2018-04-13:
OMXplayer does not avoid the issue, advancedsettings.xml is empty, audio output format does not make a difference.
I just tried to reproduce the issue without PVR and it is quite easy.
1. Start the playback of a high bitrate 1080p video from a **USB** flash drive.
2. During the playback start copying a large file to the Pi's sdcard over **LAN**.
That's when the audio dropouts will start to occur.

smp79, 2018-04-20:
Found [this](https://linuxmusicians.com/viewtopic.php?t=16920), possibly related issue.

polojoe, 2018-05-02:
The described issue also exists if a download is started during livetv watching in le. After some time there's an audio dropout, after that a/v are out of sync.
Logs required?

Krawei, 2018-06-15:
Any updates here?

smp79, 2018-06-16:
I tested with kernel 4.17 and there is no improvement.

SpeedyQ, 2018-07-22:
Any updates on this issue?

I recently tried updating to 4.14.54+ kernel (xbian on raspberry pi) which is current stable version, but the issue still exists where music playback seems to pause for millisecond(s) and continuing again where it paused.
I reverted back to 4.9.96+ kernel which is still ok regarding the stutter issue...

smp79, 2018-08-13:
Similar issue reported [here](https://lkml.org/lkml/2018/5/28/1985). I reverted commit a1b8913 but unfortunately it didn't improve things.

smp79, 2018-08-14:
I think I'm getting closer to solving this. I built 2 LibreELEC images: with kernel 4.10-rc1 and 4.10.0.
4.10-rc1: no audio dropouts at all when recording a DVB stream
4.10.0: lots of audio dropouts when recording a DVB stream

The offending commit is somewhere between 4.10-rc2 and 4.10-rc8. I will build more images and test.

smp79, 2018-08-14:
Ok, I narrowed down the issue to kernel 4.10-rc6 (no dropouts with rc5 and earlier). Any suggestions on what commits I should try to revert are welcome.

smp79, 2018-08-14:
Ok, I narrowed it down to kernel 4.10-rc6 (no issues with rc5 and older). Any suggestions on what commits I should try to revert are welcome :-)

popcornmix, 2018-08-14:
Interesting. Can you narrow it further?
Looking at the commit list (there's quite a few) the "mm" ones look like possible culprits.
Otherwise there's a number of net related commit which I guess could cause the issues here.

@smp79 Is network involved when you see the issue, or can you see it purely locally (dvb + local storage)?

mkreisl, 2018-08-14:
@SpeedyQ has reported [here](http://forum.xbian.org/thread-2877-post-35495.html#pid35495) after changing hardware from Pi3B to Pi3B+ stuttering has gone with 4.14.58+ kernel
Maybe it's network related (as everybody knows, Pi3B+ is using different network driver)

polojoe, 2018-08-14:
I'm using rpi3+ and also have the issue when watching tv during recording on local storage. Issue is also present during dvb and download.

smp79, 2018-08-14:
Ok, I'm not a git expert and I need some help with this.
I narrowed it down to those 4 commits:
67ade058ef2c65a3e56878af9c293ec76722a2e5
1fdf41941b8010691679638f8d0c8d08cfee7726
57b59ed2e5b91e958843609c7884794e29e6c4cb
99f300cf1f3877ad8ea923127de817a460b787bd

Commit a47b70ea86bdeb3091341f5ae3ef580f1a1ad822 is good, no audio issues.
Commit 99f300cf1f3877ad8ea923127de817a460b787bd is bad, I get audio dropouts.

There is something funky going on with those 3:
67ade058ef2c65a3e56878af9c293ec76722a2e5
1fdf41941b8010691679638f8d0c8d08cfee7726
57b59ed2e5b91e958843609c7884794e29e6c4cb
I am unable to build LibreELEC with those. They supposed to change only 1 file - fs/btrfs/inode.c. However, I get this:
[code]  UNPACK   linux
      APPLY PATCH (common - raspberrypi)   packages/linux/patches/raspberrypi/linux-001_kernel_build_fix.patch
patching file include/linux/log2.h
patching file tools/include/linux/log2.h
      APPLY PATCH (common - raspberrypi)   packages/linux/patches/raspberrypi/linux-999-dvb-softirq.patch
patching file kernel/softirq.c
      APPLY PATCH (device)   projects/RPi/devices/RPi2/patches/linux/linux-01-RPi_support.patch
patching file drivers/net/usb/smsc95xx.c
patching file drivers/net/usb/smsc95xx.c
patching file drivers/net/usb/smsc95xx.c
patching file kernel/resource.c
patching file mm/page_alloc.c
Hunk #1 succeeded at 7289 (offset -72 lines).
patching file drivers/irqchip/irq-bcm2836.c
patching file arch/arm/mach-bcm/Kconfig
patching file drivers/irqchip/irq-bcm2835.c
patching file drivers/irqchip/irq-bcm2835.c
patching file drivers/spi/spidev.c
Hunk #1 FAILED at 697.
1 out of 1 hunk FAILED -- saving rejects to file drivers/spi/spidev.c.rej
patching file drivers/tty/serial/8250/8250_core.c
patching file drivers/pinctrl/bcm/Kconfig
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/pinctrl/bcm/Kconfig.rej
patching file drivers/pinctrl/bcm/pinctrl-bcm2835.c
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] n
Skipping patch.
13 out of 13 hunks ignored -- saving rejects to file drivers/pinctrl/bcm/pinctrl-bcm2835.c.rej
patching file drivers/pinctrl/bcm/pinctrl-bcm2835.c
Hunk #1 succeeded at 386 (offset 5 lines).
patching file drivers/pinctrl/bcm/pinctrl-bcm2835.c
Hunk #1 succeeded at 1015 with fuzz 2 (offset -34 lines).
patching file drivers/spi/spi-bcm2835.c
patching file drivers/spi/spi-bcm2835.c
patching file drivers/spi/spi-bcm2835.c
patching file arch/arm/mach-bcm/board_bcm2835.c
patching file drivers/dma/Kconfig
patching file drivers/dma/bcm2835-dma.c
patching file include/soc/bcm2835/raspberrypi-firmware.h
Hunk #5 succeeded at 115 with fuzz 2.
patching file drivers/rtc/rtc-pcf2123.c
patching file drivers/watchdog/bcm2835_wdt.c
Hunk #2 succeeded at 105 (offset -9 lines).
Hunk #3 succeeded at 134 (offset -9 lines).
Hunk #4 succeeded at 163 (offset -9 lines).
Hunk #5 succeeded at 199 (offset -20 lines).
patching file arch/arm/kernel/reboot.c
patching file drivers/soc/bcm/Kconfig
patching file drivers/clk/bcm/clk-bcm2835.c
Hunk #1 succeeded at 1892 (offset -66 lines).
Hunk #2 succeeded at 1917 (offset -66 lines).
patching file drivers/char/hw_random/bcm2835-rng.c
[... 704 lines cut ...]
patching file sound/soc/bcm/iqaudio_digi.c
patching file sound/soc/bcm/justboom-dac.c
patching file sound/soc/bcm/justboom-digi.c
patching file sound/soc/bcm/pisound.c
patching file sound/soc/bcm/raspidac3.c
patching file sound/soc/bcm/rpi-dac.c
patching file sound/soc/bcm/rpi-proto.c
patching file arch/arm/boot/dts/bcm2708-rpi-b-plus.dts
patching file arch/arm/boot/dts/bcm2708-rpi-b.dts
patching file arch/arm/boot/dts/bcm2709-rpi-2-b.dts
patching file arch/arm/boot/dts/bcm2710-rpi-3-b.dts
patching file drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c
Hunk #1 succeeded at 6804 (offset 87 lines).
patching file drivers/net/wireless/broadcom/brcm80211/brcmfmac/common.c
patching file arch/arm/boot/dts/overlays/vc4-kms-v3d-overlay.dts
patching file drivers/clk/bcm/clk-bcm2835.c
Hunk #1 succeeded at 1144 (offset -54 lines).
Hunk #2 succeeded at 1162 (offset -54 lines).
Hunk #3 succeeded at 1218 (offset -54 lines).
Hunk #4 succeeded at 1842 (offset -66 lines).
Hunk #5 succeeded at 1863 (offset -66 lines).
Hunk #6 succeeded at 1885 (offset -66 lines).
Hunk #7 succeeded at 1901 (offset -66 lines).

[/code]

popcornmix, 2018-08-14:
How are you generating the commit hashes to test?
I wonder if you will be better off just using the merge commits
(e.g. ``git log --merges``) as the points to test.

smp79, 2018-08-14:
I'm simply picking them one-by-one. I edit PKG_VERSION= in packages/linux/package.mk (in LibreELEC build system). The last good one is a47b70ea86bdeb3091341f5ae3ef580f1a1ad822

popcornmix, 2018-08-14:
I think the problem is that the git history is not linear due to the merge commits.
You may be better off using the merge commits from:
```
 git log --pretty=oneline --merges 99f300c...a47b70e
```
(but I wouldn't claim to be a git expert either).

smp79, 2018-08-15:
I found the bad commit. It took more than 20 hours and 50+ LibreELEC/kernel builds :)

424f6c4

I also [reverted that commit](https://github.com/smp79/linux/commit/6368fb6947feaa9751814d9dcc8bceac0b8f729c) for kernel 4.14 - audio dropouts are now gone.

popcornmix, 2018-08-15:
Thank you. That commit does look plausible for causing kernel stalls.
@mkreisl can you confirm the fix?

See #2647

mkreisl, 2018-08-15:
@popcornmix 
I was never able to reproduce this issue here, so it is impossible for me to confirm the fix

popcornmix, 2018-08-16:
@smp79 without the commit reverted, does increasing CMA avoid the issue?
E.g. add ``cma=64M`` to cmdline.txt ?

smp79, 2018-08-16:
No, cma=64M does nothing.

> People with issues tend to be recording and playing live TV or running background jobs like torrents

I actually noticed the audio dropouts when only watching live TV with no extra background jobs. Those were rare but still annoying. Doing stuff like recording will just make the issue very obvious and easy to reproduce.

mkreisl, 2018-08-16:
Hmmm, that's weird. I'm running JDownloader2 in background while watching TV. Everything is on network (Videos, JDownloader saves data on network, even root fs is on iSCSI network disk) Never had audio dropouts even when playing 1080p HEVC vids

smp79, 2018-08-16:
The issue is more pronounced when the DVB tuner is used. It is harder to reproduce when just streaming video over LAN.

closed 2018-10-29 by smp79