raspberrypi/linux #2506: Raspberry Pi corrupted packets over serial

leragequit, 2018-04-12:
When reading large packets from a Serial -> USB converter I can observe corrupted packages, with invalid or missing data. 

**This behaviour is only exibited using the raspberry pi linux kernel.** 

Could test and confirm this with rpi3, cm3, rpi1 model b+ and very late and early iterations ( 4 year ago) of raspbian.
The bug was not present in debian aarch64 on rpi3 and cm3, was not present when using the beaglebone.

Can anyone confirm this bug? Read 12k serial from an arduino or s.th. at 921600 baud. bytes should be missing or currupted regardless of hardware used when using the raspberry pi kernel.

Regards,
Sascha

P33M, 2018-04-19:
Ah. You have a high baudrate. It's likely that our endpoint throttling is causing FIFO overflows inside your usb-serial adapter.

What happens if you add `dwc_otg.nak_holdoff=1` to /boot/cmdline.txt?

P33M, 2018-05-15:
No OP response after >2 weeks. Closing unless bug is reported as unfixed.

closed 2018-05-15 by P33M