raspberrypi/linux #2507: Not detecting wlan0 (Raspberry Pi 3)

davfsa, 2018-04-12:
I left my raspberry pi 3 model B some days without using it and when I turned it on again, neither the Bluetooth nor the WiFi functions worked. Any help on how to fix this?

pelwell, 2018-04-12:
Please post the output of `dmesg`.

davfsa, 2018-04-13:
`pi@raspberrypi:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.33-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1109 SMP Tue Apr 10 17:28:38 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map b9faa000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @b9f53000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936176K/966656K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22288K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                      .bss : 0x80c96f58 - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25230 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000274] Console: colour dummy device 80x30
[    0.000824] console [tty1] enabled
[    0.000861] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000900] pid_max: default: 32768 minimum: 301
[    0.001229] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001261] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002212] Disabling memory control group subsystem
[    0.002310] CPU: Testing write buffer coherency: ok
[    0.002719] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003115] Setting up static identity map for 0x100000 - 0x10003c
[    0.003251] Hierarchical SRCU implementation.
[    0.003906] smp: Bringing up secondary CPUs ...
[    0.004612] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005368] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006102] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006205] smp: Brought up 1 node, 4 CPUs
[    0.006275] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006296] CPU: All CPU(s) started in HYP mode.
[    0.006314] CPU: Virtualization extensions available.
[    0.007185] devtmpfs: initialized
[    0.017792] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.018041] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018085] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.018664] pinctrl core: initialized pinctrl subsystem
[    0.019420] NET: Registered protocol family 16
[    0.022117] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026923] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.026955] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 174 lines cut ...]
[    5.588884] logitech-hidpp-device 0003:046D:4055.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech Wireless Mouse] on usb-3f980000.usb-1.5:1
[    5.598501] input: Logitech K520 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.2/0003:046D:C52B.0003/0003:046D:2011.0005/input/input1
[    5.599328] logitech-hidpp-device 0003:046D:2011.0005: input,hidraw2: USB HID v1.11 Keyboard [Logitech K520] on usb-3f980000.usb-1.5:2
[... cut ...]
[   36.939053] Bluetooth: HCI device and connection manager initialized
[   36.939070] Bluetooth: HCI socket layer initialized
[   36.939082] Bluetooth: L2CAP socket layer initialized
[   36.939104] Bluetooth: SCO socket layer initialized
[   43.629049] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   43.629060] Bluetooth: BNEP filters: protocol multicast
[   43.629083] Bluetooth: BNEP socket layer initialized
[   69.667764] logitech-hidpp-device 0003:046D:2011.0005: HID++ 1.0 device connected.
`

davfsa, 2018-04-13:
> pi@raspberrypi:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.33-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1109 SMP Tue Apr 10 17:28:38 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map b9faa000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @b9f53000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936176K/966656K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22288K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                      .bss : 0x80c96f58 - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25230 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000274] Console: colour dummy device 80x30
[    0.000824] console [tty1] enabled
[    0.000861] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000900] pid_max: default: 32768 minimum: 301
[    0.001229] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001261] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002212] Disabling memory control group subsystem
[    0.002310] CPU: Testing write buffer coherency: ok
[    0.002719] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003115] Setting up static identity map for 0x100000 - 0x10003c
[    0.003251] Hierarchical SRCU implementation.
[    0.003906] smp: Bringing up secondary CPUs ...
[    0.004612] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005368] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006102] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006205] smp: Brought up 1 node, 4 CPUs
[    0.006275] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006296] CPU: All CPU(s) started in HYP mode.
[    0.006314] CPU: Virtualization extensions available.
[    0.007185] devtmpfs: initialized
[    0.017792] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.018041] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018085] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.018664] pinctrl core: initialized pinctrl subsystem
[    0.019420] NET: Registered protocol family 16
[    0.022117] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026923] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.026955] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 173 lines cut ...]
[    5.588272] input: Logitech Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.2/0003:046D:C52B.0003/0003:046D:4055.0004/input/input0
[    5.588884] logitech-hidpp-device 0003:046D:4055.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech Wireless Mouse] on usb-3f980000.usb-1.5:1
[... cut ...]
[   36.939053] Bluetooth: HCI device and connection manager initialized
[   36.939070] Bluetooth: HCI socket layer initialized
[   36.939082] Bluetooth: L2CAP socket layer initialized
[   36.939104] Bluetooth: SCO socket layer initialized
[   43.629049] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   43.629060] Bluetooth: BNEP filters: protocol multicast
[   43.629083] Bluetooth: BNEP socket layer initialized
[   69.667764] logitech-hidpp-device 0003:046D:2011.0005: HID++ 1.0 device connected.

pelwell, 2018-04-13:
This line is significant:
```
[ 7.422137] mmc1: error -110 whilst initialising SDIO card
```
The driver isn't getting a response from the SDIO link to the WiFi interface. This could happen if the GPIO state isn't correct. What's the output from `raspi-gpio get`? (`sudo apt-get install raspi-gpio` if you don't have it)

davfsa, 2018-04-14:
Wait I fixed it.
The reason was that I had a wireless keyboard and mouse connected to the raspberry pi 3 and it thought that it was a wifi dongle. Now every time I boot up the raspberry pi 3 I need to unplug the wireless keyboard and mouse.

Any Ideas on how to make the Raspbery Pi 3 always use the default WiFi dongle?

JamesH65, 2018-05-11:
This sounds very much like a problem with the wireless devices and their driver rather than something specific to the Pi. Not sure there is anything we can do here. What make the are wireless devices causing the issue? When plugged in, what does `lsusb` show?

davfsa, 2018-05-23:
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

davfsa, 2018-05-23:
@JamesH65 sorry for the late response, I just booted up the raspberry pi and it works perfectly fine...

JamesH65, 2018-05-23:
OK, will close. Post here to reopen if it happens again.

davfsa, 2018-05-23:
Ok, it happened again....

JamesH65, 2018-05-23:
What manufacturer/model are the wireless keyboard and mouse that are causing the problem? Logitech I presume?

davfsa, 2018-05-23:
Yes

JamesH65, 2018-05-23:
I wonder if this could be a power problem. Lack of power on boot might stop the SDIO connection to the onboard Wifi working, perhaps the logitech dongle is soaking up a lot of current at that poin. Do you get a lightening bolt symbol on the top right of the display at all?

davfsa, 2018-05-23:
Nope, nothing at all

JamesH65, 2018-06-29:
@davfsa Sorry about delay. Can you do a sudo apt update/upgrade cycle to get the latest kernel and drivers. See if problem goes away, if no,  remove the Logitech dongle, boot device, then once booted, reinsert the dongle and post output of `dmesg | tail -20` just to get some idea of what's going on with the device probe.

ymouse91, 2018-08-09:
I have the same problem. It seems to be connected with Logitech usb dongle of my mouse. Sometimes rpi3 boot up fine sometimes not. Please note I've got a 2.5A power supply.

Here part of the log for failure:
[    5.160542] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[    5.405160] brcmutil: module license 'unspecified' taints kernel.
[    5.405165] brcmutil: module license 'unspecified' taints kernel.
[    5.405169] Disabling lock debugging due to kernel taint
[    5.405661] Unable to handle kernel paging request at virtual address 7f409234
[    5.405897] pgd = b455c000
[    5.405985] [7f409234] *pgd=00000000
[    5.406103] Internal error: Oops: 5 [#1] SMP ARM
[    5.406240] Modules linked in: cfg80211 rfkill snd_soc_bcm2835_i2s snd_bcm2835(C) snd_soc_core snd_compress snd_pcm_dmaengine i2c_bcm2835 snd_pcm snd_timer snd spi_bcm2835 fixed uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[    5.406872] CPU: 0 PID: 174 Comm: systemd-udevd Tainted: P         C      4.14.52-v7+ #1123
[    5.407104] Hardware name: BCM2835
[    5.407208] task: b5c5bc00 task.stack: b455a000
[    5.407348] PC is at strcmp+0x18/0x44
[    5.407463] LR is at cmp_name+0x1c/0x20
[    5.407579] pc : [<807953b0>]    lr : [<801a4438>]    psr: 20000113
[    5.407756] sp : b455bd48  ip : b455bd58  fp : b455bd54
[    5.407907] r10: 000007f5  r9 : 7f409234  r8 : 801a441c
[    5.408060] r7 : 809c9890  r6 : 00000008  r5 : 809cd838  r4 : 000007f4
[    5.408246] r3 : 00000008  r2 : 00000fea  r1 : 809fcece  r0 : 7f409234
[    5.408434] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    5.408636] Control: 10c5383d  Table: 3455c06a  DAC: 00000055
[    5.408801] Process systemd-udevd (pid: 174, stack limit = 0xb455a210)
[    5.408988] Stack: (0xb455bd48 to 0xb455c000)
[    5.409123] bd40:                   b455bd64 b455bd58 801a4438 807953a4 b455bd94 b455bd68
[    5.409356] bd60: 804aa278 801a4428 00000008 80803bc0 b455be3c 801a4590 00000000 80803bfc
[    5.409589] bd80: 7f3890b4 7f38a22c b455bdc4 b455bd98 801a45e0 804aa230 801a441c 02000000
[    5.409822] bda0: b46a00fc 80803bd4 b455be98 801a4590 b455be3c 80803bfc b455be34 b455bdc8
[    5.410054] bdc0: 801a53b4 801a459c b455bdf4 7f045024 7f045154 7f0452e4 00000000 00000000
[    5.410285] bde0: 7f045154 7f0452e4 7f04537c 00000001 7f38a000 00000000 00000000 00000000
[    5.410517] be00: 00000002 8079d700 804a183c 00000000 b455be98 7f38a04c 7f38a040 7f38902c
[    5.410751] be20: 7f3890b4 7f38a22c b455be64 b455be38 801a54ec 801a5384 7f38a040 7f409234
[    5.410982] be40: b4550001 b455be50 8079e7c0 8079c820 b455be98 b455beac b455bf14 b455be68
[    5.411216] be60: 801a813c 801a54b8 00000000 7f000000 00000000 b5d136c0 b455bf2c 7f38a22c
[    5.411448] be80: ba06d404 7f3915d8 7f38b000 000000c3 7f38a154 b455bf38 b455bee4 b455bea8
[    5.411686] bea0: 80290fa0 8028aca8 0000017b 7f389024 00000012 00000000 00000000 00000000
[    5.411949] bec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    5.412189] bee0: 00000000 00000000 7fffffff 00000000 00000010 76e36a18 0000017b 7fffffff
[    5.412423] bf00: 00000000 00000080 b455bfa4 b455bf18 801a8d64 801a67f4 7fffffff 00000000
[    5.412657] bf20: 00000003 8016cd94 00000002 ba06c000 00003d34 00000000 ba06d077 ba06c000
[    5.412890] bf40: 00003d34 ba06f8fc ba06da90 ba06e7d4 00003000 00003810 00000000 00000000
[    5.413122] bf60: 00000000 000015d8 00000019 0000001a 00000010 00000013 00000009 00000000
[    5.413353] bf80: 8010a51c 7ff7da00 00000000 00000000 80108204 b455a000 00000000 b455bfa8
[    5.413585] bfa0: 801081e0 801a8cd4 7ff7da00 00000000 00000010 76e36a18 00000000 021b3690
[    5.413819] bfc0: 7ff7da00 00000000 00000000 0000017b 00020000 0055c110 00000000 021b3690
[    5.414052] bfe0: 7eaf68a8 7eaf6898 76e2e3d0 76d134f0 60000010 00000010 55555555 55555555
[    5.414312] [<807953b0>] (strcmp) from [<801a4438>] (cmp_name+0x1c/0x20)
[    5.414519] [<801a4438>] (cmp_name) from [<804aa278>] (bsearch+0x54/0x6c)
[    5.414723] [<804aa278>] (bsearch) from [<801a45e0>] (find_symbol_in_section+0x50/0xd4)
[    5.414956] [<801a45e0>] (find_symbol_in_section) from [<801a53b4>] (each_symbol_section+0x3c/0x134)
[    5.415219] [<801a53b4>] (each_symbol_section) from [<801a54ec>] (find_symbol+0x40/0x6c)
[    5.415455] [<801a54ec>] (find_symbol) from [<801a813c>] (load_module+0x1954/0x23a0)
[    5.415687] [<801a813c>] (load_module) from [<801a8d64>] (SyS_finit_module+0x9c/0xac)
[    5.415923] [<801a8d64>] (SyS_finit_module) from [<801081e0>] (__sys_trace_return+0x0/0x10)
[    5.416164] Code: e24cb004 ea000001 e3530000 0a000008 (e4d03001) 
[    5.416346] ---[ end trace 08968c30eee816c5 ]---
[    6.768985] uart-pl011 3f201000.serial: no DMA platform data
[    7.353693] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.354148] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   12.837058] random: crng init done
......

And here after succesful
[    8.506290] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[    8.711932] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    8.717569] logitech-djreceiver 0003:046D:C52B.0003: hiddev96,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-3f980000.usb-1.5/input2
[    8.718779] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[    8.719035] usbcore: registered new interface driver brcmfmac
[... cut ...]
[    8.951577] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
[    8.954735] input: Logitech M705 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.2/0003:046D:C52B.0003/0003:046D:101B.0004/input/input1
[    8.956230] logitech-hidpp-device 0003:046D:101B.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech M705] on usb-3f980000.usb-1.5:1
[   10.199809] uart-pl011 3f201000.serial: no DMA platform data
[   10.434124] random: crng init done