raspberrypi/linux #2508: add CONFIG_F2FS_FS_XATTR=y

monetny, 2018-04-14:
I think CONFIG_F2FS_FS_XATTR=y is good idea for use f2fs as rootfs

PS more others built-in FS have it

Thx

monetny, 2018-04-14:
Possible need add CONFIG_F2FS_FS_POSIX_ACL too
I do not know

Ferroin, 2018-04-16:
For whoever from the foundation is looking at this, xattr support is needed for most LSM's other than AppArmor (including, but not limited to SELinux, SMACK, IMA, and EVM), as well as a number of user programs that are of particular interest when dealing with low-per-node-cost systems like the Pi (such as GlusterFS, which uses xattrs to store information about the synchronization state between nodes).

Also, adding POSIX ACL support has near zero overhead if you're already adding extended attribute support, though it's a rarely used feature in most cases (90% of the usage I've seen is in `/dev`, which doesn't need support for it in the root filesystem).

JamesH65, 2018-06-29:
I believe the current kernel (4.14.50, released today) has 

CONFIG_F2FS_FS=y
CONFIG_F2FS_STAT_FS=y
CONFIG_F2FS_FS_XATTR=y
CONFIG_F2FS_FS_POSIX_ACL=y

So I believe this covers the requirements in this issue.

closed 2018-06-29 by JamesH65