raspberrypi/linux #2509: DS1621 sensor driver kernel module is missing

soosp, 2018-04-14:
I'm running Raspbian 9.4 on an Raspebrry Pi 3 with the latest kernel (Linux pi1 4.14.33-v7+ #1109 SMP Tue Apr 10 17:28:38 BST 2018 armv7l GNU/Linux). I need to use a DS1621 I2C temperature sensor but the related kernel module is missing. Please include it into the next build.

soosp, 2018-04-14:
It would be nice to add a proper overlay too.
Thanks in advance.

soosp, 2018-04-14:
Sorry I did want to post it to the firmware repository.

pelwell, 2018-04-14:
No - this is the right place.

pelwell, 2018-04-15:
Done. It will be in the next firmware build.

pelwell, 2018-04-15:
Usage is:
```
dtoverlay=i2c-sensor,ds1621
```
or
```
dtoverlay=i2c-sensor,ds1621,addr=0x4c
```
The address can range from 0x48 (the default) to 0x4f.

Note I don't have a ds1621 to test, but it ought to work.

soosp, 2018-04-17:
Thank you! It works well:
# sensors
ds1621-i2c-1-48
Adapter: bcm2835 I2C adapter
temp1:        +26.5°C  (low  = +10.0°C, high = +60.0°C)  ALARM (HIGH)

I have to use it to monitor the internal temperature of the enclosure of Pi. This sensor is supported by watchdog daemon via the sysfs interface.

soosp, 2018-04-17:
BTW: You mistyped the adress range in the README of overlays.

closed 2018-04-17 by soosp